emsesp/EMS-ESP32 #1689: Fault message on Bosch compress 3800i ew: A11-3071

mattfro, 2024-04-07:
Just got a new EMS gateway from BBQKees(Gateway S3).
It's connected to a Bosch compress 3800i ew(should be same as IVT vent 402), and now I have a alarm which does not seem to go away, A11-3071.
Tried latest stable(3.6.5) and now this issue report is on latest Dev(3.7.0-dev.2)
I have tried to reset ems-esp gateway and turned off/on the heatpump.

What I noticed, fan control is missing(at least). It's currently set on 40%
Could that communication cause some problems or maybe it's something else. New with this esm-esp thing :)

**Not related to this issue: I do wonder also if it's possible to get bosch cr10/rc100 or cr5 rf to emulate indoor temperature. Seems at least cr5 rf(and mx200 as receiver) is "only" a temperature(maybe humidity sensor). No screen on it. 

[emsesp_system_allvalues.txt](https://github.com/emsesp/EMS-ESP32/files/14896610/emsesp_system_allvalues.txt)
[emsesp_system_info.txt](https://github.com/emsesp/EMS-ESP32/files/14896611/emsesp_system_info.txt)
[Take.04.-.change.temperature.txt](https://github.com/emsesp/EMS-ESP32/files/14896613/Take.04.-.change.temperature.txt)
[log.txt](https://github.com/emsesp/EMS-ESP32/files/14896615/log.txt)
[ems-esp-api-system.log](https://github.com/emsesp/EMS-ESP32/files/14896616/ems-esp-api-system.log)

MichaelDvP, 2024-04-08:
A11-3071 indicates that there was  remote thermostat connected and is now missing. 

Indeed in ***Take.04.-.change.temperature.txt*** there is `boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 09 01 00 00 00 00 01 00 00 01 00 00 00 00 00` configured devices 0x08, 0x0B, 0x10, 0x38 and 0x50.
Device 0x38 is a remote thermostat for hc1 and 0x50 is a M200/RFM200 module (product-id  218, seen in BF message).
***emsesp-system-allvalues.txt*** shows the entity `remoteseltemp` in thermostat. This value can only be set by a remote thermostat, not by ems-esp, but in ***emsesp_system_info.txtx*** these devices are not listed. 
Also in ***log.txt*** there is `boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 01 00 00 00 00 00 00 01 00 00 00 00` only devices 0x08, 0x09, 0x0B, 0x10, 0x48, no M200 and no remote thermostat, but in 'thermostat(0x10) -W-> Me(0x0B), RC300Set(0x02B9), data: 00 00 2A 00 1E 00 00 00 FF 00 2A 00 00 00 00 00 00 2C' the remoteseltemp is still set. 

It's difficult to analyse anything if you change configuration, devices, software version, etc. and mix all in one issue-report without explaining. The error message is because you had connected a remote and disconnected it again. It is not related to ems-esp.
> What I noticed, fan control is missing(at least)

See https://emsesp.github.io/docs/FAQ/#decoding-ems-telegrams

mattfro, 2024-04-08:
Hmm... I don't understand. Just tried to power the pump off, without ems-esp no alarm. Then connected it again, I get the alarm.
I don't have any indoor temperature sensor. Only sensor connected is outdoor to the pump.

So should I again from scratch, reset ems-esp gateway. turn off pump, turn it on. Leave it for a while, then turn it off and connect ems-esp gateway again. See if it's working?

mattfro, 2024-04-08:
I'm sorry, this was my bad. I get the alarm also without it, just took some time. And now I can't get rid of it... I wonder if some cables got disconnected or something. 
Or something got stored in the pump for a for a roomsensor, when I tried to activate it. But I disabled it and even reset ems-esp. Cut the power from the pump etc.

I can't find any settings in the pump to be able to activate or disable any roomsensor.  I have a feeling, it automatically knows if it's connected or not. But maybe it's stupid, once it thinks it has a roomsensor it will never forget it. I might need to do a full reset on the pump. Maybe that helps. Sorry for this ticket then. We can close it and open a new if needed.

closed 2024-04-08 by mattfro