emsesp/EMS-ESP32 #1692: Dwh mode is not correct on bosch 3800i/rego 3000

mattfro, 2024-04-11:
### PROBLEM DESCRIPTION
Missing mode from "dwh mode"
Rego3000, I think should have 3 modes:
Normal/Eco
Comfort
Eco+

When I set it Eco+ it does not show anything
[image]
This is when I set it to Eco+:
2024-04-11 07:45:55.035 T 22356: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode(0x02F5), data: 01 00 05 00 00 FF 08 06 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00

{"name":"wwmode","fullname":"dhw mode","circuit":"","type":"enum","enum":["off","normal","comfort","auto","own prog"],"readable":true,"writeable":true,"visible":true,"value":"not set"}

### TO REPRODUCE
Set hotwater mode to different modes, normal, comfort and eco+
Eco+ does not show in ems-esp
Here it's on comfort:
[image]

mattfro, 2024-04-11:
Maybe this can be closed? 
https://github.com/emsesp/EMS-ESP32/issues/1452
I have dev2 installed... or is it still wrong also in that build?

proddy, 2024-04-13:
is this still not working for you?

mattfro, 2024-04-13:
[image]
no it does not show anything on eco+

mattfro, 2024-04-13:
I have 3 modes: comfort, normal, eco+

proddy, 2024-04-23:
haven't looked at it yet. Can you send your system info so we can check the thermostat mapping

mattfro, 2024-04-23:
This?

{
  "System Info": {
    "version": "3.7.0-dev.4",
    "uptime": "000+00:30:30.001",
    "uptime (seconds)": 1830,
    "platform": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.14",
    "sdk": "v4.4.6-dirty",
    "free mem": 195,
    "max alloc": 183,
    "free caps": 8324,
    "used app": 2161,
    "free app": 5967,
    "partition": "app0",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -53,
    "TxPower setting": 0,
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Stockholm"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 574,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "esp32-463bdaec",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time water": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 4,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 9864,
    "bus reads (tx)": 1388,
    "bus writes (tx)": 1,
    "bus incomplete telegrams": 2,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32S3",
    "locale": "en",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 50,
    "web log buffer": 28
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i",
      "device id": "0x08",
      "product id": 172,
      "version": "03.01",
      "entities": 106,
      "handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC 0x04A5",
      "handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x0486 0x04AE 0x04AF",
      "handlers ignored": "0xF7 0x04A0 0x04A3 0x04B4"
    },
    {
      "type": "thermostat",
      "name": "Rego 2000/3000",
      "device id": "0x10",
      "product id": 172,
      "version": "03.01",
      "entities": 50,
      "handlers received": "0x06 0x031D",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x0467 0x0291 0x0468 0x0292 0x02F5 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x02CE 0x02D0 0x0469 0x0293 0x02D2 0x046A 0x0294 0x02F6 0x031B",
      "handlers ignored": "0x042B 0xF7"
    },
    {
      "type": "thermostat",
      "name": "RC200/CW100",
      "device id": "0x38",
      "product id": 157,
      "version": "32.02",
      "entities": 1,
      "handlers received": "0x042B"
    },
    {
      "type": "gateway",
      "name": "KM200/MB LAN 2",
      "device id": "0x48",
      "product id": 189,
      "version": "04.08",
      "entities": 0,
      "handlers ignored": "0xF9"
    },
    {
      "type": "controller",
      "name": "Rego 3000",
      "device id": "0x09",
      "product id": 240,
      "version": "38.03",
      "entities": 1,
      "handlers received": "0x06",
      "handlers ignored": "0xC6 0xC7 0xBF"
    }
  ]
}

proddy, 2024-04-28:
@mattfro could you capture the telegram 0x2F5 when you switch to "Normal/Eco", "Comfort" as well like you did with "Eco+" so we can make sure we have the correct mapping?

@MichaelDvP The Rego3000 thermostat has different wwmode settings so suggest 
- create a new EMSdevice::EMS_DEVICE_FLAG_R3000 and set productID 172 in `device_library.h` for this thermostat
- create a new `enum_wwMode5` with the Normal/Eco, Comfort and Eco+
- update `Thermostat::process_RC300WWmode()` , `register_device_values_dhw()` and `Thermostat::set_wwmode()`

We can test with the standlone version, using `test add 0x10 172 ; call thermostat mode`

mattfro, 2024-04-28:
@proddy  normal and comfort works fine in ems-esp. But just in case I fetched them also as you requested  :)

Normal/eco:
001+19:17:32.311 N 107: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode(0x02F5), data: 01 00 01 00 00 FF 08 06 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00
Comfort:
001+19:18:10.274 N 240: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode(0x02F5), data: 01 00 02 00 00 FF 08 06 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00
Eco+:
001+19:18:45.473 N 338: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode(0x02F5), data: 01 00 05 00 00 FF 08 06 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00

proddy, 2024-04-28:
perfect, so 

Normal/Eco = 1
Comfort = 2
Eco+ = 5

mattfro, 2024-04-28:
Actually this is how it looks, missed the message before thermostat...
001+19:33:49.798 N 2639: [emsesp] gateway(0x48) -W-> thermostat(0x10), RC300WWmode(0x02F5), data: 02 (offset 2)
001+19:33:50.221 N 2640: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode(0x02F5), data: 01 00 02 00 00 FF 08 06 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00

But yes, that is how it is.

proddy, 2024-04-28:
I've added this to my PR. Are you able to build locally or do you want a .bin file?

mattfro, 2024-04-28:
> I've added this to my PR. Are you able to build locally or do you want a .bin file?

I have no clue how to build myself, sorry :)
But I'm in no hurry, if that comes in later in some later dev or release. So no stress

proddy, 2024-04-29:
@mattfro in dev, can you test?

mattfro, 2024-04-29:
@proddy  it works :) thanks
[image]

proddy, 2024-04-30:
re-opening, write of dwh mode for some users is not working

Examples:

On a Thermostat ProductID 172/Rego3000:

2024-04-29 22:16:12.984 I 86: [command] Calling command 'thermostat/mode' (mode) with value normal and id 9 on device 0x10
2024-04-29 22:16:12.984 E 87: [command] Command 'mode/normal' failed with error code 3
2024-04-29 22:16:12.984 E 88: [emsesp] Write command failed callback function failed (Error)

and on Discord someone called project said

Hi, on version 3.7.0-dev.6 , I can't change the settings " DHW mode ".  Directly in the EMS-ESP  / end with error: [emsesp] Write command failed callback function failed (Error) /   or via Home Assistant  / there is error:  [command] Command 'mode/on' failed with error code 3 / .

projectile25, 2024-04-30:
Hi, as i wrote on Discord /project/ , i use thermostat Buderus RC35 - and  up to version 3.7.0-test.0  it worked without a problem .

MichaelDvP, 2024-04-30:
@proddy Found the reason, we have now `hc/mode` and `dhw/mode`, and command.cpp tries to send to hc/mode.
We need to add the `flag` in `find_command()` to map to the right function. I think there is also a bug dhw telegram mapping for the commands. I'll check and make a PR.

proddy, 2024-04-30:
Thanks, good fix. @projectile25 and @mattfro would you like to check again with the latest dev?

projectile25, 2024-04-30:
Hi, yes its working again . thank you

mattfro, 2024-04-30:
@proddy  tested it from ems-esp and pump it works now. Thanks

proddy, 2024-04-30:
thanks both

closed 2024-04-30 by proddy