emsesp/EMS-ESP32 #1693: Improvment: Get current error state or alarm as a entity

mattfro, 2024-04-11:
I know there is at the moment Last know error code, which is really nice. 

But I would like to have the active or current alarm code, because I want to use a template in homeassistant that shows what the error code is or means which is currently active:

example:
- sensor:
  - name: "Heatpump alarm text"
    state: >
        {% set mapper = {
          '0': 'Ingen alarm',
          '1001': 'Ingen kommunikation mellan reglercentral och fjärrstyrning',
          '1010': 'Ingen kommunikation via BUS-förbindelsen EMS',
          '1037': 'Utegivare defekt - ersättningsdrift värme aktiv',
          '1038': 'Ogiltigt värde tid/datum',
          '1041': 'Spänningsbortfall under urtorkning',
          '1051': 'Ingen kommunikation med modul för extern rumsgivare',
...
        {% set x =  states.sensor.**boiler_current_code**.state %}
        {{ mapper[x] if x in mapper else 'Unknown' }}

This is for rego 3000 maybe other systems also...

Would this be a possible improvement?

proddy, 2024-04-11:
the error code is fetched on EMS-ESP startup and when there's a change. You can try and force the fetch yourself using the scheduler, and send a read request to 0x10 or 0x11. To test using the console and `read 8 10`  or `read 8 11` and see if it brings back the correct value.

MichaelDvP, 2024-04-11:
The last error shows start and end of the error, if end is noted as `now` the error is still active. But you can also try a custom entity with BF message. On error a device send the 0xBF message with offset 8/9 the error-number. offset 0 is the device-id, offset 1 the product-id, offset 5/6/7 the 3 bytes ASCII (e.g. A11). For the error number use a custom entity with USHORT and offset 8. But you have to create it for each device connected to the bus.
The other fields are also known, see Norberts1 list: https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf

proddy, 2024-04-11:
@MichaelDvP as a possible new addition, we could add a new command called 'read' to each device, so you can 'boiler/read' with value '12' (for example) from the Scheduler. Not related to this issue but got me thinking

mattfro, 2024-04-11:
I have the whole list already in hass, or almost all. I took it from Bosch webpage added in to hass. I could use the last know error code, but then I would need to grep a part from the entity so I only get that code.

Or I will try with the read as proddy mentioned, to see what I get.

mattfro, 2024-04-11:
Done similar to a Nibe heatpump(esp-nibe) integration in hass also. Some work, but not that bad :)

Anyways, thanks... will try later today or tomorrow(when I most likely have more time to test)

MichaelDvP, 2024-04-11:
Hmm, i've checked the BF/BE telegrams, they are not readable, i get only `<empty>` back. 
So they are only broadcasted on an error event (set/unset)
@mattfro for the errorcode in last_error you can create a custom entity device 0x08, type-id 0xC2, offset 8, USHORT. 
if offset 15 (UINT) is < 128 the error is still active.
ems/ht3 boilers uses telegrams 0x10 (blocking errors) and 0x11 (locking errors) and hve C2 emty, but i think you have only C2.

mattfro, 2024-04-11:
Does it help If I get logs when I will mess with the remotetemp and most likely will get an alarm? :)
Going to try that reset alarm custom entity at the same point :)