emsesp/EMS-ESP32 #1695: Improvment: Reset alarm and clear alarm history list

mattfro, 2024-04-12:
This works on bosch 3800i at least...could work on other systems also.
What I noticed sometimes it's not possible to write 255, I wonder if it's somehow related if the alarm is really active or not.
Because in those cases, I can not even reset it on the pump, it does not go away.

**controller(0x09) -W-> boiler(0x08), ?(0x05), data: FF (offset 8)**
But it seems to work, had alarm on the UI code 3071 and run this custom entity with value 255:
[image]
and the alarm went away

**controller(0x09) -W-> boiler(0x08), ?(0x05), data: 01 (offset 42)**
And this clears the alarm history list from the UI/pump:
[image]
with value 1

mattfro, 2024-04-12:
and if there is no active alarm... it also complains. Which I think is pretty obvious. :)
000+00:34:09.131 W 2455: [command] Command failed: invalid command 'alarm_reset'
000+00:34:09.131 E 2456: [emsesp] Write command failed invalid command (Not Found)

MichaelDvP, 2024-04-12:
Thanks, the first command is same as `boiler/reset maintenance`, I'll add `history` option to the `boiler/reset` command.
@proddy  Should we change the text from `maintenance` to `message` or add a `message` options with same ems-command as `maintenance`?

proddy, 2024-04-12:
> Thanks, the first command is same as `boiler/reset maintenance`, I'll add `history` option to the `boiler/reset` command. @proddy Should we change the text from `maintenance` to `message` or add a `message` options with same ems-command as `maintenance`?

yes, let's rename!

MichaelDvP, 2024-04-12:
> yes, let's rename!

Hmm, i see that maintenance reset also verifies the reset by reading the maintenance message, that is not nessesery for message reset, so maybe better two options?

proddy, 2024-04-12:
> > yes, let's rename!
> 
> Hmm, i see that maintenance reset also verifies the reset by reading the maintenance message, that is not nessesery for message reset, so maybe better two options?

yes we have to then.

MichaelDvP, 2025-04-28:
@proddy This can also be closed, implemented since a year, https://github.com/emsesp/EMS-ESP32/commit/6a16027e16b006c7652b6a698a1f62048098a8c2