emsesp/EMS-ESP32 #1696: Improvemt: Add power off/on possibility for the heatpump(bosch compress 3800i ew)

mattfro, 2024-04-13:
I made a custom entity from the message when I turned off/on the heatpump.
[image]

Value 1 turns off the pump, value 0 turns on the pump. Possible to add it as an entity?

MichaelDvP, 2024-04-14:
I think using bool as on/off is better, but with we have only 1=on/0=off, so we have to find a name that makes clear the option on/true/1 shuts the heatpump down. What about `shutdown` as name?

mattfro, 2024-04-14:
Shutdown is fine.
Not possible to invert bool, I mean like have a tag in ems-esp that you want to invert it? Maybe only for one situation it's too much work. Of course I can always do that in hass... so for me it's not a big deal.

proddy, 2024-04-14:
@MichaelDvP we could ENUM as a new type, with a list of options, in this case it would be '1, 0'

MichaelDvP, 2024-04-14:
> Not possible to invert bool,

Sure, we can add a new type BOOL_INV, but is it worth for a single entity?
> we could ENUM as a new type

Yes, this should work, but still a bit confusing when using enum as number option, then it's the only entity with on=0. 
Also i think `power_onoff` is not a good name, it's not switching the power, it just stops the pump. For modules we have always the `enable` entity, maybe here we can use `disable`? 
@mattfro What is the Bosch name in manual and UI of the thermostat for this option?

mattfro, 2024-04-15:
> > Not possible to invert bool,
> 
> Sure, we can add a new type BOOL_INV, but is it worth for a single entity?
> 
Yeah, that is the question :)
> Yes, this should work, but still a bit confusing when using enum as number option, then it's the only entity with on=0. Also i think `power_onoff` is not a good name, it's not switching the power, [...]

Hmm... bosch name in manual, what thermostat option are you thinking about? Not sure If I understand what you mean...
Are you talking what it's called when I turn off the pump?
If that is the case, it's: "Turn off the heatpump" and then when it's turned off, it ask "turn on the heatpump?"

MichaelDvP, 2024-04-15:
>  it's: "Turn off the heatpump" and then when it's turned off, it ask "turn on the heatpump?"

Changing the entity name is not possible for us, this will give a lot of issues with homeassistant. I prefere a bool state with name `disable` or `shutdown`, what is your preference or suggestion? Also you can rename the entity later in the customization., but for a bool the state `on` will be send a one to the bus and stops the heatpump.

mattfro, 2024-04-15:
In Swedish, basically: Shut down heatpump
[image]
And the next, Turn off(No/Yes), When heatpump is turned off, heatpump and display will turn off. In off state the antifreeze??prevention is going to be off. To reactivate it touch the display. 
[image]
Something like that :)

So when it's off, similar window will popup and ask if you want to start the heatpump(yes/no)

mattfro, 2024-04-22:
So will this be in, in 3.7?
In that case we can close this ticket... 
I can see shutdown entity, but it's hidden. I guess it's because it has no value or am I mistaken?

Found when using customize entity
[image]

proddy, 2024-04-22:
Entities were initially designed in EMS-ESP for both reading and/or writing. Then we introduced commands like 'reset'. In this case the shutdown is a command (there is no value to read). 

@MichaelDvP shall we introduce a new type in the WebUI for commands (`DeviceValueType::CMD`)?

MichaelDvP, 2024-04-22:
It is in 3.7dev and telegram 484 is fetched and has other values. It is not a command telegram. The value should show up.

@proddy For DeviceValueType::CMD we need to know the byte length. Custom entities show in dashboard when writeable, also without value to match such a case we need a cmd only.

proddy, 2024-05-06:
@mattfro are you still not seeing a value for hpshutdown ?

mattfro, 2024-05-06:
@proddy  yep...
[image]

proddy, 2024-05-06:
@MichaelDvP what shall we do here? It looks like shutdown is an entity with no value set, so it can't be invoked.

MichaelDvP, 2024-05-06:
We can change the type to DeviceValueType::CMD

mattfro, 2024-05-06:
@MichaelDvP I tried your latest build, it just seems to restart all the time. The one you build 10-15 min ago.

MichaelDvP, 2024-05-06:
Can not reproduce, downloaded both bin-files (esp32 and esp32S3) from github and flashed them, both working here. 
Try power-on-reset. If not successful check with serial terminal the boot messages.

mattfro, 2024-05-06:
Did not see anything else the few seconds I have... only something about ipv6. 
If I manage to load the official dev.10 it works again. Need to see if I can get serial connection...
[image]
it just reboot's without any information.
[image]

proddy, 2024-05-06:
it's either an out of bounds error or out of memory error that is causing the EMS-ESP to crash and restart.

MichaelDvP, 2024-05-06:
Looks like normal boot, i can not see a reboot with stack registers. 
Does the crash happen after the 7 sec uptime? 
Could only be the register of bool CMD `shutdown`, have to check, my boiler does not have this entity.

mattfro, 2024-05-06:
yes it looks exactly like that... it would just be a reboot without any reason. It's no exactly 7 seconds, sometimes a bit longer sometimes shorter. @MichaelDvP could you make a new build without shutdown of the latest? Just to be sure it's not that. Which is a bit strange if it would cause it to reboot.

MichaelDvP, 2024-05-06:
Yes it is, i've found the typo in the shutdown registering. Will push a new build.

mattfro, 2024-05-06:
Much better, thanks @MichaelDvP :)
Shutdown also works, on= shuts down the pump and off= turns it on. :)

mattfro, 2024-05-06:
So, it's time to close this one also then? ;)

proddy, 2024-05-08:
yes!

closed 2024-05-08 by proddy