emsesp/EMS-ESP32 #1697: Add Call feature

proddy, 2024-04-15:
I was thinking of adding a small new feature to call commands, basically mimicking what the console/telnet `call <device><cmd> <data>` does. When testing, I often need to run some commands like `test`. It's not too important to add, but also simple to code. We could limit it just to system commands and put it in the System menu as a new Tab, or make it a global feature and have its own page and extend it to call EMS devices, analogsensor, temperature sensors, etc...

What do you think @MichaelDvP  ?

```
ems-esp:# call
Showing all available commands (*=authentication not required):

 system:
  allvalues                     output all values *
  commands                      lists all commands *
  fetch                         refresh all EMS values
  info                          show system info *
  publish                       publish all to MQTT *
  response                      get response *
  restart                       restart EMS-ESP
  send                          send a telegram
  test                          run a test *
  watch                         watch incoming telegrams *

 custom:
  info:                         lists all values *
  commands:                     lists all commands *

 scheduler:
  info:                         lists all values *
  commands:                     lists all commands *

 temperaturesensor:
  info:                         lists all values *
  commands:                     lists all commands *

 analogsensor:
  info:                         lists all values *
  commands:                     lists all commands *
  setvalue                      set io value

 boiler:
  boil2hystoff                  hysteresis stage 2 off temperature
  boil2hyston                   hysteresis stage 2 on temperature
  boilhystoff                   hysteresis off temperature
  boilhyston                    hysteresis on temperature
  burnmaxpower                  burner max power
  burnminperiod                 burner min period
  burnminpower                  burner min power
  coldshot                      send a cold shot of water
  commands                      lists all commands *
  curvebase                     heatingcurve base
  curveend                      heatingcurve end
  curveon                       heatingcurve on
  emergencyops                  emergency operation
  emergencytemp                 emergency temperature
  entities                      lists all entities *
  heatingactivated              heating activated
  heatingoff                    force heating off
  heatingtemp                   heating temperature
  info                          lists all values *
  maintenance                   maintenance scheduled
  maintenancedate               next maintenance date
  maintenancetime               time to next maintenance
  nofrostmode                   nofrost mode
  nofrosttemp                   nofrost temperature
  nompower                      nominal Power
  nrgheat                       energy heating
  nrgww                         dhw energy
  pumpdelay                     pump delay
  pumpmode                      boiler pump mode
  pumpmodmax                    boiler pump max power
  pumpmodmin                    boiler pump min power
  reset                         reset
  selburnpow                    burner selected max power
  selflowtemp                   selected flow temperature
  summertemp                    summer temperature
  wwactivated                   dhw activated
  wwchargeoptimization          dhw charge optimization
  wwcirc                        dhw circulation active
  wwcircmode                    dhw circulation pump mode
  wwcircpump                    dhw circulation pump available
  wwcomfort                     dhw comfort
  wwcomfort1                    dhw comfort mode
  wwdisinfecting                dhw disinfecting
  wwdisinfectiontemp            dhw disinfection temperature
  wwflowtempoffset              dhw flow temperature offset
  wwhystoff                     dhw hysteresis off temperature
  wwhyston                      dhw hysteresis on temperature
  wwmaxpower                    dhw max power
  wwmaxtemp                     dhw maximum temperature
  wwonetime                     dhw one time charging
  wwseltemp                     dhw selected temperature
  wwseltemplow                  dhw selected lower temperature
  wwseltempsingle               dhw single charge temperature
  wwtapactivated                dhw turn on/off
  wwtempecoplus                 dhw selected eco+ temperature

 thermostat:
  [hc<n>.]boost                 boost mode
  [hc<n>.]boosttime             boost time
  building                      building type
  [hc<n>.]comforttemp           comfort temperature
  commands                      lists all commands *
  [hc<n>.]control               control device
  [hc<n>.]controlmode           control mode
  [hc<n>.]cooling               cooling
  damping                       damping outdoor temperature
  datetime                      date/time
  delayboiler                   delay boiler support
  [hc<n>.]designtemp            design temperature
  [hc<n>.]dewoffset             dew point offset
  [hc<n>.]ecotemp               eco temperature
  electricfactor                electric energy factor
  energycostratio               energy cost ratio
  entities                      lists all entities *
  [hc<n>.]fastheatup            fast heatup
  fossilefactor                 fossile energy factor
  [hc<n>.]heatingtype           heating type
  [hc<n>.]heatoffdelay          heat-off delay
  [hc<n>.]heatondelay           heat-on delay
  [hc<n>.]hpminflowtemp         HP min. flow temp.
  [hc<n>.]hpmode                HP Mode
  [hc<n>.]hpoperatingmode       heatpump operating mode
  hybridstrategy                hybrid control strategy
  info                          lists all values *
  [hc<n>.]instantstart          instant start
  intoffset                     internal temperature offset
  [hc<n>.]manualtemp            manual temperature
  [hc<n>.]maxflowtemp           max flow temperature
  minexttemp                    minimal external temperature
  [hc<n>.]minflowtemp           min flow temperature
  [hc<n>.]mode                  mode
  [hc<n>.]nofrostmode           nofrost mode
  [hc<n>.]nofrosttemp           nofrost temperature
  [hc<n>.]noreducetemp          no reduce below temperature
  [hc<n>.]offsettemp            offset temperature
  [hc<n>.]program               program
  pvenableww                    enable raise dhw
  pvlowercool                   lower cooling with PV
  pvraiseheat                   raise heating with PV
  [hc<n>.]reducemode            reduce mode
  [hc<n>.]reducetemp            off/reduce switch temperature
  [hc<n>.]remotehum             room humidity from remote
  [hc<n>.]remotetemp            room temperature from remote
  [hc<n>.]roominflfactor        room influence factor
  [hc<n>.]roominfluence         room influence
  [hc<n>.]roomtempdiff          room temp difference
  [hc<n>.]seltemp               selected room temperature
  [hc<n>.]summersetmode         set summer mode
  [hc<n>.]summertemp            summer temperature
  [hc<n>.]switchonoptimization  switch-on optimization
  switchovertemp                outside switchover temperature
  [hc<n>.]tempautotemp          temporary set temperature automode
  tempdiffboiler                temp diff boiler support
  wwcharge                      dhw charge
  wwchargeduration              dhw charge duration
  wwcircmode                    dhw circulation pump mode
  wwdailyheating                dhw daily heating
  wwdailyheattime               dhw daily heating time
  wwdisinfectday                dhw disinfection day
  wwdisinfecting                dhw disinfecting
  wwdisinfecttime               dhw disinfection time
  wwmode                        dhw mode
  [hc<n>.]wwprio                dhw priority
  wwsettemp                     dhw set temperature
  wwsettemplow                  dhw set low temperature
```

MichaelDvP, 2024-04-26:
I always use telnet, I compile with debug and my console defaults to log-debug and su activated.  
With the now fine working web-log i thought of adding watch and call input field within the log screen. 
But i think both is not good for normal users, they tend to filter too much, using `watch unknown`, etc. It's always easier to analyse with full log to see the context.

But for running test commands and check web it could be usefull, i do not use the test command often. Maybe add it only for compiling with -DEMSESP_TEST

proddy, 2024-04-26:
okay, thanks for that perspective

closed 2024-04-27 by proddy