emsesp/EMS-ESP32 #1704: New Feature: IFTTT

proddy, 2024-04-21:
Integrate webhooks into https://ifttt.com/

Not sure its useful (I won't use it) but https://husdata.se/docs/h60-manual/webhooks-ifttt/ has it and we're better ;-)

MichaelDvP, 2024-04-21:
Why contact a cloud service from every devcie if you can do the logic  in your home automation? If you really need IFTTT, you can do this also from HA or ioBroker, etc. I think it's useless. 
Husdata maybe have this for their own (payed, but optional) cloud service for users without home automation.

proddy, 2024-04-21:
100% agree. There might be some users without HA/ioBroker that want to control EMS-ESP from their mobile phones, using the native IFTTT app. Apart from that I agree its useless.

closed 2024-04-25 by proddy

MichaelDvP, 2024-06-15:
@proddy @bbqkees 
We have taked about conditions some times ago here, in scheduler discussion and when implementing custom variables. I have now a first implementaion of conditions for the scheduler. My own use case is to use the relais of gateway V2 to control the solar assist valve, now done by ioBroker script and a sonoff module. To control i need a expression parser for logical and arithmetric functions. 
relais on for `mixer/hc2/pumpstatus == 1 && solar/cylbottomtemp > (temperaturesensor/boiler_backflow + 4)`
relais off for `mixer/hc2/pumpstatus == 0 || solar/cylbottomtemp <= temperaturesensor/boiler_backflow`
First implementation looks like this:
[image]
and is working now (but still crashes if the condition could not be parsed correctly). 

Another possible use case is for users without thermostat to implement a heatingcurve, than calculate the command value:
[image]

Also implemented is a possible trigger on change, e.g. the heatingcurve can be recalculated on every change of outdoortemp, not periodic.

I'll fix the crashes and make a feature-build for testing.

proddy, 2024-06-15:
amazing! You've been busy building something that is actually super useful, while I've been busy playing around with pluggable external libraries that we don't need (but it was fun, and I learned a lot about Pio and C++ templating).

Would love to see how you programmed the conditions pre-processor. That couldn't have been easy.

MichaelDvP, 2024-06-15:
> Would love to see how you programmed the conditions pre-processor. That couldn't have been easy.

I've started with this compact shuntingYard: https://gist.github.com/t-mat/b9f681b7591cdae712f6 and added logical and command parsing.

MichaelDvP, 2024-06-15:
https://github.com/MichaelDvP/EMS-ESP32/tree/feat_conditions

But it calculates only numbers. For conditions with strings like `thermostat/hc1/mode == auto` i need some changes.

proddy, 2024-06-19:
We can use https://github.com/emsesp/EMS-ESP32/issues/1806 to track this from now on

bbqkees, 2024-06-19:
@MichaelDvP Looks great!
Get a lot of questions about a feature like this.