emsesp/EMS-ESP32 #1708: Change of Operating mode in thermostat via HomeAssistant not possible since dev.4

SeibeGitHub, 2024-04-23:
[v3.7.0-dev.4]

Since version dev4 it is not possible to change the operating mode (auto/manual) in the thermostat via Home Assistant, but when downgrading to dev2 it works again.

proddy, 2024-04-23:
does it work via the EMS-ESP WebUI or using the EMS-ESP Console? It will help us pinpoint the issue. If so, then please try again and report back any error messages in HA and also what the logs say in EMS-ESP

SeibeGitHub, 2024-04-24:
The behavior is also incorrect with the WebUI

[log.txt](https://github.com/emsesp/EMS-ESP32/files/15095189/log.txt)

proddy, 2024-04-26:
```
2024-04-24 15:39:22.781 INFO 122: [command] Calling command 'thermostat/mode' (Modus) with value heat and id 1 on device 0x10
2024-04-24 15:39:22.781 ERROR 123: [command] Command 'mode/heat' failed with code: 3
2024-04-24 15:39:22.781 ERROR 124: [mqtt] MQTT command failed with error: callback function failed (Error)
```

proddy, 2024-04-26:
Please attach your system info so we can simulate your environment.

SeibeGitHub, 2024-04-28:
Hello, now in v3.7.0 dev.6 the problem is gone. Works now completly well.

proddy, 2024-04-28:
strange, I don't know what changed. But glad it's fixed.

closed 2024-04-28 by proddy