emsesp/EMS-ESP32 #1716: Error 'invalid template (TemplateSyntaxError: expected token 'end of print statement',

bakboord4, 2024-04-25:
### PROBLEM DESCRIPTION

Logger: homeassistant.components.mqtt.mixins
Bron: components/mqtt/mixins.py:270
integratie: MQTT (documentatie, problemen)
Eerst voorgekomen: 16:05:24 (1 gebeurtenissen)
Laatst gelogd: 16:05:24

Error 'invalid template (TemplateSyntaxError: expected token 'end of print statement', got 'wayvalve') for dictionary value @ data['value_template']' when processing MQTT discovery message topic: 'homeassistant/binary_sensor/NefitCV/boiler_dhw_3wayvalve/config', message: '{'unique_id': 'boiler_dhw_3-way_valve_active', 'object_id': 'boiler_dhw_3-way_valve_active', 'name': 'dhw 3-wegklep actief', 'state_topic': 'NefitCV/boiler_data', 'payload_on': 'aan', 'payload_off': 'uit', 'value_template': "{{value_json.dhw.3wayvalve if value_json.dhw is defined and value_json.dhw.3wayvalve is defined else 'uit'}}", 'availability': [{'topic': 'NefitCV/boiler_data', 'value_template': "{{'online' if value_json.dhw is defined and value_json.dhw.3wayvalve is defined else 'offline'}}"}], 'availability_mode': 'all', 'device': {'identifiers': ['NefitCV-boiler'], 'name': 'NefitCV Boiler'}, 'platform': 'mqtt'}'

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

Systems development is not my habitat.

You are currently on version 3.7.0-dev.4 (ESP32)

```

### TO REPRODUCE

restart HA

### EXPECTED BEHAVIOUR

Not producing the reported error

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-04-25:
I get the same error withg 3.7.0-dev.4. Looking into why.

MichaelDvP, 2024-04-25:
> I get the same error withg 3.7.0-dev.4. Looking into why.

Maybe HA do not like numbers at the beginning of the name `3wayvalve` (was `ww3wayvalve`, now `dhw/3wayvalve`)

proddy, 2024-04-25:
yes, so we need to change the code and replace for example `value_json.dhw.3wayvalve` with `value_json.dhw['3wayvalve']`.

in `publish_ha_sensor_config()` this code

```
        if (is_nested() && tag >= DeviceValueTAG::TAG_HC1) {
            snprintf(val_obj, sizeof(val_obj), "value_json.%s.%s", EMSdevice::tag_to_mqtt(tag), entity);
            snprintf(val_cond, sizeof(val_cond), "value_json.%s is defined and %s is defined", EMSdevice::tag_to_mqtt(tag), val_obj);
        } else {
            snprintf(val_obj, sizeof(val_obj), "value_json.%s", entity);
            snprintf(val_cond, sizeof(val_cond), "%s is defined", val_obj);
        }
```

replacing the 3 json_value's

Will you do it or shall i do after your PR?

MichaelDvP, 2024-04-25:
My dev is with changes in thermostat dhw and the HA compatibility options. If this is ok, i can add it and make a PR. 
Does `value_json.%s.%s` become `value_json[' %s.%s']` or an other notation, maybe `value_json['%s'.'%s']` or `value_json['%s']['%s']`?

MichaelDvP, 2024-04-25:
Yes, that `"value_json.%s['%s']"` should work, but i've tried to allow also special chars in the tag. As we don't have them, stick to quote the entities only. 
@bakboord4 Please check https://github.com/MichaelDvP/EMS-ESP32/releases
If this works for you, i'll make a PR.

bakboord4, 2024-04-25:
@MichaelDvP. I'll give it a try.

bakboord4, 2024-04-25:
@MichaelDvP 
Result after a total restart of HA.

Logger: homeassistant.components.mqtt.mixins
Bron: components/mqtt/mixins.py:270
integratie: MQTT (documentatie, problemen)
Eerst voorgekomen: 20:03:11 (1 gebeurtenissen)
Laatst gelogd: 20:03:11

Error 'invalid template (TemplateSyntaxError: expected token 'end of print statement', got 'wayvalve') for dictionary value @ data['value_template']' when processing MQTT discovery message topic: 'homeassistant/binary_sensor/NefitCV/boiler_dhw_3wayvalve/config', message: '{'unique_id': 'boiler_dhw_3-way_valve_active', 'object_id': 'boiler_dhw_3-way_valve_active', 'name': 'dhw 3-wegklep actief', 'state_topic': 'NefitCV/boiler_data', 'payload_on': 'aan', 'payload_off': 'uit', 'value_template': "{{value_json.dhw.3wayvalve if value_json.dhw is defined and value_json.dhw.3wayvalve is defined else 'uit'}}", 'availability': [{'topic': 'NefitCV/boiler_data', 'value_template': "{{'online' if value_json.dhw is defined and value_json.dhw.3wayvalve is defined else 'offline'}}"}], 'availability_mode': 'all', 'device': {'identifiers': ['NefitCV-boiler'], 'name': 'NefitCV Boiler'}, 'platform': 'mqtt'}'

bakboord4, 2024-04-25:
@MichaelDvP  after applying 3.7.0 dev.5

bakboord4, 2024-04-26:
@proddy @MichaelDvP. Good to hear that. Please continue, I do not have time at this moment to do a complete re-install.