emsesp/EMS-ESP32 #1717: Improvment: Get the current power consumption of Compress 7001 Heatpumps

RaHehl, 2024-04-26:
My Setup;

Heatpump: Bosch CS7000iAW OR-S & AWMB Inside
PV System: Fronius Symo GEN24 10.0 Plus Hybrid Inverter & BYD Battery-Box Premium HVM-11.0 &  Fronius Smart Meter 63A-3
Bosch Energy manager running on a Bosch Smart Home Controller 2 https://www.bosch-homecomfort.com/de/de/wohngebaeude/wissen/heizungssteuerung/der-energiemanager/
Connected to the Heatpump via Ethernet and via TCP Modbus to my inverter
Wallbox 2x Fronius Wattpilot HOME 11J
latest Homeassistant

I would like to have this value inside my HomeAssistant, displayed in the energiemanager app:

[image]

that's what i can get via Longpoling without filter in the energie manager api:
[image]

[image]

[image]

[log (3).txt](https://github.com/emsesp/EMS-ESP32/files/15126189/log.3.txt)

MichaelDvP, 2024-04-26:
Does the graph show the value you are searching for? The y-scale is missing. We need to have the numbers of the changes for time and value. The log has~200 telegrams per minute and it's hard to see the exact time in the graph.
Can you give numbers for each change: time, from value, to value

proddy, 2024-04-26:
Adding the dicsord discussion for reference [here](https://discord.com/channels/816637840644505620/816958041345884180/1233291115386372137) as it contains some good points.

BBQKees said 

> Do you have the new Bosch Energy Manager System with the additional power meter and that the system controls the actual compressor speed based on the PV availability?
https://www.bosch-homecomfort.com/de/de/wohngebaeude/wissen/heizungssteuerung/der-energiemanager/
If so, we would like to capture with what commands on the EMS bus the Energy Manager controls the compressor speed.
Also for your electricity consumption if it's sent over the EMS bus you we can try to find it in the data. In the Gateway menu you can log the EMS bus data via System->Log. Set it to Log Level ALL and to 100 items.
Best is to enable NTP in the settings first so the time is accurate. Then make a note of values you see in the app together with the exact time they occur (up to the seconds precise). For each parameter you want to extract we need multiple different values over time.
So if a compressor speed is always 100%, its harder to see it in the raw data. We need changes. If we know f.i. it's 12% on 14:22:01 and 35% on 14:23:55 we (actually usually our hexadecimal mastermind Michael) can find the values and thus the position and data type much easier in the raw EMS bus data.

RaHehl, 2024-04-26:
The power consumption changes with compressor speed, can we filter the telegrams for these changes to have the timeframe? The app display and HA ist delayed

RaHehl, 2024-04-26:
Dataflow:
[image]

MichaelDvP, 2024-04-26:
> The power consumption changes with compressor speed, can we filter the telegrams for these changes to have the timeframe?

We know the compressor speed from telegram 0x48D offset 17, also the the Power, but i think this is heating power, not electrical from 0ffset 11. In the log the gateway writes a few time 0x4A7 with two bytes, maybe a limit. And telegram 0x4AA seems to contain a lot of large 16bt values, the gateway sometimes reads some.

But we need a known, changing value to look for. EMS messages are not realtime, they come also delayed, we always have to check all telegrams within a minute to find a value.

RaHehl, 2024-04-29:
[1.6-1.5.txt](https://github.com/emsesp/EMS-ESP32/files/15147058/1.6-1.5.txt)

It looks like not every change in compressor output performance means a change in consumption. Here is a log that I stopped immediately after the display in the app changed from 1.6 to 1.5 kw

RaHehl, 2024-04-29:
[1.8-1.7.txt](https://github.com/emsesp/EMS-ESP32/files/15147460/1.8-1.7.txt)

This is a change from 1.8 to 1.7 kw

MichaelDvP, 2024-04-29:
Can you check if this works in custom entities:
[image]

(or if you like kW, use factor 0.001

RaHehl, 2024-04-29:
[image]
I can only select UINT i don't have the option UINT16.

The current heating run is over, I'll let you know if something happens, I've already played around with 04AA but always with offset, that hasn't worked yet, I'm excited to see what happens

MichaelDvP, 2024-04-29:
Update to 3.7.-dev or set to Type: USHORT

RaHehl, 2024-04-30:
Took a while (low energy house & heating already in summer mode)

But I can confirm it works perfectly:

HomeAssistant:
[image]

Energiemanager App:

[image]

The only noticeable thing that I think is correct: The amount of energy refers to the compressor and not to the heating element. I found that out when I tried to heat manually with "extra hot water" via the HomeCom app, which apparently doesn't start the compressor. But the heating element is not relevant to me, and in cases where you need it, the whole thing can be calculated quite easily.

Therefore the information is only about the name if it is not a custom element

proddy, 2024-04-30:
Do we need to add this custom entity as a real entity, and change the name then?

RaHehl, 2024-04-30:
I don't know the criteria for a "real entity", but I would say that a real-time consumption display is a very useful use case for the EMS ESP if he's connected to a heat pump.

Another topic: I don't have a use case for it myself because I have licensed the energy manager, but could you see in the log how the energy manager influences the heat pump's consumption?

RaHehl, 2024-05-01:
F.Y.I: https://github.com/emsesp/EMS-ESP32/discussions/1678#discussioncomment-9283496

MichaelDvP, 2024-05-01:
The only candidat i see is `gateway(0x48) -W-> boiler(0x08), ?(0x04A7), data: 09 B9 00 00`
a write from gateway to boiler, sometime `00 00 00 00` , sometimes the first16 bit set. 
09B9 = 2489W could be a limit.
Please check with:
[image]
(UINT16 = USHORT in v3.6.x)

RaHehl, 2024-05-01:
@MichaelDvP 2489W watts sounds absolutely correct, which corresponds approximately to the excess that I have observed in the last few days when the heat pump started, i will let u know tomorrow

RaHehl, 2024-05-02:
@MichaelDvP 
I can confirm it seems to be something like the Powerlimit:

[image]

After reaching an excess of around 2500W that was fed into the battery, the "power limit" which was previously at 0 was set to around 2450W, and then the compressor started up with a slight delay.

So the value, as soon as it is sent from the energy manager to the heat pump, roughly corresponds to my surplus

[image]

[image]

[image]

[image]

This morning the short heating of the hot water was still without energy manager involvement, but that is clearly the case now

Maybe this topic can be implemented with these findings:
https://github.com/emsesp/EMS-ESP32/discussions/1678

Now it would be exciting to see what happens if someone without an energy manager sets a limit while the hot water tank is cold, whether that is enough for the heat pump to start or whether additional values need to be set 

@Vippis2000

Vippis2000, 2024-05-02:
Ok I willing to give it a try. Please explain what I should do.

Currently I'm running 3.6.5 on a BBQKee S3.

Vippis2000, 2024-05-02:
What is the correct test setting. Shall I set the limit before my heatpump program is about to kick in?

Currently I'm programming the switchPrograms for DHW and HC according to PV forecast. Today looks like this:

[image]

RaHehl, 2024-05-02:
@Vippis2000 
https://github.com/emsesp/EMS-ESP32/issues/1717#issuecomment-2088755872

I would just write 2500 in there and see if the compressor started after about 30 seconds

Vippis2000, 2024-05-02:
So the expected behaviour would be:

If I set the powerlimit to 2500W the compressor should start working at 8:00 AM with a max. electrical power below 2500W? Or between 8AM and 6PM with a max. of 2500W accordingly

RaHehl, 2024-05-02:
I would just write it in manually now without any schedule to see if it basically works

If that is successful, you could create a function similar to that of the energy manager from 2500W sureplus, with a certain attenuation for short clouds, report this to the heat pump, the only thing missing is a logic for Tibber, but when I see the prices, that's it probably more for the winter :D

RaHehl, 2024-05-02:
[image]
It would also be worth mentioning that I now have 4500W surplus but the power limit remains at around 2600

Vippis2000, 2024-05-02:
I tried to Add custom entities, but they don't show any value:

[image]

Vippis2000, 2024-05-02:
> I would just write it in manually now without any schedule to see if it basically works
>
> If that is successful, you could create a function similar to that of the energy manager from 2500W sureplus, [...]

Yeah that was my original plan when power limiting works fine. The y-Axis for tibber is right :)

proddy, 2024-05-02:
> I tried to Add custom entities, but they don't show any value:
> 
> ![grafik](https://private-user-images.githubusercontent.com/128602940/327401798-aa786deb-ce55-4dad-842e-428e814c98d5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2NDcxMTcsIm5iZiI6MTcxNDY0NjgxNywicGF0aCI6Ii8xMjg2MDI5NDAvMzI3NDAxNzk4LWFhNzg2ZGViLWNlNTUtNGRhZC04NDJlLTQyOGU4MTRjOThkNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMlQxMDQ2NTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMzUxMzQyOTE5NDJhNTkxNjM2ZDU4YjdiYTgzOGUxZWFjZjk1NDVmYTg3ZGMwMTExMDk2YjUxNWNmNTg4ZWM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.LN6KoZ3_CD0d-Jy80sIsXTc61dVMOGzCBj0FiTHA4IE)

did you upgrade your 3.6.5 to 3.7.0-dev?

Vippis2000, 2024-05-02:
Yes just updatet to 3.7.0-dev.7 and still not showing values

RaHehl, 2024-05-02:
maybe because u don't have a energymeter sending them, which heat pump do you own?

Vippis2000, 2024-05-02:
CS7000iAW9OR-S+AWM9

Vippis2000, 2024-05-02:
> maybe because u don't have a energymeter sending them, which heat pump do you own?

Maybe there is another type. There must be because your heatpump responds to the command of the energy manager

Vippis2000, 2024-05-05:
Any update on this?

I can try other

RaHehl, 2024-05-05:
maybe some configuration is required if u add a HP to the energy manager he configures the HP, maybe a recording of this config would help?

Vippis2000, 2024-05-05:
Can you unmount and remount the HP to your energy manager? And record all changes on the EMS

Maybe the energy manager has done a firmware upgrade on the HP?

RaHehl, 2024-05-05:
[deactivate.txt](https://github.com/emsesp/EMS-ESP32/files/15214216/deactivate.txt)
[activate.txt](https://github.com/emsesp/EMS-ESP32/files/15214217/activate.txt)

RaHehl, 2024-05-06:
[PV-Start-2.txt](https://github.com/emsesp/EMS-ESP32/files/15216752/PV-Start-2.txt)

Vippis2000, 2024-05-16:
Any update on this?

myxor, 2024-05-19:
Just wanted to confirm that 
`0x08 0x4AA 0 USHORT`
is the current power consumption of my Buderus Logatherm WLW196i AR heat pump.

Vippis2000, 2024-05-20:
Updatet to 3.7.0 dev11 and still no values of custom entities

RaHehl, 2024-05-28:
also works perfectly in cooling mode

Vippis2000, 2024-06-19:
Any new developments in this area?

proddy, 2024-06-19:
> Any new developments in this area?

I'm not sure what needs to be done. Do the custom entities from ReHehla and myxor not work for you?

Vippis2000, 2024-06-19:
> > Any new developments in this area?
> 
> I'm not sure what needs to be done. Do the custom entities from ReHehla and myxor not work for you?

Sadly the do not. Seems that the "Energy manager" does some changes to the heatpump that creates those entities

proddy, 2024-06-25:
@Vippis2000 @RaHehl - I have a request. I'd love to get some test data for a Heat pump into EMS-ESP so we can simulate and test Heat pumps (I have an old gas boiler). Same with Solar and other devices. Since you two are both active users would you mind sending over a dump of your device entity values? You'll need to upgrade to 3.7.0-dev18, click on the export button (see below) and send over the TXT file which has the JSON. The easiest way is to zip it and attach it to this GitHub issue. If you don't want to upgrade just yet. or have no time, it's also cool. Cheers! CC @bbqkees  

[image]

RaHehl, 2024-06-25:
@proddy Yes of course, but can I send you the txt via a more private method, perhaps by email?

bbqkees, 2024-06-26:
> @Vippis2000 @RaHehl - I have a request. I'd love to get some test data for a Heat pump into EMS-ESP so we can simulate and test Heat pumps (I have an old gas boiler). Same with Solar and other [...]

If someone has a mixer module MM100/200/400, an SM100 or a RC310/RC800 thermostat we would like to get these datapoints as well.

Vippis2000, 2024-06-26:
Of course, currently I'm on vacation. Will send it on Saturday

Vippis2000, 2024-07-17:
[Kessel_WP_Enviline_Compress 6000AW_Hybrid 3000-7000iAW_SupraEco_Geo 5xx_WLW196i_WSW196i.txt](https://github.com/user-attachments/files/16267489/Kessel_WP_Enviline_Compress.6000AW_Hybrid.3000-7000iAW_SupraEco_Geo.5xx_WLW196i_WSW196i.txt)

proddy, 2024-07-17:
all good, thanks all

closed 2024-07-17 by proddy

UlfHRO, 2024-08-20:
> We know the compressor speed from telegram 0x48D offset 17, also the the Power, but i think this is heating power, not electrical from 0ffset 11. In the log the gateway writes a few time 0x4A7 with [...]

Hi,
I have tested 48D offset 17 at my WLW186 Buderus heat pump and there is a continious value when hp is running (heat, cool, dhw), but value is to low to be the heating power. It is only twice the power consumption and would not fit to the cop of around 3 to 4...?!

mheyse, 2024-08-29:
Hi, are you planning to add the ElPower entity to ems-esp? FWIW, Type ID 0x04AA (UINT16) as you suggested works very well with my WLW196i.

MichaelDvP, 2024-08-29:
Yes, as hpCurrPower (4AA, offs 0, UIN16) and hpPowerLimit (47A, offs 0, UINT16). (actually in my testbuild).

UlfHRO, 2024-08-30:
Hallo,
- I can NOT confirm hpPowerLimit (47A, offs 0, UINT16) on my WLW186i, this value is not available...
- I have available power from MyEnergyManager at: 0x48 | 0x4A7 | 4 | UINT16. That value limits the hp power for cooling (if "cooling only on pv" and triggers the dhw temp set increase...
- "cooling only on pv" can be set at 0x08 | 0x4AD | 1 | INT8

I also found the tags for my hc1 pump (PC1):
pc1_run | 0x08 | 0x2D6 | 0 | INT8
pc1_rate | 0x08 | 0x488 | 13 | INT8 (%)
pc1_flow | 0x08 | 0x2D6 | 9 | INT16 (l/h)

primary pump (PC0):
pc0_flow | 0x08 | 0x0E4 | 36 | INT16 (l/h)

expansion valve of compressor circuit (VR1):
valve_vr1 | 0x08 | 0x48D | 16 | INT8

hp compressor rpm:
comp_rpm | 0x08 | 0x48D | 17 | INT16

popopirat, 2024-09-03:
> Just wanted to confirm that `0x08 0x4AA 0 USHORT` is the current power consumption of my Buderus Logatherm WLW196i AR heat pump.

I can confirm this, however with a measurement device that tracks my home power consumption from the grid, the value is some 300-400 W off (the ems-esp Value is higher than what is drawn from the grid). Did anyone else notice this?

popopirat, 2024-09-03:
> Hallo,
> 
>     * I can NOT confirm hpPowerLimit (47A, offs 0, UINT16) on my WLW186i, this value is not available...
> 

Same here, the value is 65.535 W, which is a little bit too high for a heatpump. If I overwrite it with e. g. 1500 W, it has no impact on the running heatpump

sturmi, 2025-05-03:
> Can you check if this works in custom entities: [image]
> 
> (or if you like kW, use factor 0.001

Hi, for my  WLW186i I get 25W asElPower-value - nothing but 25W.
Is there any trick or other combination to get real value?

thanks
Thomas

sturmi, 2025-05-03:
> > Can you check if this works in custom entities: [image]
> > (or if you like kW, use factor 0.001
> 
> Hi, for my WLW186i I get 25W asElPower-value - nothing but 25W. Is there any trick or other combination to get real value?
> 
> thanks Thomas

Nothings changed - but now it works - maybe I was a little bit too impatitient