emsesp/EMS-ESP32 #1720: New Feature: Easy enable/disable PV/SG on Heat pumps

proddy, 2024-04-26:
In reference to https://github.com/emsesp/EMS-ESP32/discussions/1678#discussioncomment-9236392

MichaelDvP, 2024-05-02:
I think the input configuration is a one time task. Once configured the user may want to switch the input (invert), this is n´only a write 0/1 (off/on, false/true) to the config entity, no need for a long string.
If someone really want to configure often, mybe the PV byte, he can look into the list of https://github.com/emsesp/EMS-ESP32/discussions/1600#discussioncomment-8353382 and create a custom entity for PV1 0x486, offset 39, UINT8.

proddy, 2024-06-21:
Ok. This was an idea from @bbqkees so I'll wait until we have clear requirements.

proddy, 2024-09-15:
Still not sure what this would look like (UI wise) or whether it's practical. What does the community think? Let me know....

suaveolent, 2024-10-12:
By changing the config I assume one would write to EEPROM. I’m not entirely sure if this is something intended to be used. In the worst case writing too often could damage the EEPROM?

MichaelDvP, 2024-10-13:
> By changing the config I assume one would write to EEPROM. I’m not entirely sure if this is something intended to be used. In the worst case writing too often could damage the EEPROM?

I think this affects all permanent values, eg. daytemp, etc. Only very few parameters are not permanent, seltemp, selflowtemp.
I don't know what Bosch uses to store the values. If it is a eeprom we can assume 1 mio cycles. To not damage in the next 30 years you can write 90 vales a day or ~4 per hour in average. 
If they use flash with fixed adresses (i hope not), it's factor 100 less, better not change more than once a day in average. But this is worst case, for flash a wear leveling is very common and lifetime depends on free extra spac, that we also no know.