emsesp/EMS-ESP32 #1725: "manual defrost" on heat pump unexpected behaviour

masterx1981, 2024-04-28:
### PROBLEM DESCRIPTION
Hi! On a Buderus WPLS 8.2 (recognized by ems esp as a Enviline, Compress 6000AW/...), if i unset the "manual defrost" setting, it not disable the manual defrost setting that i have visible on the hmc310 panel, it unset the setting about the auxiliary heater (gas boiler used on hybrid systems).
Also, if from the hmc310 panel i change the "manual defrost" setting, i not see changes on the exposed sensors of the ems esp. I think about a bad mapping.

### TO REPRODUCE

Enable from the hp pump auxiliary heater (as hybrid or parallel), the manual defrost setting on ems esp goes to on. If you disable from ems esp the manual defrost setting, the heatpump disable the auxiliary heater configuration.

### EXPECTED BEHAVIOUR

The manual defrost setting from ems esp must change the relative setting on the hp, and eventually must be creates the Auxiliary heater setting on ems esp for allow the configuration of the auxiliary heater from ems esp (now missing)

proddy, 2024-04-30:
Probably bad mapping indeed. Do you see any errors or warnings in the log? Can you trace the telegrams when you change it manually? https://emsesp.github.io/docs/Troubleshooting/#changing-a-value-on-an-ems-device-doesnt-work

masterx1981, 2024-05-02:
Saturday i have some spare time and i check if i can identify the correct telegrams for both settings. Thanks!

masterx1981, 2024-08-03:
I'm sorry, had no time to check it, was waiting the holidays for sone spare time to check the correct messages...

closed 2025-03-23 by proddy

masterx1981, 2025-03-23:
Sorry for the late answer, today i've decided that was the "sniffing packets" day 😅
The manual defrost is triggered from the followind command (sent from the front panel id 10 to the pdc 08)
ems-esp-pdc:$ [K000+00:30:29.056 N 732: [telegram] Rx: 10 08 05 34 01 24

On the panel the setting never turn on, remain always, off, but when the byte is set to 1, the HP start a manual defrost cycle, so i can't see if there is a way to "force" the hp to interrupt the manual defrost cycle.

Can be useful?

Thanks!

Sbried, 2025-03-26:
Yes, telegram 0x0005 offset 52 starts the manual defrost. It´s already documented in the [ems library](https://docs.emsesp.org/ems-library/)

masterx1981, 2025-03-26:
> Yes, telegram 0x0005 offset 52 starts the manual defrost. It´s already documented in the [ems library](https://docs.emsesp.org/ems-library/)

Noticed now, but if i check the manual defrost state in the ems esp entites list, i see it enabled (not possible), so it can't be mapped correctly. And worst, if i try to disable the manual defrost flag, it change the logic of the aux heater (hybrid, parallel, etc)

Sbried, 2025-03-27:
As we are discussing in #2489 it´s maybe just a wrong mapping of 0x0491 offset 0.