emsesp/EMS-ESP32 #1732: Status …Loading…

RalfLuig, 2024-05-01:
As soon if I try to see the status of the EMS bus I am getting into an endless „loading“ loop. I am using the „dev 7“ version of EMS-ESP.

MichaelDvP, 2024-05-02:
This is from v3.6.5 in your browser cache. Press reload (ctrl-R) to refresh.

RalfLuig, 2024-05-02:
This problem seems to be only in Apple`s Safari Browser on a mobile device. If I use another browser like Chrome, I don't see the problem.

MichaelDvP, 2024-05-02:
Try with clear cache in safari (for ems-esp pages).

RalfLuig, 2024-05-02:
I already did that...

proddy, 2024-05-02:
I can't reproduce on my iPhone using safari. I'll keep testing

RalfLuig, 2024-05-02:
I found the problem. I opened the web interface via shortcut on the home screen. I removed the shortcut and created a new one.

closed 2024-05-02 by RalfLuig