emsesp/EMS-ESP32 #1734: No Temp displayed on HMC 310, HC2 Control Mode is not editable

chetkelley, 2024-05-03:
### PROBLEM DESCRIPTION

The HMC310 does not show a room temperature on the display. It's just two dashes, --ºC
Also, the hc2 control mode is "weather compensated" and try as I might to change it to "room" it doesn't do it. (I am assuming that "room" is the right choice when I have created a virtual thermostat.)

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{"System Info":{"version":"3.6.5","uptime":"000+00:40:28.889","uptime (seconds)":2428,"platform":"ESP32","arduino":"Tasmota Arduino v2.0.14","sdk":"4.4.6.240105","free mem":130,"max alloc":71,"used app":1591,"free app":969,"partition":"boot","reset reason":"Software reset CPU / Software reset CPU"},"Network Info":{"network":"WiFi","hostname":"ems-esp","RSSI":-53,"BSSID":"set","TxPower setting":0,"static ip config":false,"enable IPv6":true,"low bandwidth":false,"disable sleep":false,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected","AP security":"wpa2","AP ssid":"ems-esp"},"NTP Info":{"NTP status":"connected","enabled":true,"server":"time.google.com","tz label":"Europe/Berlin"},"OTA Info":{"enabled":true,"port":8266},"MQTT Info":{"MQTT status":"connected","MQTT publishes":1525,"MQTT queued":0,"MQTT publish fails":0,"MQTT connects":1,"enabled":true,"client id":"esp32-Buderus","keep alive":60,"clean session":false,"entity format":0,"base":"ems-esp","discovery prefix":"homeassistant","discovery type":0,"nested format":1,"ha enabled":true,"mqtt qos":0,"mqtt retain":false,"publish time heartbeat":60,"publish time boiler":10,"publish time thermostat":10,"publish time solar":10,"publish time mixer":10,"publish time other":10,"publish time sensor":10,"publish single":false,"publish2command":false,"send response":false},"Syslog Info":{"enabled":false},"Sensor Info":{"temperature sensors":0,"temperature sensor reads":0,"temperature sensor fails":0,"analog sensors":0,"analog sensor reads":0,"analog sensor fails":0},"API Info":{"API calls":0,"API fails":0},"Bus Info":{"bus status":"connected","bus protocol":"Buderus","bus telegrams received (rx)":12339,"bus reads (tx)":1915,"bus writes (tx)":0,"bus incomplete telegrams":1,"bus reads failed":4,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100},"Settings":{"board profile":"S32","locale":"en","tx mode":2,"ems bus id":11,"shower timer":true,"shower alert":false,"hide led":false,"notoken api":false,"readonly mode":false,"fahrenheit":false,"dallas parasite":false,"bool format":1,"bool dashboard":1,"enum format":1,"analog enabled":true,"telnet enabled":true,"max web log buffer":50,"web log buffer":29},"Devices":[{"type":"boiler","name":"Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i","device id":"0x08","product id":172,"version":"02.02","entities":145,"handlers received":"0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlers fetched":"0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC","handlers pending":"0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x04AE 0x04AF","handlers ignored":"0x049F 0x04A0 0x04A3 0x0240 0xF7 0xE7 0x02E0 0x02EA 0x061E 0xD2 0xE2"},{"type":"thermostat","name":"RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410","device id":"0x10","product id":158,"version":"73.03","entities":101,"handlers received":"0x06 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02CC 0x0467 0x0291 0x02CE 0x0468 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E","handlers pending":"0xA3 0xA2 0x12 0x13 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E","handlers ignored":"0x0508 0x0509 0x050A 0x0507 0xBF 0x0239 0x042B"},{"type":"thermostat","name":"RC100H","device id":"0x39","product id":200,"version":"40.04","entities":0,"handlers pending":"0x042C 0x047C 0x0274 0x0A6B","handlers ignored":"0x042B"},{"type":"mixer","name":"MM100","device id":"0x21","product id":160,"version":"24.05","entities":7,"handlers received":"0x02D8","handlers fetched":"0x02CE","handlers ignored":"0x02E2 0x02EC 0xE6 0x0255 0x0292 0xBF"},{"type":"gateway","name":"MX300","device id":"0x48","product id":252,"version":"07.02","entities":0,"handlers ignored":"0x125A 0x2040 0xF9"}]}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

ems-esp should accept the change of control mode.
remote room temperature should be displayed in the hmc310.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2024-05-03:
Maybe the same as in #1691

chetkelley, 2024-05-03:
I've tried setting the remote thermostat to RC200 but the system won't accept it. It stays on RC310. On the HMC310 display I can't get to the configuration menu, go to the remote thermostat line for hc2, and there I have a choice of RC100 or RC100H. The RC100H is the selected/active remote for hc2.

MichaelDvP, 2024-05-03:
> I've tried setting the remote thermostat to RC200 but the system won't accept it.

Yes, seems the same as #1691. Update to 3.7.0dev8, set remotetemp, a RC200 should appear. No need to set control first.

chetkelley, 2024-05-03:
Progress, 
I first removed the existing remote thermostat (call thermostat hc2.remotetemp -1) in order to start fresh with your directions.
Then I set the remote temp, without choosing a controller, and the system accepted it. After a minute or two the ems-esp devices list showed a RC100H with an entity! (room temperature from remote) This was more than I had before.

chetkelley, 2024-05-03:
The hmc310 still does not show a room temperature from the virtual remote.

chetkelley, 2024-05-03:
HA! Success. I went into the service menu of the HMC310, then to Commissioning, the scrolled down to the "remote control for heating circuit 2" and chose RC100H (to match what the ems-esp shows in the device list. My only options on the boiler was none, rc100, or rc100h) When I did the room temperature show up on the display! Everything seems to be working as it should. 

Thank you very much, Michael, for your help and guidance.

closed 2024-05-03 by chetkelley