emsesp/EMS-ESP32 #1735: Improvement: Add holiday mode to bosch compress 3800i ew

mattfro, 2024-05-03:
### PROBLEM DESCRIPTION

Can't find holiday "mode"

### REQUESTED INFORMATION

{
  "System Info": {
    "version": "3.7.0-dev.8",
    "uptime": "000+01:07:15.548",
    "uptime (seconds)": 4035,
    "platform": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.14",
    "sdk": "v4.4.6-dirty",
    "free mem": 193,
    "max alloc": 183,
    "free caps": 8330,
    "used app": 2165,
    "free app": 5963,
    "partition": "app1",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -54,
    "TxPower setting": 0,
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Stockholm"
  },
  "OTA Info": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 1307,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "esp32-463bdaec",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
[... 141 lines cut ...]
000+00:51:12.569 T 1418: [emsesp] gateway(0x48) -R-> boiler(0x08), UBAMonitorFastPlus(0xE4), length: 1 (offset 40)
000+00:51:12.579 T 1419: [emsesp] boiler(0x08) -W-> gateway(0x48), UBAMonitorFastPlus(0xE4), data: <empty> (offset 40)
000+00:51:12.616 T 1420: [emsesp] gateway(0x48) -W-> thermostat(0x10), ?(0x0269), data: 09 01 01 09 01 01
000+00:51:12.706 T 1421: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPSet(0x0467), length: 25
000+00:51:12.724 T 1422: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPSet(0x0467), data: 12 18 00 01 02
000+00:51:12.908 T 1423: [emsesp] boiler(0x08) -B-> All(0x00), MenuConfig(0xF7), data: D1 00 00 07
000+00:51:13.168 T 1424: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPMode(0x0291), length: 25
000+00:51:13.186 T 1425: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPMode(0x0291), data: 01 00 00 00 00 00
000+00:51:13.214 T 1426: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0x0269), data: 09 01 01 09 01 01 02 11
000+00:51:13.473 T 1427: [emsesp] controller(0x09) -W-> thermostat(0x10), ?(0x0269), data: 12 01 01 12 01 01 00 11
000+00:51:13.569 T 1428: [emsesp] gateway(0x48) -R-> boiler(0x08), ?(0x04), length: 2 (offset 25)
000+00:51:13.579 T 1429: [emsesp] boiler(0x08) -W-> gateway(0x48), ?(0x04), data: <empty> (offset 25)
000+00:51:13.643 T 1430: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPSet(0x0468), length: 25
000+00:51:13.660 T 1431: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPSet(0x0468), data: 12 18 00 01 02
000+00:51:13.931 T 1432: [emsesp] Me(0x0B) -R-> thermostat(0x10), HPMode(0x0292), length: 25
000+00:51:13.950 T 1433: [emsesp] thermostat(0x10) -W-> Me(0x0B), HPMode(0x0292), data: 00 00 00 00 00 00
000+00:51:13.984 T 1434: [emsesp] controller(0x09) -R-> thermostat(0x10), ?(0x0269), length: 8
000+00:51:14.001 T 1435: [emsesp] thermostat(0x10) -W-> controller(0x09), ?(0x0269), data: 12 01 01 12 01 01 02 11
000+00:51:14.051 T 1436: [emsesp] thermostat(0x10) -B-> All(0x00), ?(0x0269), data: 12 01 01 12 01 01 02 11
000+00:51:14.443 T 1437: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300WWmode(0x02F5), length: 25
000+00:51:14.480 T 1438: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300WWmode(0x02F5), data: 01 00 05 00 00 FF 08 06 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00
000+00:51:14.693 T 1439: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300OutdoorTemp(0x023A), length: 25
000+00:51:14.707 T 1440: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300OutdoorTemp(0x023A), data: 00 AC
000+00:51:14.981 T 1441: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Settings(0x0240), length: 25
000+00:51:14.999 T 1442: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: 00 01 EC 00 FF (offset 8)

MichaelDvP, 2024-05-03:
See #275, the next bytes are heating mode,temperature,  dhw mode, and assign to hc/dhw circuits and this for 5 holiday entries.
All in all its 50-75 entities, depending what info we combine. Too much for RAM and mqtt-heap usage.
RC300 switchprograms and holiday settings are often discussed, but in the actual data structure is not easy to implement. 
And this are settings very rarly changed.

mattfro, 2024-05-03:
Ok I see...
But couldn't it be that if you set that date then rest will be set automatically?
Because, only parameter I can set in the pump for holiday is, a start date end date.
And if I want to turn it off, it's just reset.

Seems in holiday mode, it's set to fixed 17C.

But according to that ticket it seems not to be the case... Oh well, was hoping it would be a easy thing :)
Anyways, I could do my own "holiday/vacation" setup instead in hass.  Just lower the temperature and some other small tunings like warm water etc.

MichaelDvP, 2024-05-03:
Yes, seems to be a shortend telegram. 
`thermostat(0x10) -B-> All(0x00), ?(0x0269), data: 18 05 03 18 05 05 02 11`
means 03.05.2024-05.05.2024, fixed temperature mode (02), 17°C 
Can you check in terminal `read 8 269`, `read 8 26A`, ... `read 8 26D` to check the length and if there are also 5 dates.
It seems the the info to heatingcircuits and dhw are missing in this telegram,  Do this setting affect dhw or is there a additional setting?

mattfro, 2024-05-03:
@MichaelDvP read 8? or read 10 269
If I do read 8 269, 26A and 26D it's always empty
ems-esp:# read 8 269
000+03:23:26.362 N 81: [emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x0269), data: <empty>
ems-esp:# read 8 26A
000+03:23:34.434 N 82: [emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x026A), data: <empty>
ems-esp:# read 8 26D
000+03:23:42.745 N 83: [emsesp] boiler(0x08) -W-> Me(0x0B), ?(0x026D), data: <empty>

if I check read 10 before I set vacation:
ems-esp:#  read 10 269
000+03:34:16.485 N 89: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0269), data: 12 01 01 12 01 01 02 11
ems-esp:#  read 10 26A
000+03:35:34.816 N 90: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x026A), data: <empty>
ems-esp:#  read 10 26D
000+03:35:45.216 N 91: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x026D), data: <empty>

and this is when I set vacation:
ems-esp:#  read 10 269
000+03:36:15.699 N 92: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x0269), data: 18 05 03 18 05 05 02 11
ems-esp:#  read 10 26A
000+03:36:19.829 N 93: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x026A), data: <empty>
ems-esp:#  read 10 26D
000+03:36:22.929 N 94: [emsesp] thermostat(0x10) -W-> Me(0x0B), ?(0x026D), data: <empty>

mattfro, 2024-05-03:
btw. when I took the logs for setting vacation in the first place, I used my phone app, so you can see it comes from the gateway 0x48.

Tried again using the app and set date:
000+03:57:56.335 N 309: [emsesp] gateway(0x48) -W-> thermostat(0x10), ?(0x0269), data: 18 05 03 18 05 05

and when doing reset or removing the date:
000+03:58:22.886 N 312: [emsesp] gateway(0x48) -W-> thermostat(0x10), ?(0x0269), data: 09 01 01 09 01 01

not sure if that helps... I wonder if you could test to use that as a custom entity to try to see if it would work. Set only those?

MichaelDvP, 2024-05-03:
> read 8? or read 10 269

Youre right, read from thermostat 0x10. 
Funny communication, the gateway sets 01.01.2009 as off, and the controller (0x09) corrects to 01.01.2018. But i think we can write any date in the past.
I'll add a single entity for the rego thermostat, only dates as for RC35/RC30.

MichaelDvP, 2024-05-04:
Have you already tested if you can change the holiday temperature with custom entity:
device 0x10, type 0x0269, offset 7,factor 1

mattfro, 2024-05-04:
@MichaelDvP Oh... I think we got a misunderstanding here :)
I thought you where going to add it in the next build...
So... I did not test that,.

I have a few question in this case
[image]
should I use int8?

And how should the value be sent, what format?
[image]

MichaelDvP, 2024-05-04:
>I thought you where going to add it in the next build...

Yes,the dates, for temperature i'm not sure because you wrote  
> Seems in holiday mode, it's set to fixed 17C.

this value range int8 and uint8 is the same, does not matter. It takes a minute until the value show up. 
Just enter a temperature different from 17 maybe 15 or 18 degrees and check if it changes, You can log the telegrams while changing to see the protocol. Maybe the controller overwrite like the off-dates.

mattfro, 2024-05-04:
I meant in what format should I send the value in?
Raw or as a date?
18 05 04 18 05 05 or 240504240505
and why does it have a value 17?
[image]

Is it correct offset?

mattfro, 2024-05-04:
oh are you saying latest build has already holiday in it?
I have latest.. and I don't have any holiday or vacation settings. Only date settings is the "normal" one.

MichaelDvP, 2024-05-04:
I'm working on it, it's not yet implemented. If you like to use custom entitiy its:
`holidaydate`: device 0x10, type 0x269, offset 0, type RAW, length 6 with data `18 05 04 18 05 05`
`holidaytemp` device 0x10, type 0x269, offset 7, type UINT8, UOM °C with data `17`

mattfro, 2024-05-04:
Did not try holiday temp... 
But otherwise it worked fine, checked both the pump UI and phone app, it was on holiday mode when setting that value.
Also I did reset it with these values, as the app used to send: 09 01 01 09 01 01

So for my point of view it works fine what I could see...attached logs for this also.
[holiday.log](https://github.com/emsesp/EMS-ESP32/files/15209406/holiday.log)

activated holiday:
ems-esp:# 000+20:31:07.685 I 133: [command] Calling command 'custom/holiday' (set custom value on ems) with value 18 05 04 18 05 05

reset holiday:
ems-esp:# 000+20:33:11.683 I 761: [command] Calling command 'custom/holiday' (set custom value on ems) with value 09 01 01 09 01 01
later on it changes to this:
ems-esp:# 000+20:33:13.840 T 785: [emsesp] controller(0x09) -W-> thermostat(0x10), ?(0x0269), data: 12 01 01 12 01 01 00 11

mattfro, 2024-05-04:
And also, I don't see any point of changing the holiday temp to anything else than 17. According to the manual, you should not set it under 17C anyways. The heatpump would not work under those temperatures...at least in a way it would save any energy.

MichaelDvP, 2024-05-04:
Please check dev9: https://github.com/MichaelDvP/EMS-ESP32/releases

mattfro, 2024-05-04:
@MichaelDvP tried it already:
[image]
maybe fix the name :)
and it's setting the year wrong:
000+00:15:07.115 T 76: [emsesp] thermostat(0x10) -W-> gateway(0x48), ?(0x026A), data: <empty> (offset 4)
**000+00:15:07.125 I 77: [command] Calling command 'thermostat/holiday' (Urlaub) with value 04.05.2014-05.05.2014**
000+00:15:07.213 T 78: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300Holiday1(0x0269), data: 04 05 0E 05 05 0E
000+00:15:07.393 T 79: [emsesp] gateway(0x48) -R-> thermostat(0x10), ?(0x026B), length: 1
000+00:15:07.403 T 80: [emsesp] thermostat(0x10) -W-> gateway(0x48), ?(0x026B), data: <empty>

year first, then month and day. I think

MichaelDvP, 2024-05-04:
Oh, sorry, yes, Bosch changes the order for RC30/RC35 and RC300/Rego, Updated my dev.

mattfro, 2024-05-04:
@MichaelDvP 
works fine now:
000+00:04:25.565 I 27: [command] Calling command 'thermostat/vacations' (vacation dates) with value 04.05.2024-05.05.2024
000+00:04:25.604 T 28: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300Holiday1(0x0269), data: 18 05 04 18 05 05

and setting date to yesterday to reset it:
000+00:05:35.523 I 76: [command] Calling command 'thermostat/vacations' (vacation dates) with value 02.05.2024-02.05.2024
000+00:05:35.859 T 88: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300Holiday1(0x0269), data: 18 05 02 18 05 02
000+00:05:36.067 T 89: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Holiday1(0x0269), length: 25
000+00:05:36.107 T 90: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Holiday1(0x0269), data: 18 05 02 18 05 02 02 11
000+00:05:37.294 T 100: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Holiday1(0x0269), data: 18 05 02 18 05 02 02 11
000+00:05:37.603 T 101: [emsesp] controller(0x09) -W-> thermostat(0x10), RC300Holiday1(0x0269), data: 12 01 01 12 01 01 00 11
000+00:05:37.857 T 106: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Holiday1(0x0269), data: 12 01 01 12 01 01 02 11

mattfro, 2024-05-04:
Seems it was much easier on mine that the other ticket you had. :)

mattfro, 2024-05-05:
@MichaelDvP now when it works, shall we close this ticket?

proddy, 2024-05-05:
yes, if you've confirmed it's working

closed 2024-05-05 by mattfro