emsesp/EMS-ESP32 #1738: Remove OTA feature

proddy, 2024-05-04:
I'm planning on removing the OTA feature. You can upload firmware versions over-the-air via the WebUI upload feature or via a Python script in platformIO or standalone via the CLI.

This will also free up some memory in EMS-ESP for other features.

@MichaelDvP agree with this? I'll make a PR

MichaelDvP, 2024-05-04:
OK,
 i've used it very few times when i broke the webinterface and don't want to walk stairs, but i think normal users never need it. 
Also remove the OTA description from docs.

proddy, 2024-05-04:
Ok, I'll do the change now. I've added a script so you can still upload a .bin file from a Linux or DOS prompt

proddy, 2024-05-04:
with OTA removed there is 8kb flash free'd up and I'm my ESP32-S3 about 2kb more heap and alloc buffer space.

To upload from platformIO add this to your `pio_local.ini`

```
[env]
; upload settings
upload_protocol = custom
custom_emsesp_ip = 10.10.10.173
custom_username = admin
custom_password = admin
upload_port = /dev/ttyUSB*
```

and include the `upload.py` script like

```
extra_scripts =
  pre:scripts/build_interface.py ; comment out if you don't want to re-build the WebUI each time
  scripts/rename_fw.py
  scripts/upload.py
```

To upload from the command line use 

```sh
python3 scripts/upload_cli.py -i 10.10.10.173 -f ./build/firmware/EMS-ESP-3_7_0-dev_8-ESP32_4M.bin
```

closed 2024-05-06 by proddy