emsesp/EMS-ESP32 #1745: Automation Homeassistant - boiler disinfection

zibous, 2024-05-06:
I would like to do the boiler disinfection only once a month (first Saturday) and I am not sure if I have to set both switches (boiler, thermostat) ?

  - switch.boiler_dhw_disinfecting
  - switch.thermostat_dhw_disinfecting 

``` 
Home Assistant Core 2024.5.1 Frontend 20240501.0
EMS-ESP Version 3.6.5
GBx72/Trendline/Cerapur/Greenstar Si
Thermostat | RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410
```

  
```yaml
template:
 - sensor:
     - unique_id: "sensor_is_first_saturday"
       name: "Erster Samstag im Monat"
       state: "{% if now().day <= 7 and as_timestamp(now()) | timestamp_custom('%A') == 'Saturday' %}on{% else %}off{% endif %}"
    
    
automation:    
    - id: boiler_disinfecting_control
      alias: "Boiler Desinfektion"
      description: "Automation with the help of which the boiler disinfection is only activated on the first Saturday of a month, in the other days it is not activated."
      initial_state: true
      mode: single
      trigger:
        - platform: state
          entity_id: binary_sensor.is_first_saturday
      condition: []
      action:
        - choose:
            # ------------------------------
            # enable the boiler disinfecting
            # ------------------------------
            - conditions:
                # check if we are on the first saturday
                - condition: state
                  entity_id: binary_sensor.is_first_saturday
                  state: "on"
                # check switch.boiler_dhw_disinfecting (Boiler)
                - condition: state
                  entity_id: switch.boiler_dhw_disinfecting
                  state: "off"
              sequence:
                - service: switch.turn_on
                  entity_id: switch.boiler_dhw_disinfecting
               
        - choose:
            # -----------------------------------
            # disable the boiler disinfecting
            # -----------------------------------
            - conditions:
                - condition: state
                  entity_id: binary_sensor.is_first_saturday
                  state: "off"
                - condition: state
                  entity_id: switch.boiler_dhw_disinfecting
                  state: "on"
              sequence:
                - service: switch.turn_off
                  entity_id: switch.boiler_dhw_disinfecting
          default:[]
            
```

Thanks

MichaelDvP, 2024-05-07:
>  I am not sure if I have to set both switches (boiler, thermostat) ?

Try it. I think you need to set the thermostat switch. Boiler is controlled by the thermostat and if you set the boiler switch, the thermostat will set it back. Setting the thermostat switch advices the thermostat to set the boiler switch.

zibous, 2024-05-07:
>  I think you need to set the thermostat switch

Thank you very much, yes this is the solution, I will try it.

```
- service: switch.turn_on
   entity_id: switch.thermostat_dhw_disinfecting
```