emsesp/EMS-ESP32 #1747: Platform mqtt does not generate unique IDs. For  ID boiler_hysteresis_on_temperature and ID boiler_hysteresis_off_temperature

bakboord4, 2024-05-07:
### PROBLEM DESCRIPTION

REPORTING TWO ERROR MESSAGES.

2024-05-07 14:52:29.564 ERROR (MainThread) [homeassistant.components.number] Platform mqtt does not generate unique IDs. ID boiler_hysteresis_on_temperature is already used by number.boiler_hysteresis_on_temperature - ignoring number.boiler_hysteresis_on_temperature_2
2024-05-07 14:52:29.600 ERROR (MainThread) [homeassistant.components.number] Platform mqtt does not generate unique IDs. ID boiler_hysteresis_off_temperature is already used by number.boiler_hysteresis_off_temperature - ignoring number.boiler_hysteresis_off_temperature_2

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{"System Info":{"version":"3.7.0-dev.10","uptime":"000+00:04:52.117","uptime (seconds)":292,"platform":"ESP32","arduino":"Tasmota Arduino v2.0.15","sdk":"4.4.7.240312","free mem":175,"max alloc":107,"free caps":125,"used app":1609,"free app":375,"partition":"app1","reset reason":"Software reset CPU / Software reset CPU"},"Network Info":{"network":"WiFi","hostname":"ems-esp","RSSI":-54,"BSSID":"set","TxPower setting":0,"static ip config":true,"low bandwidth":false,"disable sleep":false,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected","AP security":"wpa2","AP ssid":"ems-esp"},"NTP Info":{"NTP status":"disconnected","enabled":false,"server":"ntp.time.nl","tz label":"Europe/Amsterdam"},"MQTT Info":{"MQTT status":"connected","MQTT publishes":166,"MQTT queued":0,"MQTT publish fails":0,"MQTT connects":1,"enabled":true,"client id":"ems-esp","keep alive":60,"clean session":true,"entity format":0,"base":"NefitCV","discovery prefix":"homeassistant","discovery type":0,"nested format":1,"ha enabled":true,"mqtt qos":0,"mqtt retain":false,"publish time heartbeat":60,"publish time boiler":10,"publish time thermostat":10,"publish time solar":10,"publish time mixer":10,"publish time water":10,"publish time other":10,"publish time sensor":10,"publish single":false,"publish2command":false,"send response":false},"Syslog Info":{"enabled":false},"Sensor Info":{"temperature sensors":0,"temperature sensor reads":0,"temperature sensor fails":0,"analog sensors":0,"analog sensor reads":0,"analog sensor fails":0},"API Info":{"API calls":1,"API fails":1},"Bus Info":{"bus status":"connected","bus protocol":"Buderus","bus telegrams received (rx)":288,"bus reads (tx)":68,"bus writes (tx)":0,"bus incomplete telegrams":0,"bus reads failed":0,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100},"Settings":{"board profile":"S32","locale":"nl","tx mode":1,"ems bus id":11,"shower timer":true,"shower alert":false,"hide led":false,"notoken api":false,"readonly mode":false,"fahrenheit":false,"dallas parasite":false,"bool format":1,"bool dashboard":1,"enum format":1,"analog enabled":true,"telnet enabled":true,"max web log buffer":50,"web log buffer":17},"Devices":[{"type":"boiler","name":"GBx72/Trendline/Cerapur/Greenstar Si","device id":"0x08","product id":123,"version":"07.00","entities":85,"handlers received":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlers fetched":"0x14 0x16 0x33 0x26 0x28","handlers pending":"0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B","handlers ignored":"0x23"},{"type":"thermostat","name":"EasyControl/CT200","device id":"0x18","product id":203,"version":"05.00","entities":3,"handlers received":"0x06","handlers fetched":"0x0A","handlers pending":"0xA3 0xA2 0x12 0x13","handlers ignored":"0x33 0x18"},{"type":"controller","name":"BC10","device id":"0x09","product id":190,"version":"01.04","entities":0}]}

```

### TO REPRODUCE

restart HA

### EXPECTED BEHAVIOUR

generate unique ID's

### ADDITIONAL CONTEXT

Looks like that boilyston and wwhyston have the same uniq_id's. The same for  boilhystoff and wwhystoff,
But I am not system develloper.

**boilhyston**
{"uniq_id":"boiler_hysteresis_on_temperature","obj_id":"boiler_hysteresis_on_temperature","cmd_t":"NefitCV/boiler/boilhyston","mode":"box","step":1,"min":-20,"max":0,"ic":"mdi:coolant-temperature","name":"Ketel aan hysterese verschil","stat_t":"NefitCV/boiler_data","val_tpl":"{{value_json['boilhyston'] if value_json['boilhyston'] is defined else -20}}","avty":[{"t":"NefitCV/boiler_data","val_tpl":"{{'online' if value_json['boilhyston'] is defined else 'offline'}}"}],"avty_mode":"all","dev":{"ids":["NefitCV-boiler"],"name":"NefitCV Boiler"}}

**wwhyston**
{"uniq_id":"boiler_hysteresis_on_temperature","obj_id":"boiler_hysteresis_on_temperature","cmd_t":"NefitCV/boiler/dhw/hyston","mode":"box","step":1,"min":-126,"max":126,"ic":"mdi:coolant-temperature","name":"dhw Inschakeltemperatuurverschil","stat_t":"NefitCV/boiler_data","val_tpl":"{{value_json.dhw['hyston'] if value_json.dhw is defined and value_json.dhw['hyston'] is defined else -126}}","avty":[{"t":"NefitCV/boiler_data","val_tpl":"{{'online' if value_json.dhw is defined and value_json.dhw['hyston'] is defined else 'offline'}}"}],"avty_mode":"all","dev":{"ids":["NefitCV-boiler"],"name":"NefitCV Boiler"}}

**boilhystoff**
{"uniq_id":"boiler_hysteresis_off_temperature","obj_id":"boiler_hysteresis_off_temperature","cmd_t":"NefitCV/boiler/boilhystoff","mode":"box","step":1,"min":0,"max":20,"ic":"mdi:coolant-temperature","name":"Ketel uit hysterese verschil","stat_t":"NefitCV/boiler_data","val_tpl":"{{value_json['boilhystoff'] if value_json['boilhystoff'] is defined else 0}}","avty":[{"t":"NefitCV/boiler_data","val_tpl":"{{'online' if value_json['boilhystoff'] is defined else 'offline'}}"}],"avty_mode":"all","dev":{"ids":["NefitCV-boiler"],"name":"NefitCV Boiler"}}

**wwhystoff**
{"uniq_id":"boiler_hysteresis_off_temperature","obj_id":"boiler_hysteresis_off_temperature","cmd_t":"NefitCV/boiler/dhw/hystoff","mode":"box","step":1,"min":-126,"max":126,"ic":"mdi:coolant-temperature","name":"dhw Uitschakeltemperatuurverschil","stat_t":"NefitCV/boiler_data","val_tpl":"{{value_json.dhw['hystoff'] if value_json.dhw is defined and value_json.dhw['hystoff'] is defined else -126}}","avty":[{"t":"NefitCV/boiler_data","val_tpl":"{{'online' if value_json.dhw is defined and value_json.dhw['hystoff'] is defined else 'offline'}}"}],"avty_mode":"all","dev":{"ids":["NefitCV-boiler"],"name":"NefitCV Boiler"}}

proddy, 2024-05-07:
Michael fixed it, in latest dev

bakboord4, 2024-05-07:
succesfull corrected.

commit 660bc633 (2024-05-07): Merge pull request #1748 from MichaelDvP/dev

dhw tag in HA format 3.4, fix #1747
commit b51b751e (2024-05-07): dhw tag in HA format 3.4, fix #1747
closed 2024-05-07 by bakboord4