emsesp/EMS-ESP32 #1764: WebUI data missing on Firefox

proddy, 2024-05-15:
It is not related to the vacation enhancements. Several values from other devices are missing as well, and it seems to affect FireFox only.

I can reproduce this as follows.

* Open a tab and type "http://ems-gw1.squirrel.nl"
* The tab opens, and is redirected to "http://ems-gw1.squirrel.nl/devices"
* Choose boiler, several values are missing
* Choose thermostat, same. Always the same set of values.
* [REFRESH] no change
* Ctrl-R goes back to "http://ems-gw1.squirrel.nl/devices".
* Choose boiler, all values are present.
* Choose thermostat, same.

Interesting is that when I open the tab and type "http://ems-gw1.squirrel.nl/devices", all values are there.

_Originally posted by @sciurius in https://github.com/emsesp/EMS-ESP32/issues/1712#issuecomment-2112296352_

proddy, 2024-05-15:
Possible a caching issue, I'll try and reproduce and fix.

sciurius, 2024-05-15:
I suspect one of my firefox extensions is getting in the way. Please stall this issue until I know more.

MichaelDvP, 2024-05-15:
Try to delete cache and website settings for this page in firefox. I've see some strange effects after updates. 
I suspect the last-modified is updated by frefox and from here https://github.com/emsesp/EMS-ESP32/blob/1487f30c433e6fc87ecbba42c04f9f9ea483dd4f/lib/framework/ESP8266React.cpp#L32  it always uses cached page.

sciurius, 2024-05-15:
That did, indeed, fix it...

proddy, 2024-05-15:
the 'last-modified' is only used for the static pages, which is ok. Things like the images and library functions.

The API data shouldn't be cached and report back a 304 so I think there is something else strange I need to look at. I first need to try and reproduce. Perhaps its here:
https://github.com/emsesp/EMS-ESP32/blob/ab9241376be7097a2239c3dcc3f7f716224c2248/interface/src/api/endpoints.ts#L12

tp1de, 2024-05-16:
I am using Firefox as well and I can't reproduce. Everything is working fine.

proddy, 2024-05-16:
> I am using Firefox as well and I can't reproduce. Everything is working fine.

thanks Thomas

proddy, 2024-05-18:
will close for now and re-open if this is a common occurrence with other users.

closed 2024-05-18 by proddy