emsesp/EMS-ESP32 #1765: boiler.nrgtotal with spikes and wrong value after update

tp1de, 2024-05-16:
nrgtotal has spikes within timeline. 
on system 1 (ems 2.0) it looks like this:

[image]

On May 10th I did a firmware upgrade from 3.6.5 to 3.7.0-dev.10. The nrgtotal value was wrongly calculated. Most probably the ww part was missing.

on system 2 (ems+) ist looks like this:

[image]

Firmware upgrade on May 3rd and spikes are positive not negative - looks strange.

As a result the home assistant energy statistics  are complete wrong and the statistic records needs to be corrected manually.

proddy, 2024-05-16:
don't know what the problem is. any ideas?

tp1de, 2024-06-05:
I can see spikes within HA (mqtt) for boiler.curburnpow (modulation) on same time when nrgheat / nrgtotal spikes happen:
[image]

Just one datapoint. On API (ioBroker) with polling every 15 seconds I do not see these spikes,
Within HA states (database) I see one entry with undefined value. 
This results in statistics with a high positive value and direct afterwars with a corresponding negative value.

Something strange. There should be no mqtt messages without any value. Please check.

proddy, 2024-06-05:
I'm see spikes with `curburnpow` as well, since May 26th

[image]

I'll look into it

tp1de, 2024-06-05:
Within ha I see states without any value. Only one state within database. Before and afterwards correct values. I assume that values are populated by mqtt.

proddy, 2024-06-05:
I'm not seeing states without values on my system

tp1de, 2024-06-06:
The states without value can only be seen with a database tool. I use MariaDB for HA and with MySQL Workbench I can analyse the states. E.g. for boiler_nrgheat there are a lot of states with current state value "unavailable":

[image]

For boiler_nrgww there are no "unavailable" states records at all. Could it be related to mqtt settings?:
[image]

tp1de, 2024-06-06:
But I do not know if this "unavailable" states are the root cause for the spikes in statistics?

closed 2024-08-03 by proddy