emsesp/EMS-ESP32 #1767: Worcester-Bosch Greenstar Summer Mode temp value upper limit too low

V4ler1an, 2024-05-16:
### PROBLEM DESCRIPTION

The 'Summer Mode' temperature setting in the Worcester Bosch Greenstar Weather Compensation settings currently has an upper limit of only 30 degrees C. When the Weather Compensation feature is enabled on the boiler and the 'Boiler Outside Temperature' thermostat value meets or exceeds this 'Summer Mode' temperature value, the boiler will disable any response to central heating demand. Effectively disabling the central heating - as intended.

The problem (at least for the way my boiler is installed), is that activation of Summer Mode will also prevent the boiler from heating my hot water tank. This appears to be because the boiler has not been connected up to use its proper designed DHW features. These are all unused in the boiler settings. Intead if has some separate setup which relies on and uses the central heating features and (presumably) some sort of separate hot water tank valve which opens when the Hive Home smart controls demand hot water.

Long story short, the upper summer mode temp limit will mean that for certain periods in the summer, if the outside temp reaches or exceeds 30 degrees C, I won't be able to heat any hot water. The workaround required would be to disable the weather compensation curve feature entirely, which disables Summer Mode.

Could I please have the 'boiler summertime temperature' value limit increased to 45 degrees C?

### REQUESTED INFORMATION

### TO REPRODUCE

Attempt to increase the 'boiler summertime temperature' value above 30 degrees C.

### EXPECTED BEHAVIOUR

Current behaviour is expected but undesirabe.

### SCREENSHOTS

Home Assistant automation failure due to Summertime Mode temperatire value limit exceeded:
[image]

### ADDITIONAL CONTEXT
I appreciate that had the central heating installers installed my boiler to use its built-in water heating (DHW) features to the full, hot water would be heated separately from central heating water. Tbh, I don't know the reasons for installing it this way. Nevertheless, as I see it, increasing the Summer Mode value limit shouldn't really cause any problems, irrespective of how the boiler is installed in respect to hot water heating.

Note: this Summer Mode feature was introduced to the EMS-Bus Gateway firmware several months ago when I requested that you add support for the official Worcester Bosch weather compensation add-on in this issue I raised: 

https://github.com/emsesp/EMS-ESP32/issues/1642

proddy, 2024-05-17:
that's like a 2 second change. I'll do it ASAP

proddy, 2024-05-18:
in dev11. Max is 45

V4ler1an, 2024-05-18:
Brilliant, thank you.

V4ler1an, 2024-05-18:
Ah... a problem. It seems the boiler's weather comp settings also limits the value to 30. My automation no longer errors but the boiler puts it back to 30. Is there any way around that do you think?

proddy, 2024-05-18:
hmm, not sure. it's it similar to https://emsesp.github.io/docs/Troubleshooting/#changing-a-value-works-at-first-but-is-then-reset-to-its-original-value ?

V4ler1an, 2024-05-18:
Side note: since applying the beta firmware, i've been seeing this. Just keeps spinning:
[image]

proddy, 2024-06-04:
> Side note: since applying the beta firmware, i've been seeing this. Just keeps spinning: [...]

refresh browser cache help?

V4ler1an, 2024-06-04:
Yes might have needed it. Seem ok now.

closed 2024-06-25 by proddy

undulanti, 2024-06-30:
@V4ler1an - Your suspicions about how your system was installed are correct - it's really rare to have boilers installed with hot water diverter kits here. In any event, I think this video might help you: https://www.youtube.com/watch?v=RJCGOtDLuNQ.

V4ler1an, 2024-07-07:
Thanks, I had seen Tom's video actually. I'm not sure if that method will ensure my boiler's weather compensation add-on kit will continue to work, it may do. Any way, I decided to stick with using my EMS-Bus gateway with Home Assistant automations because it lets me run two weather curves - one low, one high. The boiler's primary heating curve is set really low: 35 to 45 deg. Then when needed, either for hot water demand or to warm up a room that's too cold, the curve is temporarily boosted to 60 to 75 deg. Gives a nice combo of very efficient flow temps 95% of the time then very rapid warm up if a room was unoccupied and the heating was off. E.g. my bedroom is off all day in winter but it will warm up as much as 5 degrees in only an hour, just before I go to bed.