emsesp/EMS-ESP32 #1771: Modbus UI not working correctly

mheyse, 2024-05-18:
### PROBLEM DESCRIPTION

When enabling Modbus in Settings/Application only the port is pre-filled with the correct default value (502) - "Max Clients" and "Timeout" are empty.

### REQUESTED INFORMATION

Built from commit 0c52fa14eaaa0b503499af509adf0fe6c76faa32 with

```
cd interface
yarn install
yarn typesafe-i18n --no-watch
yarn build
yarn webUI
cd ..
pio run -e esp32_16M
```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

Max Clients should be pre-filled with 10, Timeout with 1000 ms.

### SCREENSHOTS

[image]

proddy, 2024-05-19:
I'll take a look.

proddy, 2024-06-04:
merged with latest dev
fixed the UI settings (changed maxClient and timeoutMillis also in C code)
made standalone compile

PR #1786 (2024-06-04): Feat modbus
merge with dev, fix #1771
closed 2024-06-04 by proddy