emsesp/EMS-ESP32 #1774: [emsesp] remotetemp timeout hc0, stop sending roomtemperature to thermostat

mattreim, 2024-05-22:
### PROBLEM DESCRIPTION

For a few days now I've been getting this error message in the log:

`WARNING 93: [emsesp] remotetemp timeout hc0, stop sending roomtemperature to thermostat`

and the remote control stops working. The only thing that helps is a restart or a resend of ...remotetemp/...remotehum.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:
{
  "System Info": {
    "version": "3.7.0-dev.11",
    "uptime": "000+08:00:31.256",
    "uptime (seconds)": 28831,
    "platform": "ESP32",
    "arduino": "Tasmota Arduino v2.0.15",
    "sdk": "4.4.7.240312",
    "free mem": 152,
    "max alloc": 83,
    "free caps": 102,
    "used app": 1609,
    "free app": 375,
    "partition": "app1",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -70,
    "BSSID": "set",
    "TxPower setting": 0,
    "static ip config": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "192.168.178.1",
    "tz label": "Europe/Berlin"
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "esp32-422a13d8",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time water": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 2261,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 57298,
    "bus reads (tx)": 10682,
    "bus writes (tx)": 1,
    "bus incomplete telegrams": 55,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "CUSTOM",
    "locale": "de",
    "tx mode": 2,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "phy type": 0,
    "rx gpio": 4,
    "tx gpio": 5,
    "dallas gpio": 14,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 6,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 25,
    "web log buffer": 25
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Condens 5000i/Greenstar 8000/GC9800IW/GB192i.2",
      "device id": "0x08",
      "product id": 195,
      "version": "05.06",
      "entities": 60,
      "handlers received": "0xBF 0xC2 0x14 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B 0x04",
      "handlers fetched": "0xE6 0xEA 0x28",
      "handlers pending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlers ignored": "0xD7 0xD8 0x02D6 0xE7 0x029B 0x02E0 0x02EA"
    },
    {
      "type": "thermostat",
      "name": "UI800/BC400",
      "device id": "0x10",
      "product id": 4,
      "version": "49.04",
      "entities": 42,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A",
      "handlers ignored": "0xC4 0x047B 0xF7 0xBF 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC100H",
      "device id": "0x38",
      "product id": 200,
      "version": "40.04",
      "entities": 3,
      "handlers received": "0x042B 0x047B",
      "handlers pending": "0x0273 0x0A6A",
      "handlers ignored": "0xF7"
    }
  ]
}

```

### TO REPRODUCE

Just wait, it usually happens at night.

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

The heater also shows an error and it looks like this under devices:

[image]

### ADDITIONAL CONTEXT

[log.txt](https://github.com/emsesp/EMS-ESP32/files/15408052/log.txt)

mattreim, 2024-05-24:
Today's log:

2024-05-22 20:32:53.511 WARNING 117: [telegram] Incomplete Rx: 38 10 01 09 15 E2
2024-05-24 11:36:20.385 WARNING 118: [emsesp] remotetemp timeout hc0, stop sending roomtemperature to thermostat
2024-05-24 11:52:43.026 WARNING 119: [telegram] Incomplete Rx: 38 10 FF 01 09 10 FE
2024-05-24 11:52:43.094 WARNING 120: [telegram] Incomplete Rx: 38 10 FF 01 C2 10 FE
2024-05-24 11:52:43.135 WARNING 121: [telegram] Incomplete Rx: 38 10 01 09 10 E7
2024-05-24 11:52:43.176 WARNING 122: [telegram] Incomplete Rx: 38 10 01 09 10 E7
2024-05-24 11:52:43.218 WARNING 123: [telegram] Incomplete Rx: 38 10 01 09 11 F7
2024-05-24 11:52:43.259 WARNING 124: [telegram] Incomplete Rx: 38 10 01 09 11 E6
2024-05-24 11:52:43.300 WARNING 125: [telegram] Incomplete Rx: 38 10 01 09 11 E6
2024-05-24 11:52:43.569 WARNING 126: [telegram] Incomplete Rx: 38 10 FF 01 09 14 FC
2024-05-24 11:52:43.637 WARNING 127: [telegram] Incomplete Rx: 38 10 FF 01 C2 14 FC
2024-05-24 11:52:43.678 WARNING 128: [telegram] Incomplete Rx: 38 10 01 09 14 E3
2024-05-24 11:52:43.701 WARNING 129: [telegram] Incomplete Rx: 38 10 01 09 14 E3
2024-05-24 11:52:43.761 WARNING 130: [telegram] Incomplete Rx: 38 10 01 09 15 E2
2024-05-24 11:52:43.802 WARNING 131: [telegram] Incomplete Rx: 38 10 01 09 15 F3
2024-05-24 11:52:43.843 WARNING 132: [telegram] Incomplete Rx: 38 10 01 09 15 E2
2024-05-24 11:52:43.884 WARNING 133: [telegram] Incomplete Rx: 38 10 01 09 15 E2
2024-05-24 11:52:43.926 WARNING 134: [telegram] Incomplete Rx: 38 10 01 09 16 E1

proddy, 2024-05-24:
It's by design to switch off after 3hrs of no activity (https://github.com/emsesp/EMS-ESP32/discussions/1680) but there may be something wrong. Michael is away at the moment, I'm sure he'll take a look when he's back in a week or two

mattreim, 2024-05-24:
Thanks for the information

MichaelDvP, 2024-05-30:
Yes, this is the timeout from #1680. Maybe a bit too short, temperature changes very slowly in night with no sushine to the house. Should we set it to one day?

mattreim, 2024-05-30:
Yes, three hours is too short. Please set it to one day, it should solve the problems.

mattreim, 2024-06-04:
Many thanks for the help.

airhead1234, 2024-06-06:
Could the timeout period be a parameter that users can set matching their individual setup? To me a day is pretty long.

proddy, 2024-06-09:
yes. we can make this an option

airhead1234, 2024-06-15:
That would be highly appreciated. Thank you!

proddy, 2024-06-21:
@airhead1234 this has been added to dev16. Can you test?

airhead1234, 2024-06-22:
Thank you so much for implementing this option. I am happy to test.

Running dev17 with a timeout of 3 hours it does not work yet. Actually, whenever my sensor send a temperature update there is a timeout immediately - see log. If I do not check "Disable remote on missing roomtemperature" there are not problems, but also no timeouts after 3 hours.

```
2024-06-22 10:54:35.839 I 66: [command] Calling command 'thermostat/remotetemp' (Raumtemperatur Remote) with value 27.3 and id 1 on device 0x10
2024-06-22 10:54:46.836 W 67: [emsesp] remotetemp timeout hc0, stop sending roomtemperature to thermostat
2024-06-22 10:55:57.756 I 68: [command] Calling command 'thermostat/remotetemp' (Raumtemperatur Remote) with value 27.3 and id 1 on device 0x10
2024-06-22 10:56:08.614 W 69: [emsesp] remotetemp timeout hc0, stop sending roomtemperature to thermostat
```

MichaelDvP, 2024-06-22:
Ups, sec/ms mismatch, i'll fix

airhead1234, 2024-06-22:
I have tested the revised dev17 and not it works perfectly. Thank you very much, this is a great feature!

commit c8300a7a (2024-05-30): Merge pull request #1783 from MichaelDvP/dev

dev12, fixes #1774 (change timeout #1680 to 24h) and #1779
closed 2024-06-23 by mattreim