emsesp/EMS-ESP32 #1775: EMS bus disconnected

wojtulab, 2024-05-24:
Hello, 
I've received the S3 Standard Wifi Version.  I have connected via service jack to Buderus GB022 and receive the information: 
"EMS bus disconnected. If this warning still persists after a few seconds please check settings and board profile"
[image]

I have read the official guide and issues on github. Unfortunately I am unable to fix it. Does anyone know how to fix it?

Board profile is: BBQKees Gateway S3
TX mode: EMS
Device: Buderus GB022

My config:
[image]

System log:
[image]

wojtulab, 2024-05-24:
I have changed the method of connection. The service jack has been replaced by BUS: 12VDC together with two BUS lines. The furnace and the RC310 controller have been found and can be seen in the dashboard.

[image]

closed 2024-05-24 by wojtulab

proddy, 2024-05-24:
glad it works.

@bbqkees do you know why it wouldn't work with the service connector jack? Need to update https://bbqkees-electronics.nl/wiki/gateway/connecting-to-specific-boilers-and-heatpumps.html#connecting-the-gateway-to-specific-boilers-and-heatpumps ?