emsesp/EMS-ESP32 #1781: change “hc1 selected room temperature” in cooling mode not possible

kwant64, 2024-05-30:
### PROBLEM DESCRIPTION

_A clear and concise description of what the problem is._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

We make use of a bbq-kees EPS gateway.
We use it for a NEFIT BOSCH Heatpump.

Wen we select Cooling it is not possible to change “hc1 selected room temperature”
When we change this value, an other field is changing. ( “hc1 manual temperature”).
On the device we kan change that value but not true the gateway.
Is this a bug or is ment to be in this way.
When we change to heating we kan change the value off the manual temperature and de selected room temperature is changing also…
We like to have the posseblillty to change de selected room temperature also in cooling mode.

Is there a possibility to fixed this?
```

### TO REPRODUCE

_Steps to reproduce the behavior:_

JUst place the heatpunp in cooling mode en try to change  “hc1 selected room temperature”

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

We expected a change of this value

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MichaelDvP, 2024-05-30:
Please log all while changing the value on the device, see https://emsesp.github.io/docs/Troubleshooting/#im-missing-certain-data-from-an-ems-device

kwant64, 2024-05-30:
[log(3).txt](https://github.com/emsesp/EMS-ESP32/files/15497735/log.3.txt)

Here is the log file.

MichaelDvP, 2024-05-30:
This is a very short log and it does not contain any change. It also looks like from an older ems-esp version. 
Send the system info and a log of > 2 min for the change of the value (working change on the device in cooling mode)

MichaelDvP, 2024-05-31:
I know what the commands seltemp and manualtemp in ems-esp do, but i don't know what changing the temp on the thermostat change in cooling mode. **You have to change the temp on the thermostat and log this.**
I can see that the monitor reports 0x28 == 20°C as selected temp. In settings i see only one 0x28 in offset 17, maybe this is the cooling temp. Just try with a custom value:
[image]

kwant64, 2024-05-31:
I think it would be wise to explain a little more.

We are working on building a remote thermostat that operates via MQTT and EMS-EPS heat pump.
So we change the values ​​via mqqt directly in the EMS-ESP.
The value of the hc1 selected room temperature is a value that we change.
If the heat pump is in heating mode, we can change this value via the EMS-EPS. However, this is not the case if it is in cooling mode, while it is a parameter that can be adjusted via the heat pump display. You will then also see the value in the EMS-EPS change.
In my opinion, it should therefore be a value that can also be changed via the EMS-EPS.
This is only partially the case, only in heating mode via the EMS-EPS and not further.
So I can send the values ​​as requested over the EMS bus in a different way, but that is not the way we want the control to work.

Summarizing:
I adjusted the value via the web interface of the EMS-EPS.
I see in heating mode that the value can be changed. (both on the heat pump display and in the EMS-ESP)
I see in coolling mode that this is not possible. (while the EMS-EPS already makes the parameter available writeble.)

I hope it gives a little more insight into the situation and the problem for us.

MichaelDvP, 2024-05-31:
Yes, it is clear, you like to have seltemp working in manual mode, auto mode and cooling mode. 
But we don't know where the setpoint in cooling mode is stored, you have to figure out. 
The thermostats reports the actual setpoint in offset 3 (0x28 = 20.0°C) of:
`RC300Monitor(0x02A5), data: 00 98 40 28 5A 00 28 28 05 A0 02 03 03 05 A0 05 A0 00 00 11 03 02 FF FF 00`
But this is only reporting and not writable, set settings are stored here:
`RC300Set(0x02B9), data: 00 2E 2A 26 1E 03 00 FF FF 05 24 01 E1 00 00 00 00 28`
and, depending on mode the thermostat picks one, in automode from offset 8 (0xFF means not active), in manual mode from offset 10 (0x24 = 18.0°C).  Because there is only one value 0x28 i guess this could be the setpoint for cooling and suggest to create a custom value to test it. If this is not working you have to do the a change on the thermostat for the setpoint while logging to see which value changes. Than we can adapt the seltemp command to write to this position in cooling mode.

proddy, 2024-08-03:
@kwant64 can we close this?

kwant64, 2024-08-03:
Yes you can.Verstuurd vanaf mijn iPhone

closed 2024-08-03 by proddy