emsesp/EMS-ESP32 #1796: Wrong state_class in HomeAssistant

tefracky, 2024-06-12:
### PROBLEM DESCRIPTION

In HomAssistant (currently 2024.6.2) the following error occurs:

`Entity sensor.heizung_eltern_solar_energylasthour (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total_increasing', 'total'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

### REQUESTED INFORMATION

- [X] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [X] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the output of http://ems-esp.local/api/system :

```lua
{
  "System Info": {
    "version": "3.7.0-dev.12",
    "uptime": "003+17:55:02.843",
    "uptime (seconds)": 323702,
    "platform": "ESP32",
    "arduino": "Tasmota Arduino v2.0.15",
    "sdk": "4.4.7.240312",
    "free mem": 133,
    "max alloc": 61,
    "free caps": 83,
    "used app": 1609,
    "free app": 375,
    "partition": "app1",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "Heizung-Eltern",
    "RSSI": -62,
    "TxPower setting": 0,
    "static ip config": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/Berlin"
  },
  "MQTT Info": {
    "MQTT status": "disconnected",
    "MQTT publishes": 101569,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 3,
    "enabled": true,
    "client id": "heizung-eltern",
    "keep alive": 60,
    "clean session": false,
    "entity format": 4,
    "base": "Heizung-Eltern",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time water": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 663324,
    "bus reads (tx)": 166257,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "de",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": false,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true,
    "max web log buffer": 100,
    "web log buffer": 100
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "Condens 5000i/Greenstar 8000/GC9800IW/GB192i.2",
      "device id": "0x08",
      "product id": 195,
      "version": "03.07",
      "entities": 70,
      "handlers received": "0xBF 0xC2 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x04",
      "handlers fetched": "0x14 0xE6 0xEA",
      "handlers pending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x2E 0x3B",
      "handlers ignored": "0x0CB8 0xE7 0xE8 0x02E0 0x02EA 0x0CB9 0x02D6 0xF7 0xF6 0x17"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.03",
      "entities": 52,
      "handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240 0x0269",
      "handlers pending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E",
      "handlers ignored": "0xBF"
    },
    {
      "type": "solar",
      "name": "SM200/MS200",
      "device id": "0x30",
      "product id": 164,
      "version": "25.06",
      "entities": 42,
      "handlers received": "0x0362 0x0363 0x0366 0x0364 0x036A",
      "handlers fetched": "0x0358 0x035A 0x035D 0x0380 0x038E 0x0391 0x035F 0x035C 0x0361",
      "handlers ignored": "0x0368 0x0369 0x03A9 0x03AA 0xBF 0x038A"
    },
    {
      "type": "controller",
      "name": "9000i",
      "device id": "0x09",
      "product id": 224,
      "version": "35.06",
      "entities": 0
    }
  ]
}
```

### TO REPRODUCE

I have a Buderus GB192i with a RC310 and a solar pump (not sure which one). The Energy Last Hour (which is exported via MQTT to HomeAssistant) has the state_class 'measurement', which is incompatible with the device_class 'energy'.

### EXPECTED BEHAVIOUR

The state_class should be 'total'.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-06-12:
Thanks, it's one line to change and we'll use `total_increasing` so its consistent if that's ok. If you can't wait for the new build you can use MQTTExplorer and just update the topic name manually.

https://github.com/emsesp/EMS-ESP32/blob/c8300a7a1284a721a3cfe7b7f2e165e5e91fee3f/src/mqtt.cpp#L1179

tefracky, 2024-06-12:
Hi,
thanks for the quick answer!

I think it should be `total`, since the values might also be decreasing.

[image]

closed 2024-06-19 by proddy