emsesp/EMS-ESP32 #1800: Setting thermostat/seltemp noit working with gui nor api

harvey637, 2024-06-15:
Hi,
running latest dev13, but seeing in logs this issue startet about beginning of april

What I do (automatically):
lower the temperature minus 2°C in the morning (Mo-Fr) and rising plus 2°C in the evening (working until start of april)

What I did manually:
changed temperature with real remote RC100H to 23°C (working!)
changed seltemp to 22°C with gui (not working)

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

related log to manually setting new temperature looks like this:

```
2024-06-15 09:23:30.200 INFO 544: [command] Calling command 'thermostat/seltemp' (Sollwert Raumtemperatur) with value 22 and id 1 on device 0x10
2024-06-15 09:23:30.394 TRACE 545: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300Set(0x02B9), data: 2C (offset 22)
2024-06-15 09:23:30.540 TRACE 546: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A7), length: 1 (offset 37)
2024-06-15 09:23:30.557 TRACE 547: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A7), data: <empty> (offset 37)
2024-06-15 09:23:32.507 TRACE 548: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 25
2024-06-15 09:23:32.580 TRACE 549: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 B7 21 2E 1E 00 2E 2A 05 A0 07 03 03 05 A0 05 A0 00 00 11 01 03 07 27 00
2024-06-15 09:23:32.820 TRACE 550: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 25 (offset 25)
2024-06-15 09:23:32.865 TRACE 551: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 08 04 00 00 00 00 0B 64 30 00 3C 01 37 (offset 25)
2024-06-15 09:23:33.052 TRACE 552: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A8), length: 2
2024-06-15 09:23:33.070 TRACE 553: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A8), data: <empty>
2024-06-15 09:23:33.596 TRACE 554: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A8), length: 2 (offset 8)
2024-06-15 09:23:33.613 TRACE 555: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A8), data: <empty> (offset 8)
2024-06-15 09:23:33.900 TRACE 556: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A8), length: 1 (offset 7)
2024-06-15 09:23:33.917 TRACE 557: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A8), data: <empty> (offset 7)
```

to reproduce:
change seltemp by api or by gui: not working

expected:
seltemp is changed not only by rc100h device, but also by api and gui

more info:
there is some related issue with seltemp and cooling, but this is different, as it is in normal operating mode. but the "summermode" might have some influence?
My finding is, that "summertemp" is the temperature with below is "heating season", above is "non heating season". This is reproduced with the reading on the rc300 display in my bosch 7000 HP. Not found/verified is the "minimum cooling temperature" from the rc300 panel to any ems-protokoll readout, there is NO temperature equal to the rc300 "cooling above" temperature.
But now we are far away from cooling :)

more background:
I changed the "summertemp" from 13°C to 11°C around beginning of april with the idea, that even if it is below 11°C in the very early morning the sun is so strong that it is possible to warm up the house during daytime without heating in the morning. 

if more info needed, just ask!
cu
Harvey

MichaelDvP, 2024-06-15:
Can you send a longer log where we can see the complete telegram `RC300Set(0x02B9)` and communicaton with 0x38

I think this related to #1779, changing the seltemp in pos 22 if mode is stored in pos21. But the CR120 thermostat seems to be special in a lot of things, maybe we should revert this and make an extra model for CR120.

The other is the RC100H, changing the temperatur is send to 0x02B9 pos 17 as a temporary change from remote. sadly we can't write to this position, the RC300 only except values from the remote device 0x38. 
You can check with custom entities:
device 0x10, 0x2B9, offset 10, factor 0.5 should set the manual temp (only if offset 17 is cleared)
device 0x10, 0x2B9, offset 17, factor 0.5 is what the remote sets, but i think wite is not accepted from other devices than 0x38.
device 0x38, 0x2B9, offset 17, factor 0.5 is worth a try, but i think the remote RC100H is not writeable.

proddy, 2024-06-15:
I'm thinking also we should create a new thermostat type for the CR120 - i will give us more flexibility and less if-then-else statements

harvey637, 2024-06-15:
it WAS working, see graph:
the seltemp is reduced 2°C in the morning and raised 2°C in the evening, but only on weekdays Mo-Do.

[image]

Fridays and weekend temperature stays as set in "prog_0".

By the way: custom entities with int8?

harvey637, 2024-06-15:
I was on holiday than, but after comming back ther is no reducing seltemp between morning and afternoon anymore.
Graph is starting in holiday with prog_1 (only small teperature peeks in the morning no to cool down the house too much),
than comming back, changing the program to prog_0 with heating from morning to early night, but NO MORE reducing as in the graph before.
I am not quite shure about the ems-esp versions, but I change them rather shortly after publishing.
Before yesterday I was running dev12, now dev13, and before my holiday at beginning april possibly dev10 ....

MichaelDvP, 2024-06-15:
>  By the way: custom entities with int8?

Yes.

harvey637, 2024-06-15:
[image]

the 16°C is found as 
[image]

the 25°C is
[image]

MichaelDvP, 2024-06-15:
Please set the customs as writable and test if you can change the value.

harvey637, 2024-06-15:
Customs are writeable, I can change the offset 10 to 16.5°C

But .... the 16.0°C (and the 16.5°C) are not the selected room temperature (which is 19.5°C as selected by the dial of the RC100H room thermostat) ?!? 

I can also change the 25.0°C to 25.5°C, which is propagated after a few seconds to the third custom entity .... but I see no 25,0 nor 25.5°C in the "normal" thermostat?

Possibly related to discussion #1779 ? There we also have the strange 16°C .......

harvey637, 2024-06-15:
some more log, I changed in the gui the seltemp from 19,5°C to 19°C .... with is not done :-(

```
000+00:05:41.124 N 329: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 6A 08
000+00:05:41.182 N 330: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 6A 08 4F 00 00 00 01 00 00 00 02 00 06 00 03 00 06 00 02
000+00:05:42.676 N 331: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 6B 08
000+00:05:42.734 N 332: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 6B 08 4F 00 00 00 01 00 00 00 02 00 06 00 03 00 06 00 02
000+00:05:44.188 N 333: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 6C 08
000+00:05:44.246 N 334: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 6C 08 4F 00 00 00 01 00 00 00 02 00 06 00 03 00 06 00 02
000+00:05:45.124 N 335: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 6D 08
000+00:05:45.182 N 336: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 6D 08 4F 00 00 00 01 00 00 00 02 00 06 00 03 00 06 00 02
000+00:05:45.590 N 337: [emsesp] thermostat(0x38) -W-> thermostat(0x10), RemoteTemp(0x042B), data: 00 BB 07 50 01
000+00:05:45.623 N 338: [emsesp] thermostat(0x38) -W-> thermostat(0x10), RemoteHumidity(0x047B), data: 0B 37 00 72
000+00:05:45.654 I 339: [command] Calling command 'thermostat/seltemp' (Sollwert Raumtemperatur) with value 19 and id 1 on device 0x10
000+00:05:45.793 N 340: [emsesp] Me(0x0B) -W-> thermostat(0x10), RC300Set(0x02B9), data: 26 (offset 22)
000+00:05:47.818 N 341: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 25
000+00:05:47.890 N 342: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 BB 61 27 00 00 27 2A 05 A0 03 03 03 05 A0 05 A0 00 00 11 02 03 07 50 00
000+00:05:48.130 N 343: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 25 (offset 25)
000+00:05:48.176 N 344: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 08 08 00 00 00 00 0B 64 30 00 3C 01 37 (offset 25)
000+00:05:48.535 N 345: [emsesp] thermostat(0x10) -B-> All(0x00), RCTime(0x06), data: 18 06 0C 0F 10 30 05 01 10 FF 00
000+00:05:51.731 N 346: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0xE7), data: 00 00 00
000+00:05:51.764 N 347: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0x02E0), data: 00 00 00 00 01
000+00:05:51.794 N 348: [emsesp] thermostat(0x10) -W-> boiler(0x08), ?(0x02EA), data: 00
000+00:05:51.845 N 349: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 08 08 00 00 00 00 0B 64 30 00 3C 01 37 (offset 25)
000+00:05:52.030 N 350: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Floordry(0x0267), data: 00 00
000+00:05:52.459 N 351: [emsesp] thermostat(0x10) -B-> All(0x00), RC300Monitor(0x02A5), data: 00 BB 61 27 00 00 27 2A 05 A0 03 03 03 05 A0 05 A0 00 00 11 02 03 07 50 00
000+00:05:52.647 N 352: [emsesp] thermostat(0x10) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 03 07
000+00:06:01.788 N 353: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 AF 08
000+00:06:01.846 N 354: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 AF 08 0B 00 00 00 14 00 00 00 19 00 00 00 2D 00 00 00 19
000+00:06:02.428 N 355: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 9B 08
000+00:06:02.486 N 356: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 9B 08 03 00 00 00 1E 00 00 00 4B 00 00 00 55 00 00 00 4B
000+00:06:03.068 N 357: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 9B 07
000+00:06:03.126 N 358: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 9B 07 07 00 00 00 1E 00 00 00 30 00 00 00 3C 00 00 00 30
000+00:06:03.468 N 359: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 9B 00
000+00:06:03.526 N 360: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 9B 00 4F 00 00 00 01 00 00 00 01 00 03 00 07 00 03 00 01
000+00:06:06.488 N 361: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A6), length: 1 (offset 3)
000+00:06:06.492 N 362: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A6), data: <empty> (offset 3)
000+00:06:07.384 N 363: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A6), length: 1 (offset 4)
000+00:06:07.388 N 364: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A6), data: <empty> (offset 4)
000+00:06:08.000 N 365: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A6), length: 2
000+00:06:08.004 N 366: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A6), data: <empty>
000+00:06:08.624 N 367: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A6), length: 2 (offset 8)
000+00:06:08.628 N 368: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A6), data: <empty> (offset 8)
000+00:06:09.856 N 369: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 25
000+00:06:09.928 N 370: [emsesp] thermostat(0x10) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 BB 61 27 00 00 27 2A 05 A0 03 03 03 05 A0 05 A0 00 00 11 02 03 07 50 00
000+00:06:10.793 N 371: [emsesp] Me(0x0B) -R-> thermostat(0x10), RC300Monitor(0x02A5), length: 25 (offset 25)
[... cut ...]
000+00:06:23.868 N 435: [emsesp] gateway(0x48) -R-> thermostat(0x10), ValueConfig(0xF9), length: 17, data: FF 01 9D 00
000+00:06:23.926 N 436: [emsesp] thermostat(0x10) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 01 9D 00 4B 00 00 00 01 00 00 00 01 00 03 00 07 00 03 00 01
000+00:06:25.959 N 437: [emsesp] gateway(0x48) -R-> thermostat(0x10), RC300Monitor(0x02A7), length: 1 (offset 37)
000+00:06:25.962 N 438: [emsesp] thermostat(0x10) -W-> gateway(0x48), RC300Monitor(0x02A7), data: <empty> (offset 37)

```

harvey637, 2024-06-15:
Hi, I experimented around an created custom entity with device_id 10, type 2B9 and offset 8 int8 factor 0.5 °C writable.

And a change to THIS entity changed the "seltemp"  !!!!!!!

harvey637, 2024-06-15:
confirmed:
writing to this custom entity with offset = 8 works, changes the seltemp in the ems-esp thermostat device an also in the display of the rc100H room thermostat device.

[image]

MichaelDvP, 2024-06-15:
Oh, there is an initialization error of the new mode from #1779, your 0x98B does not have offset 21/22, so mode is also wrong. Offset 8 is seltemp in automode.
I'll fix.

MichaelDvP, 2024-06-15:
Please check if https://github.com/MichaelDvP/EMS-ESP32/releases works.

harvey637, 2024-06-15:
Hi, thank you for the fast solution!

I checked the changing of the "seltemp" (hc1/settemp) -> working now, changes are reflected in gui and rc100H device also.
OK!!!!!

The only thing I am not shure is "hc1/modetype": it is "Komfort" now, readonly.
Hmmmm, was it writable with "eco/comfort" before? ... dont remember.

MichaelDvP, 2024-06-15:
modetype was/is not writeable, it only shows the state in automode.

harvey637, 2024-06-15:
I was just thinking .... thank you for the explanation.

by the way: IMHO a better word as "ADDITION" in the customs might be "OFFSET" even in german.
happy weekend!
cu
Harvey

MichaelDvP, 2024-06-15:
> by the way: IMHO a better word as "ADDITION" in the customs might be "OFFSET" even in german.

Yes, or POSITION?

harvey637, 2024-06-15:
yes (but use it in a consistant way in your support when creating custom entities too :-)

proddy, 2024-06-25:
ok to close this one?

commit 55316334 (2024-06-15): Merge pull request #1803 from MichaelDvP/dev

fix #1800
closed 2024-07-05 by proddy