emsesp/EMS-ESP32 #1801: set the duration trigger of a shower

proddy, 2024-06-15:
Hi, is there any possibility in ESP config to set the duration trigger of a shower higher than 2min? Many thanks in advance...

_Originally posted by @NongJoWo in https://github.com/emsesp/EMS-ESP32/discussions/1293#discussioncomment-9776370_

proddy, 2024-06-15:
Todo: make SHOWER_MIN_DURATION a setting

proddy, 2024-06-20:
in dev14 - can you test @NongJoWo

proddy, 2024-06-20:
> I don't have any experience with GitHub projects. How can I test it?

Download the latest dev release from https://github.com/emsesp/EMS-ESP32/releases/tag/latest, or use the EMS-ESP Web user interface to download this for you. It'll be version 3.7.0 so if you're coming from 3.6.x there will be a few surprises!

NongJoWo, 2024-06-20:
OK, the update was really easy! I've adjusted the shower timer. Now I just have to take a shower. I'll let you know when everything is OK. Thanks for the super-fast support!!

NongJoWo, 2024-06-22:
So just tested. The shower trigger works perfectly. Unfortunately, the shower duration no longer works. Only the value 0 is returned here. There is also a new MQTT-device: ems-esp Shower. However, this does not contain any entities.

proddy, 2024-06-23:
thanks for reporting. I porbably broke something. Will take a look and test soon

NongJoWo, 2024-06-23:
Should I download and test the new version?

NongJoWo, 2024-06-23:
Just tested. Now everything is as expected. Many thanks again!

closed 2024-06-23 by proddy