emsesp/EMS-ESP32 #1838: add Thermostat RC220 for BC400 HP

Bingo2023, 2024-07-03:
We startet to make some comands for the BC400 already.
I feel like we need to do more in this direction to also cover the thermostats for the new HPs.

My starting point for this was to set up a virtual Remote-Thermostat.
This can not be done because the virtual need to be same as the real thermostat (in my case RC220, see pic below).
Unfortunately I can not choose "RC220" in the dropdown of (see second pic).

I started to pull the dev from github, to do some experiments.
Unfortunately I need so help how to start, thermostat.cpp is long&complex.
From trace below u see e.g. last line that RC220 has ID 0x38.
Should we add an own section for only 0x38 or should be combine it with more thermostats (keep the range ID approach)?
@MichaelDvP or @proddy : maybe u can point this out?

[image]
[image]

`001+20:56:35.523   TRACE 868:    [emsesp]     thermostat(0x10) -W-> Me(0x0B), RC300OutdoorTemp(0x023A), data: 00 F7
001+20:56:35.850   TRACE 869:    [emsesp]     Me(0x0B) -R-> thermostat(0x10), RC300Settings(0x0240), length: 25
001+20:56:35.920   TRACE 870:    [emsesp]     thermostat(0x10) -W-> Me(0x0B), RC300Settings(0x0240), data: 00 03 00 00 00 00 00 FF 01 F4 06 FF 00 00 02 00 32 00 00 00 00 00 01 00 (offset 1)
001+20:56:35.920   TRACE 871:    [emsesp]     Me(0x0B) -R-> thermostat(0x10), HybridSettings(0xBB), length: 27
001+20:56:35.939   TRACE 872:    [emsesp]     thermostat(0x10) -W-> Me(0x0B), HybridSettings(0xBB), data: 00 (offset 14)
001+20:56:36.337   TRACE 873:    [emsesp]     Me(0x0B) -R-> thermostat(0x10), PVSettings(0x023E), length: 25
001+20:56:36.370   TRACE 874:    [emsesp]     thermostat(0x10) -W-> Me(0x0B), PVSettings(0x023E), data: 02 02 00 FF 00 00 00
001+20:56:36.582   TRACE 875:    [emsesp]     gateway(0x48) -R-> boiler(0x08), ValueConfig(0xF9), length: 17, data: FF 03 99 01
001+20:56:36.619   TRACE 876:    [emsesp]     boiler(0x08) -W-> gateway(0x48), ValueConfig(0xF9), data: FF 03 99 01 8F 00 00 00 3A 00 00 00 40 00 00 00 40 00 00 00 40
001+20:56:44.687   TRACE 877:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 10 00 CC 00 00 00 00 02 B2 00 00 00 CB 00 00 00 00 00 CC 00 CC
001+20:56:44.878   TRACE 878:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 02 00 10 00 00 00 00 00 00 00 00 00 CC 00 00 50
001+20:56:45.074   TRACE 879:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 32 02 55 01 95 00 00 00 00 41 3C 00 00 01 00 00 00 00 00 00 00 00 00 32 02
001+20:56:48.691   TRACE 880:    [emsesp]     thermostat(0x10) -R-> boiler(0x08), MenuConfig(0xF7), length: 1, data: E4 00 00 (offset 4)
001+20:56:48.695   TRACE 881:    [emsesp]     boiler(0x08) -W-> thermostat(0x10), MenuConfig(0xF7), data: E4 00 00 (offset 4)
001+20:56:51.745   TRACE 882:    [emsesp]     thermostat(0x10) -B-> All(0x00), RCTime(0x06), data: 18 07 12 03 08 3B 02 01 10 FF 00
001+20:56:54.751   TRACE 883:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 10 00 CC 00 00 00 00 02 B2 00 00 00 CB 00 00 00 00 00 CC 00 CC
001+20:56:54.939   TRACE 884:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 02 00 10 00 00 00 00 00 00 00 00 00 CC 00 00 50
001+20:56:55.136   TRACE 885:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 32 02 55 01 95 00 00 00 00 41 3C 00 00 01 00 00 00 00 00 00 00 00 00 32 02
001+20:56:59.883   TRACE 886:    [emsesp]     thermostat(0x38) -W-> thermostat(0x10), RemoteHumidity(0x047B), data: 0D 32 00 86
001+20:57:04.690   TRACE 887:    [emsesp]     boiler(0x08) -R-> thermostat(0x10), RC300Settings(0x0240), length: 1 (offset 12)
001+20:57:04.694   TRACE 888:    [emsesp]     thermostat(0x10) -W-> boiler(0x08), RC300Settings(0x0240), data: FF (offset 12)
001+20:57:04.750   TRACE 889:    [emsesp]     boiler(0x08) -R-> thermostat(0x10), HPMode(0x0291), length: 1
001+20:57:04.772   TRACE 890:    [emsesp]     thermostat(0x10) -W-> boiler(0x08), HPMode(0x0291), data: 01
001+20:57:04.813   TRACE 891:    [emsesp]     boiler(0x08) -R-> thermostat(0x10), RC300Curves(0x029B), length: 1 (offset 1)
001+20:57:04.836   TRACE 892:    [emsesp]     thermostat(0x10) -W-> boiler(0x08), RC300Curves(0x029B), data: 03 (offset 1)
001+20:57:04.876   TRACE 893:    [emsesp]     boiler(0x08) -R-> thermostat(0x10), RC300Curves(0x029B), length: 2 (offset 7)
001+20:57:04.900   TRACE 894:    [emsesp]     thermostat(0x10) -W-> boiler(0x08), RC300Curves(0x029B), data: 37 41 (offset 7)
001+20:57:04.961   TRACE 895:    [emsesp]     boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 09 01 00 00 00 00 01 00 01 00 00 00 00
001+20:57:05.163   TRACE 896:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 00 10 00 CC 00 00 00 00 02 B2 00 00 00 CB 00 00 00 00 00 CC 00 CC
001+20:57:05.338   TRACE 897:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAOutdoorTemp(0xD1), data: 00 F7 00
001+20:57:05.602   TRACE 898:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus2(0xE3), data: 02 00 10 00 00 00 00 00 00 00 00 00 CC 00 00 50
001+20:57:05.798   TRACE 899:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorSlowPlus(0xE5), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 2)
001+20:57:05.998   TRACE 900:    [emsesp]     boiler(0x08) -B-> All(0x00), UBAMonitorWWPlus(0xE9), data: 32 02 55 01 95 00 00 00 00 41 3C 00 00 01 00 00 00 00 00 00 00 00 00 32 02
001+20:57:06.252   TRACE 901:    [emsesp]     thermostat(0x38) -W-> thermostat(0x10), RemoteTemp(0x042B), data: 00 F3 09 82 01 00`

MichaelDvP, 2024-07-03:
The data from the RC220 are one byte longer than RC200 `RemoteTemp(0x042B), data: 00 F3 09 82 01 00` the meaning of the last two bytes are unknown, 
All ems+ remote thermostats send from device-id 0x38.
We need to know the settings for control, telegram 0x291, offset 3. The known settings for RC3xx are (rc310), (rc200), (rc100),(rc100h), (tc100)). What is the advantage of RC220 over RC100 or RC200 as remote? It's always just sending the roomtemp,

Bingo2023, 2024-07-03:
Telegram 0x291:

- 0 = keine
- 2 = RC100 / RC100.2
- 3 = RC100 H / RC100.2 H
- 5 = RC120 RF
- 6 = RC220
- 7 = Einzelraumregelung

[image]
[image]

Bingo2023, 2024-07-03:
I'm not aware about the differences in the thermostats. Also I dont know what the additional 1 Byte is carring.

I guess RC220 is newer model with more features.

Bingo2023, 2024-07-04:
one more idea:
can we eliminate the "ems-esp/thermostat/hc1/control" command to activate the virtual remote?
We can just read the real Thermostat and use exact same for virtual remote, because if they are different the HP will not accept and throw a error message.

MichaelDvP, 2024-07-04:
> I'm not aware about the differences in the thermostats. 

Why do you want the RC220 then to be implemented? Just use the RC100/RC100H emulation as you can select it.
I can add the new choices to control, but to implement a RC220 emulation a lot of more data is needed  and we have to understand all the telegram settings.

Bingo2023, 2024-07-04:
let me explain again... sorry for german, but that is easier for both of us.

Wenn ich an der WP in dem Menü (also dem BC400) eine andere Fernbedienung als RC220 einstelle, meckert die WP und geht auf Fehler (-> falsche Fernbedienung ausgewählt).
D.h. im BC400 MUSS RC220 ausgewählt werden. Um jetzt eine virtuelle zu erstellen MUSS diese die gleiche sein wie die echte angeschlossene, also auch eine RC220.

Deshalb bin ich auf das Problem aufmerksam geworden.

MichaelDvP, 2024-07-04:
Wenn ein echter RC220 angeschlossen ist und du wählst eine andere FB aus, klar gibt das Fehler. Wenn du an einem anderen HK eine auswählst gibt das Fehler bis remotetemp gesendet wird, dann geht der Fehler weg. Echte RC220 und simulierte am gleichen HK gibt Buskollision und Fehler. 
Klemm den RC220 ab, setze Control aud RC100 und sende dann remotetemp.

Same in en:
If aRC200 is connected and you change to another control it will give error. If you set a control on a different hc it will show error until you send the remottemp. A real RC220 and a simulated on the same hc gives bus collision and will show a different error.
Disconnect the RC220, set control to RC100 and then send remotetemp.

Bingo2023, 2024-07-04:
> The data from the RC220 are one byte longer than RC200 `RemoteTemp(0x042B), data: 00 F3 09 82 01 00` the meaning of the last two bytes are unknown, All ems+ remote thermostats send from device-id [...]

how can I configure the "individual entity" to capture the new features?

proddy, 2024-08-03:
Was added to 3.7.0-dev

closed 2024-08-03 by proddy

Bingo2023, 2024-08-08:
I confirm, virtual RC220 on BC400 is working with
3.7.0-dev.30

great job !