emsesp/EMS-ESP32 #1849: Reboot Loop with latest development build 22

RalfLuig, 2024-07-06:
I flashed the latest development build yesterday. The device is connecting to the EMS bus and I see the entities coming back from my
heating system. After couple seconds there might be a reset and it is startting again discovering the entities.

I moved back to development build 19 and this beavior disapeared.

My heating system is a Bosch Condens 8000if.

Gateway E32 V2 (Ethernet + WiFi Edition V2)

MichaelDvP, 2024-07-06:
Discussed on discord, a PR is on the way, disable HA discovery until dev.23 is out.

proddy, 2024-07-06:
thanks @RalfLuig for helping out. It was a tricky one.

RalfLuig, 2024-07-06:
you are welcome, thanks :-)

RalfLuig, 2024-07-06:
Now there is another issue with development build 23. MQTT is not working anymore. I don‘t see any previous topics in MQTT Explorer and Home Assistant, with development build 19 everything is ok.

proddy, 2024-07-07:
> Now there is another issue with development build 23. MQTT is not working anymore. I don‘t see any previous topics in MQTT Explorer and Home Assistant, with development build 19 everything is ok.

please send system info

RalfLuig, 2024-07-07:
Problem solved - I had to clear all MQTT topics in HA - reboot the EMS-ESP. Now everything is  working as expected.

closed 2024-07-07 by RalfLuig