emsesp/EMS-ESP32 #1857: Missing entities for cooling devices

thombec, 2024-07-10:
### PROBLEM DESCRIPTION

I have a Buderus WPLS.2 heat pump running with the HMC310 Controller. This devics can in addition to heat in wintertime start cooling in summer time. In the EMS-ESP are a lot of possible switches, controls and values be found. But the values for the cooling feature are missing.

These can only be programmed in the HMC310 interface:

[image]

Can you implement the entities for the cooling start temperature, and the switch-off and on delay?

proddy, 2024-07-10:
We certainly can, but need your help. See https://emsesp.github.io/docs/Troubleshooting/#im-missing-certain-data-from-an-ems-device

proddy, 2024-07-10:
and search for cooling in https://emsesp.github.io/docs/All-Devices/#all-devices-and-entities

thombec, 2024-07-10:
thank you for information!

I checked https://emsesp.github.io/docs/All-Devices/#all-devices-and-entities -> there I dont find my requested values.

Where can I find a description to read the EMS telegrams and test it at Custom Entities page in the WebUI to create a new entity. 

Here I was not able to find these points at my EMS-ESP32 Web interface from the device.

proddy, 2024-07-10:
First thing we always ask is for the system info (https://emsesp.github.io/docs/Support/). I suspect you're using an older version of the software.

thombec, 2024-07-10:
I am using 3.6.5

thombec, 2024-07-11:
I checked the missing entities via the recording of the telegrams and found the reuired ones as follows:

[emsesp] thermostat(0x10) -B->   All(0x00), RC300Summer2(0x0471), data: _xx_ (offset 5)
   Range: 20-35; Unit: °C; German Description: HK1   Kühlbetrieb ab

[emsesp] thermostat(0x10) -B-> All(0x00),   RC300Summer2(0x0471), data: _xx_ (offset 6)
  Range: 1-48; Unit: hours; German Descrition: HK1 Einschaltverzögerung Kühlen

[emsesp] thermostat(0x10) -B-> All(0x00),   RC300Summer2(0x0471), data: _xx_ (offset 7)
   Range: 1-48; Unit: hours; German Descrition: HK1 Ausschaltverzögerung Kühlen

At the entity "HK1 Sommertemperatur" I found, that the range is set to high in EMS-ESP. It should be 10-19 instead of 10-30. In addition the German name in the 310 is called "HK1 Heizbetreib ab" not "HK1 Sommertemperatur". I think a name adjustment makes sense due to the same structure of the cooling mode temperature.

proddy, 2024-07-19:
@MichaelDvP shall we add these 3 entities and rename the other one? I'll need help with the German to English translation.

MichaelDvP, 2024-07-19:
I think the mqtt/en/ger should be:
```
MAKE_TRANSLATION(coolstart, "coolstart", "cooling starttemp", "Kühlbetrieb ab") // TODO translate
MAKE_TRANSLATION(coolondelay, "coolondelay", "cooling on delay", "Einschaltverzögerung Kühlen") // TODO translate
MAKE_TRANSLATION(coolondelay, "cooloffdelay", "cooling off delay", "Ausschaltverzögerung Kühlen") // TODO translate
```

For renaming the summertemp and limits we have the customizations. Bosch renames too often. In a lot of RC3xx it is summertemp and we get the next comply few hours after renaming. Also renaming always changes mqtt/HA entities, etc. and a lot of users have to change scripts automations etc. 

Should i make the addition?

proddy, 2024-07-19:
yes please, thanks Michael.

MichaelDvP, 2024-07-19:
@thombec  Please check if all works: 
https://github.com/MichaelDvP/EMS-ESP32/releases

thombec, 2024-07-20:
Thank you for your help! 

I am currently on vacation and will check it directly afterwards.

thombec, 2024-08-04:
@MichaelDvP 

 I have checked it - it is ok.

Thank you!

proddy, 2024-08-16:
this can be closed when merged into dev

proddy, 2024-08-16:
coolstart, coolondelay & cooloffdelay added to 3.7.0-dev-30.

closed 2024-08-16 by proddy