emsesp/EMS-ESP32 #1859: Edit Temperature Sensor - Default name from ID to long

jeschofnik, 2024-07-11:
### PROBLEM DESCRIPTION

When I want to change the offset of a temperature sensor, it isn't possible to keep the device ID as the name because it is longer than 15 characters
### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  {"System Info":{"version":"3.7.0-dev.23","uptime":"000+00:13:11.593","uptime (seconds)":791,"platform":"ESP32","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","free mem":156,"max alloc":87,"free caps":106,"used app":1676,"free app":308,"partition":"app1","reset reason":"Power on reset / APP CPU reset by PRO CPU"},"Network Info":{"network":"WiFi","hostname":"ems-esp","RSSI":-81,"TxPower setting":78,"static ip config":false,"low bandwidth":false,"disable sleep":false,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected","AP security":"wpa2","AP ssid":"ems-esp"},"NTP Info":{"NTP status":"connected","enabled":true,"server":"time.google.com","tz label":"Europe/Vienna"},"MQTT Info":{"MQTT status":"connected","MQTT publishes":540,"MQTT queued":0,"MQTT publish fails":0,"MQTT connects":1,"enabled":true,"client id":"ems-esp","keep alive":60,"clean session":false,"entity format":0,"base":"ems-esp","discovery prefix":"homeassistant","discovery type":0,"nested format":1,"ha enabled":true,"mqtt qos":0,"mqtt retain":false,"publish time heartbeat":60,"publish time boiler":0,"publish time thermostat":0,"publish time solar":0,"publish time mixer":0,"publish time water":10,"publish time other":0,"publish time sensor":0,"publish single":false,"publish2command":false,"send response":false},"Syslog Info":{"enabled":false},"Sensor Info":{"temperature sensors":3,"temperature sensor reads":474,"temperature sensor fails":0,"analog sensors":0,"analog sensor reads":0,"analog sensor fails":0},"API Info":{"API calls":0,"API fails":0},"Bus Info":{"bus status":"connected","bus protocol":"Buderus","bus telegrams received (rx)":1219,"bus reads (tx)":404,"bus writes (tx)":0,"bus incomplete telegrams":0,"bus reads failed":0,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100},"Settings":{"board profile":"S32","locale":"en","tx mode":1,"ems bus id":11,"shower timer":false,"shower min duration":180,"shower alert":false,"hide led":false,"notoken api":false,"readonly mode":false,"fahrenheit":false,"dallas parasite":false,"bool format":1,"bool dashboard":1,"enum format":1,"analog enabled":true,"telnet enabled":true,"max web log buffer":50,"web log buffer":7},"Devices":[{"type":"boiler","name":"GBx72/Trendline/Cerapur/Greenstar Si","device id":"0x08","product id":123,"version":"07.00","entities":83,"handlers received":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlers fetched":"0x14 0x16 0x33 0x26 0x28","handlers pending":"0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B","handlers ignored":"0x23"},{"type":"thermostat","name":"RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410","device id":"0x10","product id":158,"version":"18.06","entities":55,"handlers received":"0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269","handlers pending":"0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E","handlers ignored":"0xBF"},{"type":"controller","name":"BC25","device id":"0x09","product id":125,"version":"03.05","entities":0}]}

```

### TO REPRODUCE

Attach Dallas sensor, goto sensors, try to change just the offset - leave name to default

e.g.: 28_0C2C_4592_0502 or 28_9CA8_4592_1402

since the default name is longer than 15 char, it's impossible to save

### EXPECTED BEHAVIOUR

offset can be changed without touching the name

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-07-11:
thanks! I made a fix in the next PR

jeschofnik, 2024-07-11:
perfect thanx!

closed 2024-07-11 by jeschofnik