emsesp/EMS-ESP32 #1860: Change key-names in JSON to be compliant and consistent

proddy, 2024-07-11:
The key names are inconsistent in the downloaded `emsesp_settings.json` file which is the same as the API call `http://ems-esp.local/api/system`.

I'd like to rename the keys to either camel-case or snake-case, following the standards that the JSON org and Google recommend: https://restfulapi.net/valid-json-key-names/ https://google.github.io/styleguide/jsoncstyleguide.xml?showone=Key_Names_in_JSON_Maps#Key_Names_in_JSON_Maps

This would mean names like `"publish time solar"` would become `"publishTimeSolar"` using camel-case. And `"Network Info"` changes to just `"network"`.

Only the System Info would be impacted. The devices and other entities are in lowercase without spaces and can stay like that.

@tp1de @MichaelDvP

tp1de, 2024-07-12:
Please consider to align mqtt (ha) and api names. 
I understand that you prefer CamelCase. Is this right?

proddy, 2024-07-12:
> Please consider to align mqtt (ha) and api names. I understand that you prefer CamelCase. Is this right?

yes camelCase. 

in the API, the URL won't be case sensitive. In HA I need to check if the sensor names are case sensitive. right now, it's all lower-case which is fine. The recommendation from https://github.com/Trikos/Home-Assistant-Naming-Convention is to use underscores, which I don't like.

proddy, 2024-07-13:
So I propose, for starters, to change to use camelCase and no spaces or special characters in:
- the API endpoint `/system/info`
- the MQTT topic `heartbeat`. This will also change the MQTT device ems-esp in HA, so old dashboards need refreshing. Code is in `Mqtt:ha_status()`
- the MQTT topic `info`
- change Web code as Michael mentioned to keep all RegEx consistent (max 20 length), comparing case-insensitive except for the special conditions with temperaturesensor, empty scheduler name etc...

tp1de, 2024-07-13:
@proddy please allow the following remarks / questions:

Whenever names are changed for mqtt (HA) the old entities became inactive and the old statistic records are kept. This is not nice and needs manual intervention next to adjusting dashboards.

Is there any proposed way how to do this with minimum effort? Or can this be done without manual intervention?
While doing the changes in steps. These exercise must be done more than once. I do not recommend. 

Within ioBroker I changed the code to recreate the object structure for /system/info on every instance restart.

proddy, 2024-07-13:
I agree we should avoid making any major changes to the HA historical data. With my plan above, only 3 sensor IDs will change in HA:  bus_status, uptime_sec, ntp_status

But we can decide not change that, and it would mean nothing changes in HA.

That leaves 

1. change the MQTT topics payload for `heartbeat` and `info`. This is purely cosmetic. 
2. continue with the initial plan to change the `/api/` endpoints, starting with `system/info` so it can be used in the new Scheduler conditions logic and the new Message command. For example today {"System Info"."uptime (seconds)"} is nsaty.

proddy, 2024-07-25:
I changed heartbeat and info, but it also impacted HA's entities, so I'll leave those two and only change system/info.

proddy, 2024-07-25:
@tp1de note system/info is changed so you'll need to adjust your ioBroker code accordingly

closed 2024-08-16 by proddy