emsesp/EMS-ESP32 #1867: New Bosch HP only recognizes CR10/CR11, CR10H/CR11H, CR20RF, RT800 as remote thermostats

christic73, 2024-07-16:
I have a new Bosch heat pump (compress 6800i AW) with a UI800 user interface. I also got a RT800 remote interface on top, which is currently not connected.
This HP apparently only accepts CR10/CR11, CR10H/CR11H, CR20RF, RT800 as remote thermostats. However I only can select RC100, RC100H, RC120RF or RC220 in EMS-ESP.

If I select RC100H in EMS-ESP and send remotetemp and remotehum periodically via scheduler, then the HP shows CR10H as remote thermostat, together with an error code 3111 (wrong remote configuration).

Is there anything I can do to solve the issue? Do you want me to share some logs of the RT800?

Thanks for looking into this :-)

[log.txt](https://github.com/user-attachments/files/16248789/log.txt)
[IMG_20240716_125457](https://github.com/user-attachments/assets/a3fd7cbe-4432-46b0-8753-3d4024d6614d)

(used EMS-ESP version 3.7.0-dev.25)

christic73, 2024-07-16:
I added a log of RT800 remote thermostat. Maybe this is of interest
[log_RT800.txt](https://github.com/user-attachments/files/16249357/log_RT800.txt)

MichaelDvP, 2024-07-16:
> I added a log of RT800 remote thermostat. Maybe this is of interest

Yes, this helps much. Could you please also upload the support info.

From the log i can see that:
- control is set to 6, so ems-esp shows RC220, we can change this text to RT800.
- version of RT800 is v21.03 with no subversion and  profduct-id 03
- temperature telegram is longer than known. Two bytes at the end read `01 09`. Don't know what this is, but we can trywith fixed numbers.
- humidity telegram is as know from RC100H with humidity and dewpoint.

I'll prepare a RT800 emulation to test...Takes a bit... 

BTW: You have a connect module 0x50, What is this for? We have seen such modules as RF base for RF-sensors and RF-thermostats, but no logs for this.

MichaelDvP, 2024-07-16:
Please test: https://github.com/MichaelDvP/EMS-ESP32/releases

christic73, 2024-07-16:
> Yes, this helps much. Could you please also upload the support info.
>
> From the log i can see that:
>
> * control is set to 6, so ems-esp shows RC220, we can change this text to RT800.
> * version of [...]

There is a module, which is inserted into the HP. On the manual not is called "K 30 RF". It delivers a "Gateway Module" (0x48) and a "Connect Module" (0x50)

[image]

christic73, 2024-07-16:
> There is a module, which is inserted into the HP. On the manual not is called "K 30 RF". It delivers a "Gateway Module" (0x48) and a "Connect Module" (0x50) [...]

I

> Yes, this helps much. Could you please also upload the support info.
>
> From the log i can see that:
>
> * control is set to 6, so ems-esp shows RC220, we can change this text to RT800.
> * version of [...]

I'm sorry, I'm new here... Could you tell me what do you mean with "support info"?

proddy, 2024-07-16:
there's a link in the box at the bottom of this page: https://emsesp.github.io/docs/Support/

christic73, 2024-07-16:
sorry 
[support.info.txt](https://github.com/user-attachments/files/16254987/support.info.txt)

christic73, 2024-07-16:
> Please test: https://github.com/MichaelDvP/EMS-ESP32/releases

thanks for the version!

- I successfully selected in "RT800" in EMS-ESP
- I successfully set a Temperature and Humidity value in EMS-ESP
- In the Bosch-HP User Interface I could see, that "RT800" has been set
- In the Bosch-HP User Interface I could see, that set temp and hum has been accepted from virtual thermostat
- The Bosch-HP didn't trough any error message (so far)

many thanks! you are awesome!!

closed 2024-07-17 by christic73