emsesp/EMS-ESP32 #1871: Device Naming Confusion

dleding, 2024-07-18:
### PROBLEM DESCRIPTION

There seems to be confusion about naming of several devices that are sold with different brandings e.g. Bosch an Buderus.

In my case there is an
Buderus GB172i.2 boiler
Buderus BC400 base thermostat (physically part of the boiler, plugged like a tablet behind the front, but separate device on bus)
Buderus RC220 room thermostat

The Boiler is identified with Buderus as manufacturer, but a generic type. The thermostats dont show an manufacturer name

```lua
  System information output here:
{"System":{"version":"3.7.0-dev.25","uptime":"000+00:31:42.117","uptime (seconds)":1902,"platform":"ESP32","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","free mem":153,"max alloc":87,"free caps":103,"used app":1679,"free app":305,"partition":"app0","reset reason":"RTC watch dog reset: CPU+RTC / APP CPU reset by PRO CPU"},"Network":{"network":"WiFi","hostname":"ems-esp","RSSI":-84,"BSSID":"set","TxPower setting":0,"static ip config":false,"low bandwidth":false,"disable sleep":false,"enable MDNS":true,"enable CORS":false,"AP provision mode":"disconnected","AP security":"wpa2","AP ssid":"ems-esp"},"NTP":{"NTP status":"connected","enabled":true,"server":"time.google.com","tz label":"Europe/Amsterdam"},"MQTT":{"MQTT status":"disconnected","enabled":false,"client id":"esp32-44a4ae30","keep alive":60,"clean session":false,"entity format":0,"base":"ems-esp","discovery prefix":"homeassistant","discovery type":0,"nested format":1,"ha enabled":false,"mqtt qos":0,"mqtt retain":false,"publish time heartbeat":60,"publish time boiler":10,"publish time thermostat":10,"publish time solar":10,"publish time mixer":10,"publish time water":10,"publish time other":10,"publish time sensor":10,"publish single":false,"publish2command":false,"send response":false},"Syslog":{"enabled":false},"Sensor":{},"API":{"API calls":0,"API fails":0},"Bus":{"bus status":"connected","bus protocol":"Buderus","bus telegrams received (rx)":3242,"bus reads (tx)":888,"bus writes (tx)":0,"bus incomplete telegrams":0,"bus reads failed":0,"bus writes failed":0,"bus rx line quality":100,"bus tx line quality":100},"Settings":{"board profile":"CUSTOM","locale":"de","tx mode":1,"ems bus id":11,"shower timer":false,"shower min duration":120,"shower alert":false,"phy type":0,"rx gpio":23,"tx gpio":22,"dallas gpio":0,"pbutton gpio":0,"led gpio":2,"hide led":false,"notoken api":false,"readonly mode":false,"fahrenheit":false,"dallas parasite":false,"bool format":1,"bool dashboard":1,"enum format":1,"analog enabled":false,"telnet enabled":true,"max web log buffer":100,"web log buffer":16},"Devices":[{"type":"boiler","name":"Condens 5000i/Greenstar 8000/GC9800IW/GB192i.2","device id":"0x08","product id":195,"version":"05.04","entities":87,"handlers received":"0xBF 0xC2 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B 0x04","handlers fetched":"0x14 0xE6 0xEA 0x28","handlers pending":"0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A","handlers ignored":"0xE7 0x02E0 0x02EA 0xD7 0xD8 0x02D6"},{"type":"thermostat","name":"BC400 Heizkreis","device id":"0x10","product id":4,"version":"49.04","entities":57,"handlers received":"0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267","handlers fetched":"0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240","handlers pending":"0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E 0x0269","handlers ignored":"0xC4 0xBF"},{"type":"thermostat","name":"RC220 Raum","device id":"0x38","product id":3,"version":"21.03","entities":3,"handlers received":"0x042B 0x047B","handlers pending":"0x0273 0x0A6A","handlers ignored":"0xBF"}]}

```

### TO REPRODUCE

Include an RC220 device in your system

### EXPECTED BEHAVIOUR

To show an RC220 as RC220.. :)

### SCREENSHOTS
[image]

proddy, 2024-07-19:
in EMS-ESP, we distinguish an EMS device by its unique deviceID and productID. Maybe there is a better way but that's always how the code was architected based on the knowledge we had a few years ago. We pull the information from a 'Version' telegram (0x02). If we're lucky the Brand (Buderus, Bosch, Worcester...) is included in that Telegram, but not always. We can only use the data we receive.

Could you use the Telnet Console and fetch for us 
```
read 8 2
read 10 2
read 38 2
```

maybe there is more information from those responses we can use.

...and as you've seen we recently added a rename function so you call the device anything you want now ;)

MichaelDvP, 2024-07-19:
This option was renamed in #1867. 
We don't have a RC220 in our database, but the RT800. Bosch renames often and use different names for their brands.
In #1867 it is reported that emulation of RC100 gives error when CR10H is set. So there is a difference in the brands, but we don't know how Bosch differs. I can set the option to `RC220/RT800`. It seems these are mostly identical. Your system info shows a RT800 and you have renamed it to RC220
RT800: product-id 3, version 21.03, now subversion, no brand (from telegram 02 in #1869 
Your RC220 seems the same version, as proddy mentioned read the 02 telegram if there are any subversions/brands in the telegram.

dleding, 2024-07-19:
> Could you use the Telnet Console and fetch for us
> ```
> read 8 2
> read 10 2
> read 38 2
> ```

of course:
```
ems-esp:$ read 8 2
000+16:52:35.242 N 1: [emsesp] boiler(0x08) -W-> Me(0x0B), Version(0x02), data: C3 05 04 EF 02 0E 00 00 00 03 01 14
ems-esp:$ 
ems-esp:$ read 10 2
000+16:52:49.197 N 2: [emsesp] thermostat(0x10) -W-> Me(0x0B), Version(0x02), data: 04 31 04 00 FF 00 00 00 00 00
ems-esp:$ 
ems-esp:$ read 38 2
000+16:53:03.117 N 3: [emsesp] thermostat(0x38) -W-> Me(0x0B), Version(0x02), data: 03 15 03 00 00
ems-esp:$ 
```

> 
> ...and as you've seen we recently added a rename function so you call the device anything you want now ;)

Right, it only starts looking "Wonky" when there is an option menu like shown where components of different manufacturers get mixed up in one selection. 

Up to now, i dont get how the distinction works in case of the remote selection menu of the UI800/BC400.
The Bosch UI800 shows the CRs and RTs, as a Buderus there are RCs instead. Does it get back to the "Kodierstecker" in the boiler controller, to tell the Buderus system not to act "like a bosch" and to use other names instead?

For my understanding, the only difference between Bosch RT800 and Buderus RC220 seems to be the colored front bezel.

proddy, 2024-07-19:
We guessed the Brand is stored at offset 9 in the Version(0x02) telegram, so your boiler has a value of 03, which means Buderus in our code. Obviously, we got the logic wrong, and there is an extra byte at the end with a value of 01. We'll need to collect more examples from users to decipher the brand accurately.

dleding, 2024-07-19:
> We guessed the Brand is stored at offset 9 in the Version(0x02) telegram, so your boiler has a value of 03, which means Buderus in our code. Obviously, we got the logic wrong..

Sorry, what is wrong there? The Boiler (and thermostats too) are indeed Buderus ones, and for the boiler that is shown correctly in the device Information tab.

Only the thermostats dont seem to indicate their branding, as for my naive understanding :)

By the Way: Thank you guys big times for this incredible piece of software! It works like a swiss clock so far and is a breeze to use.

MichaelDvP, 2024-07-19:
The thermostats do not report the brand. 
Your BC400 is product-id 4 (seen as BC400 and UI800 models), in #1867 with different options it was product-id 253 (seen as BC400, . UI800, rego3000, WSW196) Some rego3000 also comes with product-id 172 and different options. It's very difficult to differ the models. But it seems we need another flag for the 4/253 ids. 
@proddy i think we should reserve a flag range for the RC300 compatible, flags 32-63, so we can use a bitmask for `isRC300()` instead of many or.

proddy, 2024-07-24:
> The thermostats do not report the brand. Your BC400 is product-id 4 (seen as BC400 and UI800 models), in #1867 with different options it was product-id 253 (seen as BC400, . UI800, rego3000, WSW196) [...]

yes, we can add a new mask for these RC300 variations.

dleding, 2024-08-03:
*kisses* :)

closed 2024-08-03 by dleding