emsesp/EMS-ESP32 #1882: MQTT Discovery in HA throws an error after upgrade to 2024.7.3

zaphood1967, 2024-07-22:
### PROBLEM DESCRIPTION

MQTT Error is shown in the log. Not sure what it means, but all ESM-ESP Entities are offline in HA

### REQUESTED INFORMATION

```lua
2024-07-22 08:22:20.489 ERROR (MainThread) [homeassistant.components.mqtt.mixins] Error ''max' must be > 'min'' when processing MQTT discovery message topic: 'homeassistant/number/ems-esp/boiler_burnmaxpower/config', message: '{'object_id': 'boiler_burnmaxpower', 'icon': 'mdi:percent-outline', 'device': {'name': 'ems-esp Boiler', 'identifiers': ['ems-esp-boiler']}, 'unique_id': 'boiler_burnmaxpower', 'step': 1, 'min': 100, 'availability': [{'topic': 'ems-esp/boiler_data', 'value_template': "{{'online' if value_json.burnmaxpower is defined else 'offline'}}"}], 'value_template': '{{value_json.burnmaxpower if value_json.burnmaxpower is defined else 100}}', 'max': 100, 'name': 'Maximale Brennerleistung', 'mode': 'box', 'state_topic': 'ems-esp/boiler_data', 'availability_mode': 'all', 'command_topic': 'ems-esp/boiler/burnmaxpower', 'platform': 'mqtt'}'
```

### TO REPRODUCE

Error occurs since update to 2024.07.3

### EXPECTED BEHAVIOUR

MQTT should deliver the attributes again

zaphood1967, 2024-07-22:
All of a sudden, EMS-ESP had "forgotten" the MQTT configuration and switched it off as well. Weird. 
So after enabling MQTT again, all is fine again. Strange coincidence and I am not sure, why EMS-ESP has deleted the MQTT Configuration.

zaphood1967, 2024-07-22:
Fixed

proddy, 2024-07-22:
this is strange. I'm happy it is fixed but would still like to know what may have caused it.

 - has anything rebooted, like HA, or EMS-ESP or the MQTT broker?
 - did you upgrade anything (HA, EMS-ESP, MQTT) ?
 - was their a WiFi/Network reconnect?
 - to get it working did you just have to re-enable MQTT in EMS-ESP? Or were all the settings lost and you had to re-enter them?

zaphood1967, 2024-07-22:
Agree, this is really strange. I updated HA yesterday to 2024.7.3 and did not notice anything unusual. This morning, one value from the ESP was missing, so I rebooted HA, just to make sure it was not HA. After the reboot, all values of EMS went offline, so I checked the EMS website, but all looked good. I then rebooted EMS ESP as well, but nothing changed. After finally rebooting Mosquitto I filed the issue. 
Don't know why, but only then I checked the MQTT settings on the ESP and found MQTT to be disabled and the IP address of the server as well as the login credentials were gone. After keying in the data and enabling MQTT discovery for HA again, all worked as expected. No idea what made the ESP forget the data. There was no power outage, no WLAN issues and no other problems I would be aware of.

proddy, 2024-07-22:
Thanks for explaining. This has happened to me before as well, it's very odd and I really don't know what is causing it. Sometimes the MQTT settings are removed or not being loaded after a re-boot of EMS-ESP. It's mentioned before in the last 1-2 years, like in https://github.com/emsesp/EMS-ESP32/issues/1265#issuecomment-1711152212. I just don't know what is causing it.

This is one to keep track of. Thanks for reporting.

zaphood1967, 2024-07-22:
Just to clarify: I cannot tell for sure, if the MQTT settings had already been deleted BEFORE I rebooted the ESP, as I only checked afterwards. But I would assume they had been gone already beforehand, as there was no other reason on why HA did not receive any data from the ESP.

closed 2024-08-03 by proddy