emsesp/EMS-ESP32 #1884: Allow custom entity names with reference to hc1 .... hc4 within api

tp1de, 2024-07-22:
custom entity names like "hc1.abcd" can be defined and read by http get   .../api/custom.

Within ioBroker  the custom entity list is processed name by name ... e.g. .../api/custom/hc1.abcd .
This http request is not resolved as soon as a  "." or "_" is within custom name.

This needs to be corrected.

proddy, 2024-07-22:
we only allow alphanumeric in a name. spaces, . and _ should not be allowed

tp1de, 2024-07-22:
... we need a method to assign a circuit

MichaelDvP, 2024-07-23:
Actually we allow https://github.com/emsesp/EMS-ESP32/blob/9c20e93b3c49887b285da1e657190e0f0722b332/interface/src/project/validators.ts#L311

The underscore is part of temperaturesensor default name (Sensor-id), we should keep it, but the dot is critical, we allow it as seperator as shown in `show commands`, better disallow it in the name.

There is a bug in https://github.com/emsesp/EMS-ESP32/blob/9c20e93b3c49887b285da1e657190e0f0722b332/src/command.cpp#L233
Any circuit name at start of the command will be converted to id, independ of separator. 
So also `api/custom/hc1abcd` is not found.

I suggest: 
- remove dot form name validator, allow onlynames alphanumeric and underscore.
- check in `parse_command_string` for a valid separator, keep command unchanged for `hc1_abcd`

Then you can name a custom entiy to `hc1_abcd` and it is treated as name, not as circuit. Naming `hc1.abcd` or `hc1/abcd` are not allowed.
If this is ok, i can make the change.

tp1de, 2024-07-23:
@MichaelDvP  your suggestion sounds good for me

tp1de, 2024-07-23:
I still miss the possibility to create enum custom entities. Any chance to get this added as well?

MichaelDvP, 2024-07-23:
> Any chance to get this added as well?

Difficult, the enums have translated options as flash-strings, Have to find a way without complete refactoring.

tp1de, 2024-07-23:
> > Any chance to get this added as well?
> 
> Difficult, the enums have translated options as flash-strings, Have to find a way without complete refactoring.

custom elements do not need translations ... shall I open a new GH issue?

proddy, 2024-07-23:
> @MichaelDvP your suggestion sounds good for me

works for me too. Probably worth making a warning note somewhere that commands starting with 'hc' 'dhw' 'id' 'ahs' 'hs' are all reserved system words.

MichaelDvP, 2024-07-23:
Starting with hc1.. is not a problem, but what about name `hc1` 
**before:** `api/thermostat/hc1` returns the list as `api/thermostat/hc1/values`, but `hc1` as name does not work 
**now:** hc1 is also a valid name,  `api/thermostat/hc1` returns the value_info or `unknown command`, but not the list,  `api/thermostat/hc1/values`  still works with the list. Don't know if someone uses the short form of circuit list command and complains the change. (the complete list `api/thermostat` is not affected and still works as before)

closed 2024-07-25 by proddy