emsesp/EMS-ESP32 #1886: system send / response does not seem to work anymore correctly with mqtt and api

tp1de, 2024-07-23:
The raw command send and response logic seems to not work correctly anymore.
Testcase with telegram 01B9 (raw). I want to read offset 0x13 (1 byte)

**MQTT**

sending with offset 0: {"cmd":"send", "data":"0B 90 FF 00 20 01 B9"} 
data: "FF 2E 2C 3C 28 01 00 FF FF 0A 0A 01 FE 20 01 0F 05 00 00 01 22 FF 2E 2C 3C 28 01 00 FF FF 0A 0A 01 FE 20 01 0F 05 00 00 01 22"

sending with offset 13 length 1 byte: {"cmd":"send", "data":"0B 90 FF 13 01 01 B9"}
data: "01 FF 2E 2C 3C 28 01 00 FF FF 0A 0A 01 FE 20 01 0F 05 00 00 01 22"
*** offset works length not

**API**

same data but the api get result contains a "message": "cannot find data in response",
which is interpreted  as an error in the actual ioBroker adapter.

Do I have an error within the raw telegrams?

proddy, 2024-07-23:
yes, it's broke. I reproduced. None of the system commands work. I'll see if I can fix it.

```
ems-esp:# call system fetch
{
  "message": "cannot find data in fetch"
}
```

MichaelDvP, 2024-07-23:
Yes, the changes made for #1862 have a side effect, length is not processed correctly. 
I will test a bit how to solve.

The api message was fixed yesterday, i think.

MichaelDvP, 2024-07-23:
> yes, it's broke. I reproduced. None of the system commands work. I'll see if I can fix it.

yes, once again a https://github.com/emsesp/EMS-ESP32/blob/dfe85b9ba7b9044dc4c1658935142337fed1b53d/src/system.cpp#L1357

I had the output.clear() after cf found in command.cpp left in my build, so couldn't reproduce.

tp1de, 2024-07-23:
... and the combination of send / response does not give 100% accuracy. (this is new)
Same for custom entities. The values are not 100% stable. In approx 5-10% of times there are wrong / faulty values in custom entities and/or poll requests responses. (do they come from ems-bus or ems-esp coding?)

proddy, 2024-07-23:
I did some refactoring too to the command() and added some more tests for system commands like send so we can quickly find any breaking changes next time.

tp1de, 2024-07-23:
> Please test my build https://github.com/MichaelDvP/EMS-ESP32/releases if it is fixed.

First look is good.  API/MQTT send/response seems to work as well as custom entity name hc1_xxxx.
I will continue testing.

tp1de, 2024-07-23:
@MichaelDvP I recognized that I do not get the full telegram length with 20 bytes anymore. So switchTimes with 82 Bytes are truncated.

MichaelDvP, 2024-07-23:
> @MichaelDvP I recognized that I do not get the full telegram length with 20 bytes anymore. 

Can you explain with an example. Don't know what you mean. 
I can read the full telegram, but response contains <empty> if longer is requested. Here request 0x6A = 106 bytes:
```
ems-esp-s3:# call system send "00 90 49 00 6A"
000+00:03:54.088 I 15: [command] Calling command system/send (send a telegram) with value 00 90 49 00 6A
000+00:03:54.447 I 16: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0
000+00:03:54.576 I 17: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90
000+00:03:54.700 I 18: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7
000+00:03:54.801 I 19: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90
000+00:03:54.878 I 20: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90
ems-esp-s3:# call system response
{
  "src": "10",
  "dest": "49",
  "type": "49",
  "offset": "00",
  "data": "01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 00 00 00 10 05 18 18 05 18 01 01 00 01 01 00 <empty>"
```

tp1de, 2024-07-23:
Quite some time ago you implemented that when requesting 20 bytes a function to collect the different response telegrams into on one response for mqtt/api. For mqtt response this still works, for api response not.

MichaelDvP, 2024-07-23:
Ah, yes that was a patch. You can't request 32 (0x20) bytes this way. Just use FF instead to get the whole telegram. or 0x54 to request 84 bytes.

tp1de, 2024-07-23:
Thanks ... this works. Not so easy to keep track about all this changes

MichaelDvP, 2024-07-24:
> (actually I send a double request with 300 msec in between

I have to check, i think a second request while the first is still collecting data can lead to strange things.

MichaelDvP, 2024-07-24:
Sometimes there is something slow on the bus, depends on the bus-devices and telegrams. 
I've logged the reading and merging of the response and checked the times:
1. try: 2.5 sec
2. try: 9,05 sec
3. try: 0.64 sec

In 2. try it was only 0.7 sec for the first 3 parts of the telegram, the last byte came in >8 sec later. But this is bus timing and we can't do anything.
```
ems-esp-s3:# call system send "00 90 49 00 ff"
000+00:00:55.887 I 1: [command] Calling command system/send (send a telegram) with value 00 90 49 00 ff
000+00:00:56.353 I 2: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0
000+00:00:56.480 I 3: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90
000+00:00:56.605 I 4: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7
000+00:00:58.456 I 5: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90
ems-esp-s3:# call system send "00 90 49 00 ff"
000+00:01:59.273 I 6: [command] Calling command system/send (send a telegram) with value 00 90 49 00 ff
000+00:01:59.726 I 7: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0
000+00:01:59.855 I 8: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90
000+00:01:59.980 I 9: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7
000+00:02:08.331 I 10: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90
ems-esp-s3:# call system send "00 90 49 00 ff"
000+00:02:26.102 I 11: [command] Calling command system/send (send a telegram) with value 00 90 49 00 ff
000+00:02:26.352 I 12: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0
000+00:02:26.479 I 13: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90
000+00:02:26.605 I 14: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7
000+00:02:26.706 I 15: [emsesp] response: 01 23 00 81 21 23 20 81 41 23 40 81 61 23 60 81 81 23 80 81 A1 26 A0 81 C1 2A C0 81 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90

```

MichaelDvP, 2024-07-24:
If it helps: i can clear the last response if a new is in progress. Then you can just poll and get empty until the telegram is completed.

tp1de, 2024-07-24:
> If it helps: i can clear the last response if a new is in progress. Then you can just poll and get empty until the telegram is completed.

This is not necessary since I check response type. I will continue to test tomorrow.

tp1de, 2024-07-25:
@MichaelDvP 
Response to telegram with offset and one byte length does not work correctly:
0B 90 FF 13 01 01 B9 --> {src: "10",dest: "0B",type: "02B9",offset: "13",data: "02 02",value: 2}
data has 2 bytes  double entry - why?

In other cases (20-30%) the response data is starting from offset 0x13 but length is up to end of telegram:
data: 02 FF 2E 2C 3C 28 01 00 FF FF 0A 0A 01 FE 20 01 0F 05 00 00 02 22
but data position 0x14 is missing --> see complete telegram

complete telegram: 
data :FF 2E 2C 3C 28 01 00 FF FF 0A 0A 01 FE 20 01 0F 05 00 00 02 22 FF 2E 2C 3C 28 01 00 FF FF 0A 0A 01 FE 20 01 0F 05 00 00 02 22

tp1de, 2024-07-25:
Wenn I request a non existing telegram type in between the data is correct. Clearing data with new request should work.

MichaelDvP, 2024-07-25:
Stop testing with this version, i'll make some changes that should also work with the boiler 0xeA with holes (#1862).
Have your boiler 0xEA? Is it complete shown?

tp1de, 2024-07-25:
> Have your boiler 0xEA? Is it complete shown?

Yes ist has: 
data: 08 0B EA 00 00 00 00 00 00 01 32 F8 00 14 01 02 41 D8 00 B4 32 3C 2A 0A 3C 64 64 00 00 01 45

MichaelDvP, 2024-07-25:
Seems to be complete starting from offset 0. In #1862 reading 0xea only gives offset 5-18 and we need a second read with offset > 18 and get only offset 27 back. For response i need to insert the gapped bytes as zero. 

Please try the response with my new build. A single request should, but multi requests are ok. Response is cleared while collecting the data from the bus.

tp1de, 2024-07-25:
I tried your new build. First impression is good. Data seems to be ok now for existing telegram types.
When trying types which are not existing the data format is now different:

old version: "data": "<empty>"
new version: "data": 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

The old version for empty data is much better.

MichaelDvP, 2024-07-25:
Do you prefere ` "data": "<empty>"` or ` "data": ""`?

tp1de, 2024-07-25:
> Do you prefere ` "data": "<empty>"` or ` "data": ""`?

old version was {"src":"10","dest":"0B","type":"02C6","offset":"00","data":"<empty>","value":0}

Better to keep the old version. (within ioBroker I test both versions)

MichaelDvP, 2024-07-25:
Ok, updated. For empty it is now "" and there is no value calculation now, as for and length > 4. 
Read your edit to late, so it's "", but can change to `<empty>` if needed.
You should get `response:"{"src":"10"..., "data":""}"` for a empty reply and `response:""` for not yet received.

tp1de, 2024-07-26:
I still struggle with send / response  timing. 

I implemented a poll cycle every 2 minutes. Actually I am testing with 6 telegram types. (4 are switchTimes)
Within a polling cycle I send requests with axios POST, and try to read after one second.
Trying to read every second for 40 seconds duration with 1 second delay between the axios GET reads. 

Let me share my findings so far:

* In most cases  response is resolved within 1 second (type requested is the same as type responded and data is available)

* In some cases response type is different then requested (with data) -  I try to read every second until response type is right

* In other cases response is "" - this should happen only when type is not available. After 10 seconds I send a 2nd request - this works. Same for cases where response has still wrong type after 10 seconds. Without second POST the response will be not fulfilled.

* I do recognize some (very seldom) axios post errors:  ECONNRESET

The average polling time for the 6 telegrams are between 6.5 and 30 seconds. (average 12-15 seconds).
I will continue testing.

MichaelDvP, 2024-07-26:
Let me explain the timing:
1. on command `api/system/send` the telegram is placed in front of the sending queue
2. waiting for the busmaster to allow us to send
3. send out first part of request, clear response data
4. on receiving the bus answer store the telegram part in a buffer and request next part, repeat 4. until complete
5. build the json for response-data, publish to mqtt

I'm not sure if it is better to clear the response data in step 3 or 1?
If you repeat a send  command the new message is queued and begin with step 1 immediate after step 5, 
You can poll `system/response` all the time, acutally it return the old result in step 1-2, empty message in step 3-5 and new result after step 5.

tp1de, 2024-07-26:
I do understand the logic and I can process the outcome.

I just ask myself if I should accept that sometimes I do not get any valid reply with data from response? 
(I am waiting 50 seconds now)
And shall I repeat the request after 10 seconds - or not? ... and how long shall I wait? (is 50 seconds to long?)
The total polling time should be significant lower than polling interval (2 minutes)

My first version some months ago was to use syslog server within ioBroker and collecting data from there. Just regular polling requests where needed for some telegram types. This required all telegrams in hex to be sent from ems-esp and to analysed. This created quite heavy CPU-load for the ioBroker instance.  So I went to send / response logic.

_I am working on switchPrograms for my RC310 EMS+ thermostat. I found out that when changing the switchProgramType from levels (eco/comfort) to absolute (absolute temps) different telegram types are used. (same structure).
KM200 puts them into same api entity, content depending on switchProgramType per hc.
From ems-esp telegram-type point of view, I would need to define 4 switchPrograms per hc. A and B for levels and absolute._

tp1de, 2024-07-26:
Without 2nd send I recognized that reading response errors correspond to system log entries:
2024-07-26 15:40:44.720 E 2116: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 10 FF 00 01 C3 FF
2024-07-26 15:41:12.867 E 2117: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 10 FF 00 02 09 FF

MichaelDvP, 2024-07-26:
Oh, in this case the the respose_id is not cleared i have to stop waiting for the rest of telegrams.

proddy, 2024-08-03:
@tp1de close this?

closed 2024-08-03 by tp1de