emsesp/EMS-ESP32 #1887: different information of mm100 modules in EMS network

Seziksezik, 2024-07-23:
2.65 software version. I have 2 mm100 in network. 

But can see onlyb4 parameters inside first module. And 7 inside second. They are almost the same.

[image]
[image]
[image]

### PROBLEM DESCRIPTION

_A clear and concise description of what the problem is._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-07-23:
I'm not sure which version you're using, but it looks very old. Try and upgrade to the latest 3.7.0. Also please provide your system info.

Seziksezik, 2024-07-23:
There. 

Yahoo Почта: Поиск, организация, успех

proddy, 2024-07-23:
Upgrade and try again.

closed 2024-07-23 by proddy

Seziksezik, 2024-07-23:
3.7 is ok..Both are 7 items..

Yahoo Почта: Поиск, организация, успех