emsesp/EMS-ESP32 #1890: Keenetic cloud authorised web access not working

Seziksezik, 2024-07-24:
Problem with authorized access trough keenetic cloud.

Keenetic is.... a  former part Zyxell 
We have there big keenetic routers installed base. 

This company is very close to Zyxell..

So... they have a cloud....For example. I.can access to my home network with easy address. 

http://ems.blablabla.keenetic.pro

this is service description:

https://help.keenetic.com/hc/en-us/articles/360008818279-Keenetic-Cloud-web-portal

Difficult repeat..Need access to Keenetic cloud. But I can add read only account to check it. 

Login/password are default..

May be will be the same problem with ddns service or other who support user authorisation for remote access. 

It works directly from my browser without any authorisation. But.... If I try set up login/password based access... everything not working. 

I can see only login page :) Enter  login/password for.the cloud and see an error message..

I think there are no delay somewhere. See attached screen shots. 

[image]
[image]
[image]

proddy, 2024-07-25:
turn on the browsers dev console (F12) and see what errors are reporting after you click login. It will help us pin point the problem. You may also need to enable CORS in ems-esp.

Seziksezik, 2024-07-25:
A question. How to enable cors?:) 

Look my video on Google Chrome. 
I don't have enough time to enter password. Windows closed in 10 seconds around. Yes, I have access but see almost nothing. 

And F12 screen photo. 

I used first time  a yandex browser. From my smartphone. Don't know how to press F12 there. 

3d small screen copy is Chrone from my Smartphone. I enter the password then see network timeout. 

Without password protection it works. 

https://github.com/user-attachments/assets/59b9a8ee-2884-4861-b84b-c41b578e7aa1

@
[image]

Seziksezik, 2024-07-25:
[image]

3d file. Smartphone. Google Chrome. Password entered correctly.

proddy, 2024-07-25:
- what do you mean by "without password protection, it works"? Is this something on your VPN
- CORS can be enabled in EMS-ESPs Network Settings
- When you're in F12, go the Network tab and then CTRL-R. Then we can see all the traffic. Something is being blocked.

Seziksezik, 2024-07-25:
look. I have two option when access to remote site. With password and without password. If without password - it works. 

But if I switch on password protection something not working. See details.. @

https://github.com/user-attachments/assets/08c21b1a-6625-4168-bf06-8b1b548d50b7

[image]
[image]
[image]

proddy, 2024-07-25:
click on one of those red signIn errors (where it says cancelled). then in the tab Headers and also Response show me what you see. Like:

[image]

proddy, 2024-07-25:
also do you have examples of other ESP based applications where it does work, like Tasmota, ESPHome ? I'm thinking its either because we use HTTP and not HTTP/S or there's an HTTP header missing in EMS-ESP that you need to authorize the client

Seziksezik, 2024-07-25:
see attached. 

Yes. Cloud use https protocol. 

[image]

[image]

Sorry. I dont have  Tasmota or ESP Home  Just Tuya :) 

I can access remote to the device, but need start SSTP client, then put IP address. Not a problem. But looks like it is not compatible with keenetic cloud. urls based access is more easy.

Seziksezik, 2024-07-25:
So strange. 

I can enter inside troughbthe cloud when use the link lice... https://ems.sezik.keenetic.pro/devices

Then enter cloud login/password, device login password... and see something. 

[image]
[image]

See attached picture.

proddy, 2024-07-25:
I can see non of the backend API calls are working (they all show as red and cancelled). I don't know what is causing it, sorry.

Seziksezik, 2024-07-25:
ok. no problem. 
[image]
Anyway thank you for your support. I think it is cloud problem somewhere. 

See direct link access.

proddy, 2024-07-26:
Okay. I have EMS-ESP running in a VPN/DDNS in Cloudflare and it works, but nothing as sophisticated as Keenetic. I'll turn this into a discussion.