emsesp/EMS-ESP32 #1892: Update from 3.6.0 to 3.6.5 through web interface fails

florian-reus, 2024-07-25:
### PROBLEM DESCRIPTION

update from 3,6.0 to 3.6.5 through web interface not possible
log is not showing any cause.
- is updating to -dev mandatory?

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

I searched, and found nothing similar.

- [ ] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [ ] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the output of http://ems-esp.local/api/system :
{
  "System Info": {
    "version": "3.6.0",
    "platform": "ESP32",
    "uptime": "000+00:02:05.046",
    "uptime (seconds)": 125,
    "free mem": 121,
    "max alloc": 67,
    "free app": 8,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Info": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -78,
    "IPv4 address": "192.168.9.241/255.255.255.0",
    "IPv4 gateway": "192.168.9.1",
    "IPv4 nameserver": "192.168.9.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": true,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Info": {
    "NTP status": "connected",
    "enabled": true,
    "server": "de.pool.ntp.org",
    "tz label": "Europe/Berlin"
  },
  "OTA Info": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Info": {
    "MQTT status": "connected",
    "MQTT publishes": 223,
    "MQTT queued": 0,
    "MQTT publish fails": 0,
    "MQTT connects": 1,
    "enabled": true,
    "client id": "ems-esp",
    "keep alive": 10,
    "clean session": true,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Info": {
    "enabled": false
  },
  "Sensor Info": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Info": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Info": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 291,
    "bus reads (tx)": 89,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": true,
    "shower alert": true,
    "shower alert coldshot": 10,
    "shower alert trigger": 7,
    "hide led": true,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true,
    "max web log buffer": 100,
    "web log buffer": 20
  },
  "Devices": [
    {
      "type": "boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si",
      "device id": "0x08",
      "product id": 123,
      "version": "07.00",
      "entities": 68,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33 0x26",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
      "device id": "0x10",
      "product id": 158,
      "version": "74.01",
      "entities": 81,
      "handlers received": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
      "handlers ignored": "0x35 0x02E1 0x02E2 0x02EB 0x02EC"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x20",
      "product id": 160,
      "version": "24.05",
      "entities": 3,
      "handlers received": "0x02D7",
      "handlers ignored": "0x1E 0x1A 0x0255"
    },
    {
      "type": "mixer",
      "name": "MM100",
      "device id": "0x21",
      "product id": 160,
      "version": "24.05",
      "entities": 4,
      "handlers received": "0x02D8",
      "handlers ignored": "0x1A 0x0255"
    },
    {
      "type": "controller",
      "name": "BC25",
      "device id": "0x09",
      "product id": 125,
      "version": "03.05",
      "entities": 0
    }
  ]
}

```lua
  System information output here:
- how?

```

### TO REPRODUCE

running 3.6.0
downloaded 3.6.5 from web
uploaded to ems-esp through web interface

### Log
Log Level
Max Buffer Size
Compact
2024-07-25 14:14:40.000    INFO 1:      [emsesp]     Starting EMS-ESP version 3.6.0
2024-07-25 14:14:40.000    INFO 2:      [emsesp]     Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2024-07-25 14:14:40.000    INFO 3:      [emsesp]     Loading board profile S32
2024-07-25 14:14:40.000  NOTICE 4:      [system]     Downgrading to version 3.6.0-
2024-07-25 14:14:40.000    INFO 5:      [emsesp]     Starting Scheduler service
2024-07-25 14:14:40.000    INFO 6:      [emsesp]     Starting Custom entity service
2024-07-25 14:14:40.000    INFO 8:      [emsesp]     Loaded EMS device library (116 records)
2024-07-25 14:14:40.000    INFO 9:      [emsesp]     Starting Access Point with captive portal on 192.168.4.1
2024-07-25 14:14:43.852    INFO 10:     [emsesp]     Starting NTP service
2024-07-25 14:14:43.852    INFO 11:     [emsesp]     WiFi connected with IP=192.168.9.241, hostname=ems-esp
2024-07-25 14:14:43.859    INFO 12:     [emsesp]     mDNS responder service started
2024-07-25 14:14:46.994    INFO 14:     [system]     NTP connected
2024-07-25 14:14:50.001    INFO 15:     [emsesp]     Stopping Access Point
2024-07-25 14:14:56.745    INFO 16:     [emsesp]     Recognized new boiler with deviceID 0x08
2024-07-25 14:14:57.617    INFO 17:     [emsesp]     Recognized new controller with deviceID 0x09
2024-07-25 14:14:58.228    INFO 18:     [emsesp]     Recognized new thermostat with deviceID 0x10
2024-07-25 14:14:58.818    INFO 19:     [emsesp]     Recognized new mixer with deviceID 0x20
2024-07-25 14:15:00.441    INFO 20:     [emsesp]     Recognized new mixer with deviceID 0x21

### EXPECTED BEHAVIOUR

update through web interface is working
(provide log why update was failing)

**(Please, remember to close the issue when the problem has been addressed)**

proddy, 2024-07-25:
I suspect it's because starting with 3.6.x we changed the partitions on the firmware. See https://emsesp.github.io/docs/#breaking-changes-since-v36x

3.5.0 is a year old, and there have been many changes. I would export the settings (or write down the important ones), then use the flash tool to go straight to the dev version 3.7.0.

florian-reus, 2024-07-25:
I already used the flash tool back then when updating to 3.6.0
Now there shouldn't be another change!?

proddy, 2024-07-25:
Then it should work. Your log shows it's version 3.6.0. What exactly isn't working?

florian-reus, 2024-07-25:
I don't know why, but it's restarting after upload the image through the web interface. The log only shows "reverting to 3.6.0"
Mit freundlichen Grüßen,
Florian Reus

On 25 July 2024 17:23:02 CEST, Proddy ***@***.***> wrote:
>Then it should work. Your log shows it's version 3.6.0. What exactly isn't working?
>

proddy, 2024-07-25:
okay, let me think about this.

florian-reus, 2024-07-25:
Anything I should try (dev version)?
Any further logs helpful?
Mit freundlichen Grüßen,
Florian Reus

On 25 July 2024 18:09:47 CEST, Proddy ***@***.***> wrote:
>okay, let me think about this.
>

proddy, 2024-07-25:
only thing I would suggest is make an export of your settings, use the latest Flash tool to erase the ESP32 and upgrade to 3.7.0. If that didn't work then I suspect its a hardware issue

florian-reus, 2024-07-29:
worked last time, so I think it will work again.
It's a more work to get it out of the installation, so I will leave it till I really need to update (as this will also not help to find any cause for the update problem)

florian-reus, 2024-07-29:
Tested more, and found out: 
Update 3.6.0 - 3.6.5 -- not working
Update 3.6.1 - 3.6.5 -- not working
Update 3.6.2 - 3.6.5 -- working

So In this case intermediate steps were needed. 
Update fast, update often!
Mit freundlichen Grüßen,
Florian Reus

On 25 July 2024 20:55:40 CEST, Proddy ***@***.***> wrote:
>only thing I would suggest is make an export of your settings, use the latest Flash tool to erase the ESP32 and upgrade to 3.7.0. If that didn't work then I suspect its a hardware issue
>

proddy, 2024-07-29:
thanks for testing, it's good to know for others. Sadly there's not a lot we can do to fix older versions. I'll move this to a GitHb discussion.