emsesp/EMS-ESP32 #1893: call schedule entries via command

proddy, 2024-07-25:
From an earlier discussion:

I'm thinking, @MichaelDvP, shall we extend the Scheduler so you can execute the event? Even if its disabled. That way you could create quite complex commands and conditions, keep it disabled and invoke it manually via the API.

_Originally posted by @proddy in https://github.com/emsesp/EMS-ESP32/discussions/1835#discussioncomment-10148907_

MichaelDvP, 2024-07-26:
Some more thoughts about that:
- A single action command does not have a schedule trigger, so only show command, value, and name  if no trigger is selected:
[image]
- The Execute button make it active. In the list the schedule and tme field is empty.
[image]
[image]
- You can call a command from mqtt/api/terminal to activate and the schedule-command is executed and deactivated again:
```
ems-esp-s3:# call scheduler info
{
  "webtest": 0,
  "singleshot": 0
}
ems-esp-s3:# call scheduler singleshot
{
  "name": "singleshot",
  "type": "string",
  "active": 0,
  "command": "custom/test",
  "value": "custom/test + 3",
  "readable": true,
  "writeable": true,
  "visible": true
}
ems-esp-s3:# call scheduler singleshot on
Command executed. Check log for messages.
000+00:01:11.485 I 1: [command] Called command scheduler/singleshot (enable schedule item) with value on
000+00:01:11.485 I 2: [command] Called command custom/test (set custom value on ems) with value 8
ems-esp-s3:# call scheduler singleshot
{
  "name": "singleshot",
  "type": "string",
  "active": 0,
  "command": "custom/test",
  "value": "custom/test + 3",
  "readable": true,
  "writeable": true,
  "visible": true
}
ems-esp-s3:#
```
( BTW: i still don't like the `Command executed. Check log for messages.` message in console)

This is easy to implement, it only needs in scheduler::loop
```
    for (ScheduleItem & scheduleItem : *scheduleItems_) {
        if (scheduleItem.active && scheduleItem.flags == 0) {
            command(scheduleItem.name.c_str(), scheduleItem.cmd, compute(scheduleItem.value));
            scheduleItem.active = false;
        }
    }
```

proddy, 2024-07-26:
That would work nicely, just checking for flags = 0.  And I'll think about how we can make the WebUI more intuitive so you knows it's an "on command" type.

Please feel free to change or remove the "Command executed" console message.

proddy, 2024-07-27:
@MichaelDvP I've made some small changes to the Web Scheduler, I'll do PR so you can test.

proddy, 2024-07-27:
There's a demo online at https://ems-esp.derbyshire.nl/scheduler - let me know what you think

proddy, 2024-08-03:
I will add the translations to the web and then we can close this new feature

closed 2024-08-05 by proddy