emsesp/EMS-ESP32 #1894: Modbus Issue Writing to Heating Temperature

wint100, 2024-07-26:
Following on from the Modbus support thread, before it was closed, there is still an issue with writing to the Heating Temperature on the latest Dev build. The error log fills up as below. It does not affect the new S3 boards, only the older ESP32-D0WD-V3

2024-07-26 06:25:43.789 E 916: [modbus] Modbus write command failed with error: unrecognized path (Error)
2024-07-26 06:25:53.901 E 917: [modbus] Modbus write command failed with error: unrecognized path (Error)
2024-07-26 06:26:03.995 E 918: [modbus] Modbus write command failed with error: unrecognized path (Error)
2024-07-26 06:26:14.085 E 919: [modbus] Modbus write command failed with error: unrecognized path (Error)
2024-07-26 06:26:24.191 E 920: [modbus] Modbus write command failed with error: unrecognized path (Error)
2024-07-26 06:26:34.307 E 921: [modbus] Modbus write command failed with error: unrecognized path (Error)
2024-07-26 06:26:44.410 E 922: [modbus] Modbus write command failed with error: unrecognized path (Error)
2024-07-26 06:26:54.511 E 923: [modbus] Modbus write command failed with error: unrecognized path (Error)

proddy, 2024-07-26:
Which version of the E32 are you using? The V2 has 8MB of PSRAM (and 16MB Flash) which is needed for modbus. In fact I think we should disable the option for enabling Modbus on 4MB variants.

But it doesn't explain the error you are getting. I think I can make that log message more intuitive, so we'll start there.

proddy, 2024-07-26:
The latest dev version has more info the logs, if you could try that out

bbqkees, 2024-08-13:
I believe it's the S32 @wint100 is referring to and not the E32 V1.5.
The S32 has 16MB flash but very limited RAM (No PSRAM), maybe it's a memory issue.
The S3 and E32 V2 have 8MB of PSRAM which allows for much more features to be active.

proddy, 2024-08-13:
closing this, please feel free to re-open @wint100 when you have more details.

closed 2024-08-13 by proddy