emsesp/EMS-ESP32 #1897: Device specific commands

MichaelDvP, 2024-07-26:
There are some device specific commands that react different on devicetypes. I'm  thinking how to make a clear line:
- `api/devices/commands`
  is in line for all devices
- `api/device/values`

proddy, 2024-07-26:
I like it, the code does need a clean-up. My initial thoughts

- `<device>/command` - yes it should be consistent. There's a lot of repeated code, like manually adding values, info etc which I don't like. 
- Create a function that tests for `info` and `value` and other special commands. Now we have duplicate code doing string comparisons.
- add `system/values` as it is indeed missing
- `entities` I can't remember why I added it, but we don't need it now that we generate these using `dump_entites.sh`. I think we can comment out all that code and remove the command.
- JsonArray looks nicer, but JsonObject is easier to traverse and find objects in the JSON tree. E.g. just pick the node `custom[mytemp]` instead of looping through doing a string comparison on the `name` key.
- Decide if we want to nest or not, as there is a difference between `info` and `values`. For example

`ems-esp:# call thermostat info`
```json
{
  "hc1 selected room temperature (seltemp)": 23.0,
  "hc1 current room temperature (currtemp)": 20.7,
  "hc1 mqtt discovery current room temperature (haclimate)": "roomTemp",
  "hc1 mode type (modetype)": "comfort",
  "hc1 target flow temperature (targetflowtemp)": 0,
  "hc1 summer mode (summermode)": "winter",
  "hc1 room temperature from remote (remotetemp)": null,
  "hc1 room humidity from remote (remotehum)": null
}
```
`ems-esp:# call thermostat values`
```json
{
  "hc1": {
    "seltemp": 23.0,
    "currtemp": 20.7,
    "haclimate": "roomTemp",
    "modetype": "comfort",
    "targetflowtemp": 0,
    "summermode": "winter",
    "remotetemp": null,
    "remotehum": null
  },
  "dhw": {}
}
```

I would say nest, so the only different between values and info would be whether its the shortname or translated longname.

MichaelDvP, 2024-07-26:
> Decide if we want to nest or not, as there is a difference between info and values. For example

Didn't remember this, but i think this a point for users don't like nested. We can use the same scheme for other devices. 
Custom/scheduler/analog/temperature does not have nests, so info/value stays the same, but for system we have then
system/info: `{"network rssi": -55}` and system/values gives `{"network": {"rssi": -55}}`
Then the default for `api/system` would also point to /values. So no change in response for @tp1de ioBroker

proddy, 2024-07-26:
ok, makes sense.

I also think we can create a single function to find the attribute_s as this code is also repeated

MichaelDvP, 2024-07-29:
I'll make a PR with some refactoring.

proddy, 2024-08-05:
All done I think. The new tests pass. I'll close this one.

closed 2024-08-05 by proddy