emsesp/EMS-ESP32 #1904: Buderus BC400: HK1 Summer Mode not available

Hank2419, 2024-07-29:
### PROBLEM DESCRIPTION

The entity “HK1 Summer Mode” ist visible in Home Assistant but is constantly shown as “unavailable”

### REQUESTED INFORMATION

- [ ] Provide the output of http://ems-esp.local/api/system :

```lua
  System information output here:

```
Will be provided later if needed

### TO REPRODUCE

The error is constantly visible if entity is displayed in HA

### EXPECTED BEHAVIOUR

The entity should be available with a valid status

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT

None

proddy, 2024-07-29:
please attach the support info

proddy, 2024-08-03:
and see what value is shown in the EMS-ESP - either on the WebUI or via the Console with `show values` or an API call like `/api/thermostat/info`

Hank2419, 2024-08-03:
> please attach the support info

I have added this info above

> and see what value is shown in the EMS-ESP - either on the WebUI or via the Console with `show values` or an API call like `/api/thermostat/info`

{
    "name": "summermode",
    "fullname": "HK1 Sommerbetrieb",
    "circuit": "hc1",
    "type": "enum",
    "enum": [
        "Winter",
        "Sommer"
    ],
    "readable": true,
    "writeable": false,
    "visible": true,
    "value": "not set"
}

[image]

proddy, 2024-08-03:
There is no value (it says not set) so its not sent to HA.  Modes for BC400 was recently added. i would upgrade to the latest dev, try again and if it still doesn't work then help us find the right value in the telegram - https://emsesp.github.io/docs/Troubleshooting/#im-missing-certain-data-from-an-ems-device

Hank2419, 2024-08-03:
> There is no value (it says not set) so its not sent to HA. Modes for BC400 was recently added. i would upgrade to the latest dev, try again and if it still doesn't work then help us find the right [...]

I’m on esp firmware version 3.7.0-dev.23 (ESP32-S3)

Since I have installed ESP the summer mode in the BC400 has never changed and remains on “Sommer”. Next change to “Winter” will be next autumn when temperature is below 15°. Could that be the reason that the value is not transmitted?

Hank2419, 2024-08-03:
Another observation: during restart the different entities are changing their status. First Sommer mode is valid and other entities are unavailable. At last when “Arbeitsweise” turns available at the very same moment “summer mode” turns unavailable. Is this coincidence?

[image]

MichaelDvP, 2024-08-04:
See #550 and #566, heatpumps does not show summermode, they show operatingmode instead. Since more than 2 years now.

Hank2419, 2024-08-04:
> See #550 and #566, heatpumps does not show summermode, they show operatingmode instead. Since more than 2 years now.

In my case there is a different behavior of the signal. It is the only signal from the Buderus heat pump that is unavailable. All other approx 400 signals are available (thereof many with operating modes) and seem to work as the should. Also I found out, that during restart process of the ESP Gateway for a couple of minutes the values is correctly displayed in HA (see screenshot above) before it turns back to “unavailable”. However the signal is for me only “nice to have” and I fully depend on experts to solve the issue. Thanks to all who provide support!