emsesp/EMS-ESP32 #1909: BBQKees Gateway S32 crashes after update to EMS-ESP-3_7_0-dev_27-ESP32_16M or later

Andi252, 2024-08-01:
### PROBLEM DESCRIPTION

BBQKees Gateway S32 crashes and does not restart after update to EMS-ESP-3_7_0-dev_27-ESP32_16M - EMS-ESP-3_7_0-dev_29-ESP32_16M

The version EMS-ESP-3_7_0-dev_26-ESP32 is the last which works.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the problem in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the problem in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the problem in the [docs](https://emsesp.github.io/docs/Troubleshooting/)
- [x] Searched the problem in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the output of http://ems-esp.local/api/system :

## Last good version
```lua
{
  "System": {
    "version": "3.7.0-dev.26",
    "uptime": "000+00:01:44.224",
    "uptime (seconds)": 104,
    "platform": "ESP32",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "free mem": 180,
    "max alloc": 107,
    "free caps": 130,
    "used app": 1690,
    "free app": 870,
    "partition": "boot",
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network": {
    "TxPower setting": 0,
    "static ip config": false,
    "low bandwidth": false,
    "disable sleep": false,
    "enable MDNS": true,
    "enable CORS": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP": {
    "NTP status": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "MQTT": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client id": "esp32-7d55b594",
    "keep alive": 60,
    "clean session": false,
    "entity format": 0,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "discovery type": 0,
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time heartbeat": 60,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
[... 191 lines cut ...]
[21:30:04]
[21:30:04]E (725) esp_littlefs: mount failed,  (-84)
[21:30:04]E (729) esp_littlefs: Failed to initialize LittleFS
[21:30:04][   196][E][vfs_api.cpp:105] open(): /littlefs/config does not exist, no permits for creation
[21:30:04]E (918) wifi_init_default: netstack cb reg failed with 12289
[21:30:04][   301][E][vfs_api.cpp:105] open(): /littlefs/config/networkSettings.json does not exist, no permits for creation
[21:30:04][   316][E][vfs_api.cpp:105] open(): /littlefs/config does not exist, no permits for creation
[21:30:05][   388][E][vfs_api.cpp:105] open(): /littlefs/config/apSettings.json does not exist, no permits for creation
[21:30:05][   419][E][vfs_api.cpp:105] open(): /littlefs/config/ntpSettings.json does not exist, no permits for creation
[21:30:05][   452][E][vfs_api.cpp:105] open(): /littlefs/config/mqttSettings.json does not exist, no permits for creation
[21:30:05][   536][E][vfs_api.cpp:105] open(): /littlefs/config/securitySettings.json does not exist, no permits for creation
[21:30:05][   658][E][vfs_api.cpp:105] open(): /littlefs/tmp_upload does not exist, no permits for creation
[21:30:05][   671][E][vfs_api.cpp:105] open(): /littlefs/config/emsespSettings.json does not exist, no permits for creation
[21:30:05][   682][E][Preferences.cpp:483] getString(): nvs_get_str len fail: boot NOT_FOUND
[21:30:05][   772][E][vfs_api.cpp:105] open(): /littlefs/config/emsespCustomization.json does not exist, no permits for creation
[21:30:05][   813][E][vfs_api.cpp:105] open(): /littlefs/config/emsespScheduler.json does not exist, no permits for creation
[21:30:05][   855][E][vfs_api.cpp:105] open(): /littlefs/config/emsespEntity.json does not exist, no permits for creation
[21:30:06]E (2167) lan87xx: lan87xx_pwrctl(409): power up timeout
[21:30:06]E (2167) lan87xx: lan87xx_init(491): power control failed
[21:30:06]E (2167) esp_eth: esp_eth_driver_install(215): init phy failed
[21:30:06][  1551][E][ETH.cpp:324] begin(): esp_eth_driver_install failed
[21:30:06][  1578][E][vfs_api.cpp:105] open(): /littlefs/config/emsespModules.json does not exist, no permits for creation
[21:30:06][21:30:06]

```

proddy, 2024-08-02:
Thanks for reporting, that doesn't look good. We've upgraded to the latest Core espressif chip libraries so that might have something to do with it. We'll try and reproduce and fix.

You're not getting a boot loop are you? just unable to read the config files from the little File System?

proddy, 2024-08-02:
ahh, I see the problem. You uploaded the wrong firmware. The older S32 gateways are 4MB and you used the latest S3 which is 16MB. Take this build: https://github.com/emsesp/EMS-ESP32/releases/download/latest/EMS-ESP-3_7_0-dev_29-ESP32.bin

There was an old bug in the WebUI which downloaded the wrong firmware version. That has now been fixed.

closed 2024-08-02 by proddy