emsesp/EMS-ESP32 #1910: Translation unclarity for german language at setburnpow vs. selburnpow

dleding, 2024-08-01:
Just noticed, while going through the lists out of curiosity, that i get 2 values which display as identical german names:

https://github.com/emsesp/EMS-ESP32/blame/1487f30c433e6fc87ecbba42c04f9f9ea483dd4f/src/locale_translations.h#L299

i think this line would need an additional " maximale " in between.

Sorry for myself being "such a Caren", this product is getting more awesome every day. *thumbs up*

proddy, 2024-08-02:
I see.

`selBurnPow` and `setBurnPow` are both "Sollwert Brennerleistung"

MichaelDvP, 2024-08-02:
Yes, it's the same value. selburnpow is the value from boiler telegrams, setburnpow is taken from the thermostat->boiler command. 
I think in the beginning it was not clear if this is the same value, so keep both,
- remove setburnpow - we'll get omplains about missing value
- rename one of them - we'll get complains about renaming and questions about the difference
- keep it - we'll get complains about doule names
What to do?
(i prefere to remove setburnpow)

dleding, 2024-08-02:
> Yes, it's the same value. selburnpow is the value from boiler telegrams, setburnpow is taken from the thermostat->boiler command. I think in the beginning it was not clear if this is the same value, [...]

For my system, removing setburnpow would be sufficient, as it never got a value.

In general, the german translation is the only one where these names are identical, therefore my idea to insert " maximale " so that it equals the english and other translated description.

Keeping it as it is would leave the doubled name only for german users.

I'm not entirely sure, but think that there may come cases where the value is actually used:
In the bottom right part of page 4 in the manual to CW100 and CR100 is a description of the device directly changing the burner power when set to "Leistungsregelung":
https://docplayer.org/41788014-Installationsanleitung-fuer-den-fachmann-cr100-cw100-bedieneinheit-ems-o-2014-07.html

So i tend to prefer just altering the translation to match the others.

MichaelDvP, 2024-08-02:
In my build i've removed the useless values from Boiler: setBurnPow, setFlowTemp, wwSetPumpPow (the last is completly wrong)
The corresponding boiler values are: selburnpow, selflowtemp and wwmaxpower. 
setburnpow and setflowtemp are used in alert module when configured as analog command module.
@proddy removing values have a side-effect that modbus list has holes in register numbering. Don't know if this harm.

proddy, 2024-08-02:
It's good - check it in.

@mheyse please note some changes in the ordering

dleding, 2024-08-03:
Thank you for adressing!

commit 276a1964 (2024-08-03): Merge pull request #1917 from MichaelDvP/dev

fix #1910 and small changes
closed 2024-08-03 by dleding