emsesp/EMS-ESP32 #1920: Updates to webUI

proddy, 2024-08-04:
I'm planning to make some changes to the WebUI, and noting them down here so I don't forget

- [ ] upgrade Alova from version 2 to 3. This is done, but there's a bug in Alova (rendering with preact) that is blocking the PR
- [ ] refactor all the updateState() to use methods
- [ ] use AutoRefreshRequest() in devices, sensors, custom entities, status (activity, network, hardware, mqtt, ntp, ap) and remove the Refresh button where appropriate.
- [ ] uppercase "Never" "Always" etc in the AP, to make it consistent
- [ ] Rename Customization to Customizations in .tsx filenames to be consistent with naming in UI
- [ ] Remove the trailing spaces code, which router-dom does now
- [ ] Add translations for Immediate, used in the Scheduler
- [ ] Tidy up Application settings, moving access control to the User settings page, and creating a Hardware Settings section
- [ ] Remove history package
- [ ] Replace Functional Components (FC) with normal JSX functions if we're not passing down children props

mattreim, 2024-08-04:
And maybe a few more translations.

proddy, 2024-08-04:
> And maybe a few more translations.

that would be nice. We really leave it up to the community to do the translations (I'm not fluent in 10 languages, unfortunately!). 

I'll do a broadcast and ask for people to help.

proddy, 2024-08-10:
Looks like Alova 3 is working again so I'll make a PR. Only remaining issue seems to be the Upload progress bar.

proddy, 2024-08-12:
For the Application page I was thinking 

```
Services
 - API
   - Bypass Access Token
 - Telnet (on/off)
 - Modbus (on/off)
   - Max Clients, Port, Timeout
 - Syslog (on/off)
   - Host, Port, Log level, Mark interval
 - Sensors
	- enable Analog Sensors
	- enable Temperature parasite power
 - Shower Timer
	- Wait time
	- enable shower alert
	  - Trigger Time, Cold shot duration
 
Formatting Options
 - Language (for device entities)
 - Boolean Format Dashboard (change Dashboard to Web)
 - Boolean Format API/MQTT
 - Enum Format format
 - Convert temperature values to Fahrenheit
 - Log EMS in hex

Hardware Options
 - Board profile
 - EMS Tx Mode
 - EMS BUS ID
 - Enable read-only
 - Hide LED
 - underclock CPU speed

Other Options
 - Start boiler forced heating off
 - Disable remote on missing roomtemperature (..and fixing the spacing here)
 ```

bbqkees, 2024-08-13:
Looks good but I would put the shower timer in the Other Options as the logic is generated in EMS-ESP like the 'start boiler forced heating off''

And Sensors maybe its own header?
Also maybe change 'enable Temperature parasite power' to something like 'enable temperature sensor 1-Wire Parasite-Power'', as that's the official term for clarity.
[image]

proddy, 2024-08-15:
@bbqkees can you review the latest changes to the application settings page at https://demo.emsesp.org/settings/application

proddy, 2024-08-17:
I'll add some code to automatically download and update the firmware from the GitHub assets.

proddy, 2024-08-18:
> I'll add some code to automatically download and update the firmware from the GitHub assets.

Created this PR https://github.com/emsesp/EMS-ESP32/pull/1951 but still not 100% working from the WebUI.

closed 2024-09-01 by proddy