emsesp/EMS-ESP32 #1922: WiFi scan not working when SSID is empty

proddy, 2024-08-04:
with dev-29:

Two use cases
1. On ethernet (E32V2 gateway), Ethernet only and SSID is blank. Scan WiFi comes up empty. The option should be disabled if we're not using WiFi and have an Ethernet connection.
2. EMS-ESP starts its own Access Point. SSID is blank. Scan WiFi comes up blank. URL is http://192.168.4.1/settings/network/scan

@tp1de FYI

MichaelDvP, 2024-08-04:
I don't think there is a bug. 
1. To change from ethernet to Wifi it's usefull to scan networks and connect to wifi. After restart wifi is connected and ethernet is disabled. Works as expected. 
2. Works as expected.

Tested both with E32V2, actual 3.7.0-dev29-16M from this GH.

proddy, 2024-08-04:
Ok, I'll need to test again.

I erased an ESP32, flashed it, went to the Access Point on 192.168.4.1 and WiFi scan didn't work. But I'll try again.

proddy, 2024-08-05:
I was testing on Windows as well. I try with Android and iPhone...

** edit - on iPhone, in AP/Captive Portal Wifi Scan still comes up as blank **

tp1de, 2024-08-05:
@proddy Does WiFi work for you if you manually enter SSID and password?

proddy, 2024-08-05:
> @proddy Does WiFi work for you if you manually enter SSID and password?

yes, that works.

thorstenfleischmann, 2024-08-08:
I think I have the same issue. Just installed ems-esp and scan in initial ap mode does not find any networks. Quite sure there should be at least one. Will try manually later.

proddy, 2024-08-09:
I think I found a fix. We need to add a `WiFi.softAPenableIpV6();` in `APSettingsService::startAP()` before the WiFi.softAP() call. Tested and works.

proddy, 2024-08-16:
in dev-31

closed 2024-08-16 by proddy