emsesp/EMS-ESP32 #1926: serial shell not working

MichaelDvP, 2024-08-05:
in actual dev29 i can't connect to serial shell. Tested with 16M version from this server. Seems to be this:
https://github.com/emsesp/EMS-ESP32/blob/38ae744a10f11db709caedf2376cc2f0d9e4bfd0/src/emsesp.cpp#L1542

proddy, 2024-08-05:
ooops, I'll fix now

MichaelDvP, 2024-08-05:
Confirm, it's working again, closing.
BTW: With ^D or exit the shell closes, also serial shell, that can not be reactivated (except restart ems-esp). Is this intended?

proddy, 2024-08-05:
> Confirm, it's working again, closing. BTW: With ^D or exit the shell closes, also serial shell, that can not be reactivated (except restart ems-esp). Is this intended?

No, its not supposed to do this. 

I remember in the first version when the serial was started you had to do a key stroke to start the new Shell console session. It was to save memory on the ESP8266. I'll see if I can find that old code. What we can also do is test for `shell_` being null and if so start it again with `shell_->start();`. I'll leave this one open.

commit 66ed8a1e (2024-08-05): fixes #1926 - serial shell not working
closed 2024-08-07 by MichaelDvP